ros-navigation/navigation2 #1557: Turtlebot world doesn't show up when running tb3_simulation_launch.py

crdelsey, 2020-02-21:
## Bug report

When running the nav2 stack through the provided tb3_simulation_launch.py launch file, Gazebo starts and the turtlebot shows up, but not the turtlebot world

#### Steps to reproduce issue
```
> source navigation2_ws/install/setup.bash
> ros2 launch nav2_bringup tb3_simulation_launch.py
```

#### Expected behavior

[image]

#### Actual behavior

[image]

#### Additional information

I'm just submitting the issue for documentation purposes, in cas someone else runs into this.

This problem is caused by failing to add the `turtlebot3_simulations/turtlebot3_gazebo/models/` directory to the gazebo model path. The fix is to run

```
export GAZEBO_MODEL_PATH=~/<path to turtlebot3_ws>/src/turtlebot3/turtlebot3_simulations/turtlebot3_gazebo/models/
```

before running the launch file

crdelsey, 2020-02-21:
I'm just submitting this issue because several of us were running into the problem. The fix at the end of the description above. We can close this as soon as a few people have had a chance to see this.

SteveMacenski, 2020-02-21:
It appears I have it in my bashrc

```
export TURTLEBOT3_MODEL=waffle
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/home/s.macenski/Documents/turtlebot3_ws/src/turtlebot3/turtlebot3_simulations/turtlebot3_gazebo/models
```

Good thinking past me.

closed 2020-02-23 by SteveMacenski