ros-navigation/navigation2 #1558: Navigation with SLAM Toolbox - nothing happens

Leonti, 2020-02-23:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 19.10
- Version or commit hash:
  - 8793b3a806b51f21bc88526e0896e332bffc3e7d
- DDS implementation:
  - default one

#### Steps to reproduce issue
1. Start gazebo simulation with Lidar and Slam Toolbox  
Here is my repo where my config is located:  https://github.com/Leonti/ros-rover
It's nothing more than a robot model, room world and launch script to start simulation, RViz and Slamm Toolbox  
2. In another window start Navigation with:  

```
ros2 launch nav2_bringup nav2_navigation_launch.py
```

This is what I see in RViz after navigation has started:  
[image]

#### Expected behavior
After running
```
ros2 topic pub /move_base_simple/goal geometry_msgs/PoseStamped "{header: {stamp: {sec: 0}, frame_id: 'map'}, pose: {position: {x: 0.2, y: 0.0, z: 0.0}, orientation: {w: 1.0}}}"
```
I expect to see robot move or at least see some log messages in the navigation output
#### Actual behavior
Robot doesn't move, console output stays the same
#### Additional information

I did change scan topic in `nav2_params.yaml` to be:
`local_costmap` -> `local_costmap`  -> `ros__parameters` -> `obstacle_layer` -> `topic: /rover/laser_scan`  
`global_costmap` -> `global_costmap`  -> `ros__parameters` -> `obstacle_layer` -> `topic: /rover/laser_scan`  
and `robot_base_frame` to be:
`local_costmap` -> `local_costmap`  -> `ros__parameters` -> `robot_base_frame: chassis`  
`global_costmap` -> `global_costmap`  -> `ros__parameters` -> `robot_base_frame: chassis`

Here is the full console output:  
[navigation2_output.txt](https://github.com/ros-planning/navigation2/files/4240516/navigation2_output.txt)

It has a bunch of warnings like:  
```
[controller_server-1] Warning: Invalid frame ID "chassis" passed to canTransform argument source_frame - frame does not exist
[controller_server-1]          at line 133 in /home/leonti/ros2_eloquent/src/ros2/geometry2/tf2/src/buffer_core.cpp
```
but they stop after everything is started.  
I also intemittently get these after everyhting has started:  
```
[controller_server-1] [INFO] [local_costmap.local_costmap_rclcpp_node]: Message Filter dropping message: frame 'chassis' at time 177.450 for reason 'Unknown'
[planner_server-2] [INFO] [global_costmap.global_costmap_rclcpp_node]: Message Filter dropping message: frame 'chassis' at time 176.220 for reason 'Unknown'
[planner_server-2] [INFO] [global_costmap.global_costmap_rclcpp_node]: Message Filter dropping message: frame 'chassis' at time 176.240 for reason 'Unknown'
```

SteveMacenski, 2020-02-23:
Does this work with amcl rather than slam toolbox?

There should be no relationship between slam toolbox and navigation if you’re not seeing tons of TF errors every time you try to go to pose. 

Have you started with our basic tutorials and moved from there? Why not common with rviz using our provided action server tool? Does that work?

Also what branch is that on? We don’t support specific issues in 19.04 but I don’t think that’s likely related to your issue.

Leonti, 2020-02-23:
Hi @SteveMacenski!
Thanks for the reply!

My repo has no navigation launch files because I'm using the ones from this repo:
https://github.com/ros-planning/navigation2/blob/8793b3a806b51f21bc88526e0896e332bffc3e7d/nav2_bringup/bringup/launch/nav2_navigation_launch.py
`nav2_params.yaml` is the same as in the repo, no changes at the moment. 
Is `nav2_navigation_launch.py` not the one I should be using?
My launch file for Gazebo and Slam Toolbox is here:
https://github.com/Leonti/ros-rover/blob/master/rover_gazebo/launch/rover.launch.py

That "chassis" frame error was intermittent and caused by a `gazebo` process running in the background, I killed it and now I get no errors except for an occasional:
```
Message Filter dropping message: frame 'base_link' at time 76.610 for reason 'Unknown'
```
But you are right, it's not related to navigation, I get in in Slam Toolbox as well and it doesn't affect the mapping.

It seems like my transforms are all fine (I renamed `chassis` to `base_link` for simplicity):
[image]

I followed this tutorial: https://ros-planning.github.io/navigation2/getting_started/index.html#running-the-example and it worked fine, encouraged by it I moved to SLAM and this is where I have issues.

I'm on a branch `eloquent-devel`, commit hash: 8793b3a806b51f21bc88526e0896e332bffc3e7d
I'm running `eloquent` built from source on 19.10. I don't think it's a 19.10-specific issue since everything else works fine

`ros2 topic list` shows that `/move_base_simple/goal` exists and when I use RViz `2D Goal Pose` it sends messages there.

What do you mean by "Action Server Tool", is it this panel?
[image]

Is there an example of how I can run it with `amcl` instead of Slam Toolbox?

Thanks again for your reply!

Leonti, 2020-02-23:
@SteveMacenski you were right - there are 2 topics:
`/goal_pose` and `/move_base_simple/goal`

Sending a message to `/goal_pose` makes it work (the robot is still not moving, but I can deal with it, at least I see something is happening) 
```
ros2 topic pub /goal_pose geometry_msgs/PoseStamped "{header: {stamp: {sec: 0}, frame_id: 'map'}, pose: {position: {x: 0.2, y: 0.0, z: 0.0}, orientation: {w: 1.0}}}"
```

I still don't understand how Turtlebot example works - if Rviz is sending messages to `/move_base_simple/goal` when I click on the `2D Goal Pose`, how is it able to receive them on `/goal_pose`? Or maybe RViz is somehow configured to send messages to `/goal_pose`?

SteveMacenski, 2020-02-23:
Rviz does not send on the move base topic for ROS2. It sends on the goal pose topic. 

Sounds like you found your issue! Closing out

closed 2020-02-23 by SteveMacenski

Leonti, 2020-02-24:
Ok, now I know what happened - my setup was copied from another project which had topic set to:
```
    - Class: rviz_default_plugins/SetGoal
      Topic: /move_base_simple/goal
``` 
This is why RViz didn't work.

Next thing I did is I looked at the docs https://ros-planning.github.io/navigation2/tutorials/docs/navigation2_with_slam.html which had `/move_base_simple/goal` as well so this is why sending message to a topic directly didn't work either.

I created a PR to fix the docs:
https://github.com/ros-planning/navigation2/pull/1559