ros-navigation/navigation2 #1567: xy_goal_tolerance param name is different in simple_goal_checker and rotate_to_goal

Jconn, 2020-02-26:
The [rotate to goal](https://github.com/ros-planning/navigation2/blob/master/nav2_dwb_controller/dwb_critics/src/rotate_to_goal.cpp#L58) `xy_goal_tolerance` looks for a different param than the [simple goal checker]( https://github.com/ros-planning/navigation2/blob/master/nav2_dwb_controller/dwb_plugins/src/simple_goal_checker.cpp#L64) `xy_goal_tolerance`

This is a difference from dashing, I assume it is unintended but didn't look hard to see if it is. 

It looks like the simple goal checker param name is what matches the `nav2_params.yaml`, so I'm thinking that is the right name?

Side note: param name mismatches between yaml and code is the most common bug I hit in ros2 in every package I work with and it can be hard to find, especially when you don't have an intuition for what the param does. Can anyone imagine a way to reduce these bugs?

SteveMacenski, 2020-02-26:
> This is a difference from dashing, I assume it is unintended but didn't look hard to see if it is.

I can't imagine it was, PRs welcome.

> so I'm thinking that is the right name?

What are the 2 names?

>  Can anyone imagine a way to reduce these bugs?

Not providing any default configfile ;-) but that's not a good answer either. We could run some static analysis tests to check all declares and check the yaml and report an error if not aligned. That would be a little hard though.

Jconn, 2020-02-26:
in `SimpleGoalChecker` we have:
```
  nav2_util::declare_parameter_if_not_declared(
    nh,
    plugin_name + ".xy_goal_tolerance", rclcpp::ParameterValue(0.25));

````

and in `rotate_to_goal` we have:
```
  xy_goal_tolerance_ = nav_2d_utils::searchAndGetParam(
    nh_,
    dwb_plugin_name_ + "." + name_ + ".xy_goal_tolerance", 0.25);
```

SteveMacenski, 2020-02-26:
Ah ok. Mhm. 

I definitely do not want to make the `FollowPath.<critic>.xy_goal_tolerance` the new default, it should be be probably the `FollowPath.xy_goal_tolerance` while the goal checker is still inside of DWB's implementation.

Jconn, 2020-02-26:
makes sense to me. I'll get a PR for this early next week

SteveMacenski, 2020-03-02:
Any update here? 

Moving to "In Progress" on the V1 project

Jconn, 2020-03-02:
I'll have a PR soon. I forgot about this, thanks for the ping

closed 2020-03-02 by SteveMacenski