ros-navigation/navigation2 #1570: Planner update logic change

SteveMacenski, 2020-02-27:
Rather than planning at 1 hz, I'd like to propose the following, where in the prior iteration we stored the path and the path cost:

At 10hz (or something) we:
- Check if old path has the same, lower, or higher cost
- If lower, same or slightly higher, we continue on it
- If much higher by some statistical value, we replan and replace the original plan
- If legal obstacle, we replan

The goal here is to make it so that we're only replanning when required and then can use planning methods that take a bit longer than 100ms because they're not updating necessarily often. Additionally, that helps widely remove oscillatory behavior where N similar cost plans down different routes are being changed between. Also, if you have many robots in an environment, going through the orchestration of many paths changing at 1hz for `N` robots sounds like a pain. I'd rather only update my global model of robots when required

orduno, 2020-03-09:
Just to clarify, on your proposal, the path is computed (replanned) at a given fixed-rate but will only be updated (on the controller) if some criteria are met?

I like the idea. We would need to clarify the implementation, are you thinking of keeping the logic for assessing an update on `BtNavigator` or the `PlannnerServer`. There are some small issues on either.

I think we should treat replanning because of invalid replan as a separate issue.

SteveMacenski, 2020-03-09:
> Just to clarify, on your proposal, the path is computed (replanned) at a given fixed-rate but will only be updated (on the controller) if some criteria are met?

No, 1 of the 2 options have the exact opposite interpretation of this. There will be a very frequent check if replanning is required, but it will not replan unless actually necessary.

> I like the idea. We would need to clarify the implementation, are you thinking of keeping the logic for assessing an update on BtNavigator or the PlannnerServer. There are some small issues on either.

One step at a time. I would rather come to consensus about what makes sense thematically before being bogged down in technical specifics. We should chose the right solution and then figure out how to do it.

orduno, 2020-03-09:
Ok, so as the robot is making progress on a path, the navigator runs the path periodically through some evaluations (validity, current pose deviation, etc.) and triggers replanning when needed. Navigator would still be replanning at a (smaller than current) rate, however, the new and previous paths are compared (cost, variation, etc.) and if beyond some threshold.

SteveMacenski, 2020-03-09:
>  as the robot is making progress on a path, the navigator runs the path periodically through some evaluations 

Correct, by option 1

> and triggers replanning when needed

Correct

> Navigator would still be replanning at a (smaller than current) rate,

No, it would never replan unless required, by option 1

>  the new and previous paths are compared (cost, variation, etc.) and if beyond some threshold.

That's option 2

orduno, 2020-03-09:
Ok, so these are mutually exclusive. I don't have a good example at the moment for having both enabled and work as I describe. 

You've mentioned issues/requirements around path oscillation, path computation time, path computation halting. Any others?

SteveMacenski, 2020-03-09:
Those are the big ones.

We could also have some modest replanning in the background - but I generally like the idea of no replanning unless there's value in it to conserve resources. And as I mentioned, if you're working with distributed multirobot systems, I imagine replanning on a frequent basis can get real-old real-fast.

But we could still enable the ability to do replanning, but have it severely pulled back (once every 30 seconds or something).

orduno, 2020-03-10:
I like the event-based approach of option 1.

I think it would be helpful to discuss some scenarios. For example, how will a map update propagate to trigger replanning?

SteveMacenski, 2020-03-10:
New sensor data being processed so you see that the old path now has something in the way like a box or a moved shelf. Or has something that is closer to the path than is comfortable and may want to make a new path.

If a person is moving in the environment and blocks the path then that would also trigger a replan. Its just a trivial integration of costs and then some heuristic or probabilistic model for "the cost has increased enough". I don't know off hand what I'd choose to do that with, but I'm sure we can come up with something as we go.

orduno, 2020-03-12:
That would be for the case where replanning is triggered by obstruction from dynamic obstacles.

We also need to consider how it would be triggered (for the event-based approach) if the static map was updated (that's what I meant to ask above).

SteveMacenski, 2020-03-12:
That would also be reflected in the costmap and if the new path causes a collision or a much higher cost, then the it would trigger a replan.

Keep in mind that the planner only see a costmap -- so whether the updates are from sensors, from moving obstacles, or from the map, via the static layer, it appears to the planner the same way as a new or resized planning space. 

The situation that brings up that is a little unique is what happens if the updated map shows a shorter path option. The motivational question I ask is if a plan exists that is valid, is it important to cause a replan due to this criteria? My knee-jerk answer is "no", but there may be cases that is useful.

We could in theory have a replanning rate in addition to the event-based replanning, but that should be very infrequent (on the order of every minute or more) so if its that infrequent, you'll miss a number of these opportunistic shorter paths. Which circles back to the "is this that important?" 

Thoughts?

shrijitsingh99, 2020-05-05:
Bringing back the discussion.
Opinions on doing something similar to [dlux_global_planner](https://github.com/locusrobotics/robot_navigation/tree/master/dlux_global_planner) + event based planning?

SteveMacenski, 2020-05-05:
@orduno re-reading now, I think I get your point now. You mean how do we deal with a map update loop if we're replanning at inconsistent intervals. The costmap is updated on a constant rate of a background thread. We could optionally pull that back to also being event based and really leveraging the observation buffers to cull out old information, or keep it spinning at whatever rate but its just not being utilized regularly. I like that first option, but we'd have to carefully look through the stack and parameters and make sure we do that right. Option 2 is more overhead but less up front effort. 

@shrijitsingh99 I think David's method was my option 2 in the upper thread: replan regularly but only take paths that are "sufficiently" different. What I proposed instead was to replan only if the existing path has changed cost to be "sufficiently" higher than than when originally planned. Ex plan -> drive -> drive -> drive -> new sensor readings come in and there's something off to the left side of the path, but doesnt change the cost of the path too much and not overlapping lethal space -> drive ->  drive -> drive a raccoon pops out from behind a corner and overlaps with the path as a lethal obstacle -> replan -> drive. So the replanning only happens based on the cost/length of the last path becoming itself "sufficiently" different to warrant a change. David is taking the length and cost after replanning and comparing them to decide if to use the new or old path recursively. 

I like my idea better but open for disagreements. I can see arguments for both.

shrijitsingh99, 2020-05-07:
I can see the use case for both approaches. 
The first one would be useful to deal with planners who are quick to update but suffer from oscillating changing paths.
The second approach fits with what you said to avoid replanning when the planner is long-running.

The best way to go about would be a hybrid approach, check if a better path is available when the time passed since the last update is greater than a threshold ( >10s)  and re-plan if the current's paths cost has sufficiently changed. 
If people want the first approach only they can set the sufficiently changed threshold to something large.
If they want the second approach they can increase the time since last update threshold.

SteveMacenski, 2020-05-07:
Replanning less often also deals with oscillation caused by any  non-SE2 planner like all the ones we currently have. Because it can't distinguish orientation, if you're in the middle of an aisle from a goal (similar to your DWB ticket) it can oscillate back and forth based on really small changes in cost or sensor readings. I have experienced this first hand in a product.

I'm not sure how the option to replan but compare costs / length would help that situation because the length would be similar and the cost would also be similar, but in completely different directions. There'd also have to be some integral change test to reject similar length / cost paths that are too "different". Similarly, the argument could be made that you should actually be rejecting paths that are _too_ similar. Unclear how to resolve the oscillation problem by comparing 2 very different path.

 > If people want the first approach only they can set the sufficiently changed threshold to something large

I think what we'd actually do is have a `use_caching` param and if off, only does replanning. If on, it does caching and when the planner is called we check the changes and decide if we replan or not. I think this might involve having a couple of new behavior trees though so we can call it based on a replan rate like we do today, but then also have a call to the planner server at the BT tick rate to check the path cost and if it should replan at that iteration. 

The nice thing about that would be that we can replan super often in highly-dynamic situations.

SteveMacenski, 2021-09-14:
https://github.com/ros-planning/navigation2/issues/2374 meets the intent and is being actively worked on. Closing

closed 2021-09-14 by SteveMacenski