ros-navigation/navigation2 #1585: Fix flaky tests

SteveMacenski, 2020-03-07:
The costmap planner test and the action_utils test are flaky. We should get them back up to >80%

gimait, 2020-04-07:
I've taken a look at the action_utils tests, and I found that variables are not used correctly, so I updated them to see if that fixed the issue ([735858e](https://github.com/gimait/navigation2/commit/735858ed2150adbf8b0f616528e35c06041d0fec)).
But that didn't really fix it for me, I still get the error: ```"Goal handle is not known to this client."``` from this line:
https://github.com/ros2/rclcpp/blob/01a6befdde7f78b6f7788e6180cc5ed38390e955/rclcpp_action/include/rclcpp_action/client.hpp#L407
Caused because the client's ```goal_handles_``` variable is empty when the call to ```async_get_result(preepmt_handle)``` is done.

There is two functions where the goals can be erased from the ```goal_handles_``` list: 
```handle_status_message``` https://github.com/ros2/rclcpp/blob/01a6befdde7f78b6f7788e6180cc5ed38390e955/rclcpp_action/include/rclcpp_action/client.hpp#L586
and ```make_result_aware``` https://github.com/ros2/rclcpp/blob/01a6befdde7f78b6f7788e6180cc5ed38390e955/rclcpp_action/include/rclcpp_action/client.hpp#L616

What I've seen is that the ```handle_status_message``` function is called with all actions finished (both action and preemption) before the call to  ```async_get_result(preepmt_handle)``` in the test.
Because of that, the client does no longer contain any ```goal_handle``` when ```async_get_result(preepmt_handle)``` is called, producing the mentioned exception and failing the test.

Have you seen this issue before, @SteveMacenski ?

SteveMacenski, 2020-04-07:
Don't worry about the action utils issues, I believe its an upstream issue

gimait, 2020-04-07:
Yes I think that's the issue with the action utils.
Although I would still like to change ```future_goal_handle``` with ```future_preempt_handle``` here:
https://github.com/ros-planning/navigation2/blob/db4193fd87dbd9ffc12237f47814704793ec405b/nav2_util/test/test_actions.cpp#L446

SteveMacenski, 2020-04-07:
Mhm, that test is complex to dissect since its intentionally trying to test a race condition. Its not immediately clear to me if that was the intention to cause the race condition or not.

gimait, 2020-04-07:
I mean, I guessed that the intention here
https://github.com/ros-planning/navigation2/blob/db4193fd87dbd9ffc12237f47814704793ec405b/nav2_util/test/test_actions.cpp#L442-L446
would be to wait for the second goal to be sent, otherwise it doesn't really matter to have that spin (is the same one as few lines before).

AlexeyMerzlyakov, 2020-04-15:
I've found that segfaults in `nav2_system_tests/test_planner_random_node` and `nav2_system_tests/test_planner_costmaps_node` appear because of dereferencing zero-length vector in path generated by planner in cases when it can not produce right path.

The following patch seems to fix segfaults in these tests for me:
```
diff --git a/nav2_system_tests/src/planning/planner_tester.hpp b/nav2_system_tests/src/planning/planner_tester.hpp
index a9dfaebd..bf574860 100644
--- a/nav2_system_tests/src/planning/planner_tester.hpp
+++ b/nav2_system_tests/src/planning/planner_tester.hpp
@@ -83,6 +83,12 @@ public:
     }
     try {
       path = planners_["GridBased"]->createPlan(start, goal);
+      // The situation when createPlan() did not throw any exception
+      // does not guarantee that plan was created correctly.
+      // So it should be checked additionally, that path is correct.
+      if (!path.poses.size()) {
+        return false;
+      }
     } catch (...) {
       return false;
     }

```

However, it does not fix the segfaults at all: new problem uncovers after patch was applied. The second problem appears after testcase was passed and `PlannerTester` is starting to destroy all objects and threads. After some digging with GDB it looks like that problem appears somewhere in Costmap2D's `mapUpdateLoop()` when it is trying to use already destroyed instance:
```
...
[WARN] [PlannerTester]: Path deviates from requested start and end points
[WARN] [PlannerTester]: Failed with start at 185.00, 325.00 and goal at 149.00, 88.00
[INFO] [PlannerTester]: Tested with 100 tests. Planner failed on 83. Test time 337 ms
[Thread 0x7fffe1599700 (LWP 18460) exited]
[INFO] [nav2_planner]: Cleaning up
[Thread 0x7fffd1ffb700 (LWP 18465) exited]
[INFO] [global_costmap.global_costmap]: Cleaning up
[Thread 0x7fff59ffb700 (LWP 18493) exited]
[Thread 0x7fff5b7fe700 (LWP 18490) exited]
[Thread 0x7fff5affd700 (LWP 18491) exited]
[Thread 0x7fff5a7fc700 (LWP 18492) exited]
[Thread 0x7fff5bfff700 (LWP 18489) exited]
[Thread 0x7fff597fa700 (LWP 18494) exited]
[Thread 0x7fff977fe700 (LWP 18476) exited]
[INFO] [nav2_planner]: Cleaning up plugin GridBased of type NavfnPlanner
[INFO] [nav2_planner]: Destroying plugin GridBased of type NavfnPlanner
[Thread 0x7fffe27fc700 (LWP 18459) exited]
[INFO] [nav2_planner]: Destroying
[Thread 0x7fff68ff9700 (LWP 18488) exited]
[INFO] [global_costmap.global_costmap]: Destroying
[Thread 0x7fff697fa700 (LWP 18487) exited]
[Thread 0x7fff6affd700 (LWP 18484) exited]
[Thread 0x7fff6a7fc700 (LWP 18485) exited]
[Thread 0x7fff69ffb700 (LWP 18486) exited]
[Thread 0x7fff6bfff700 (LWP 18482) exited]
[Thread 0x7fff6b7fe700 (LWP 18483) exited]

Thread 43 "test_planner_ra" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff58ff9700 (LWP 18495)]
0x00007ffff7648426 in nav2_util::getCurrentPose(geometry_msgs::msg::PoseStamped_<std::allocator<void> >&, tf2_ros::Buffer&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, double) ()
   from /home/leha/navigation2_ws/install/nav2_util/lib/libnav2_util_core.so
(gdb) bt
#0  0x00007ffff7648426 in nav2_util::getCurrentPose(geometry_msgs::msg::PoseStamped_<std::allocator<void> >&, tf2_ros::Buffer&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, double) ()
   from /home/leha/navigation2_ws/install/nav2_util/lib/libnav2_util_core.so
#1  0x00007ffff78b9a8a in nav2_costmap_2d::Costmap2DROS::getRobotPose (this=this@entry=0x555555e07190, 
    global_pose=...)
    at /home/leha/navigation2_ws/src/navigation2/nav2_costmap_2d/src/costmap_2d_ros.cpp:537
#2  0x00007ffff78be40f in nav2_costmap_2d::Costmap2DROS::updateMap (this=this@entry=0x555555e07190)
    at /home/leha/navigation2_ws/src/navigation2/nav2_costmap_2d/src/costmap_2d_ros.cpp:438
#3  0x00007ffff78bea85 in nav2_costmap_2d::Costmap2DROS::mapUpdateLoop (this=0x555555e07190, 
    frequency=<optimized out>)
    at /home/leha/navigation2_ws/src/navigation2/nav2_costmap_2d/src/costmap_2d_ros.cpp:396
#4  0x00007ffff567d6ef in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff59506db in start_thread (arg=0x7fff58ff9700) at pthread_create.c:463
#6  0x00007ffff4d3a88f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

SteveMacenski, 2020-04-15:
Looking at the common logs for these failures: https://10204-135363400-gh.circle-artifacts.com/0/opt/overlay_ws/log/test/nav2_system_tests/stdout.log my impression was something to do with TF, do you see a crash there that I'm missing? 

That still looks like an issue that may cause random problems once its up - please do submit that diff as a PR and I'll merge it. Thanks!

AlexeyMerzlyakov, 2020-04-16:
The problem appeared in CI log you've mentioned - is another new one problem. It concerned with timeout while waiting for transform in `Costmap2DROS::on_activate()`. Unfortunately, the exact failure did not reproduced locally on my PC. However I can see similar messages for other testcases:
```
[80.470s] 13: [planner_server-8] [INFO] [global_costmap.global_costmap]: Activating
[80.472s] 13: [planner_server-8] [INFO] [global_costmap.global_costmap]: Checking transform
[80.473s] 13: [planner_server-8] [INFO] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[80.943s] 13: [planner_server-8] [INFO] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[81.445s] 13: [planner_server-8] [INFO] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[81.943s] 13: [planner_server-8] [INFO] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
...
```
causing them to timeout sometimes (for example `test_waypoint_follower` is being timed out from time to time locally). Therefore this might be a system problem appeared on some testcases.

The patch I've proposed above is fixing another failure concerned with segfaults on `test_planner_random_node` and `test_planner_costmaps_node` appeared with messages like:
```
...
[20.427s] 10: [test_planner_random_node-1] [INFO] [global_costmap.global_costmap]: Activating
[20.427s] 10: [test_planner_random_node-1] [INFO] [global_costmap.global_costmap]: Checking transform
[20.427s] 10: [test_planner_random_node-1] [INFO] [global_costmap.global_costmap]: start
[20.427s] 10: [test_planner_random_node-1] [INFO] [nav2_planner]: Activating plugin GridBased of type NavfnPlanner
[20.427s] 10: [test_planner_random_node-1] [INFO] [PlannerTester]: Loading map with file_path: /home/leha/navigation2_ws/src/navigation2/nav2_system_tests/maps/map.pgm
[20.428s] 10: [test_planner_random_node-1] [INFO] [mapio]: Loading image_file: /home/leha/navigation2_ws/src/navigation2/nav2_system_tests/maps/map.pgm
[20.476s] 10: [test_planner_random_node-1] [DEBUG] [mapio]: Read map /home/leha/navigation2_ws/src/navigation2/nav2_system_tests/maps/map.pgm: 400 X 400 map @ 1 m/cell
[20.477s] 10: [test_planner_random_node-1] [INFO] [PlannerTester]: Costmap: Setting static costmap
[20.477s] 10: [test_planner_random_node-1] 1586872498059401141
[20.481s] 10: [test_planner_random_node-1] [WARN] [nav2_planner]: GridBased: failed to create plan with tolerance 2.00.
[20.481s] 10: [test_planner_random_node-1] Magick: abort due to signal 11 (SIGSEGV) "Segmentation Fault"...
[21.475s] 10: [ERROR] [test_planner_random_node-1]: process has died [pid 3551, exit code -6, cmd '/home/leha/navigation2_ws/build/nav2_system_tests/src/planning/test_planner_random_node --ros-args -p use_sim_time:=True'].
[21.511s] 10: -- run_test.py: return code 250
[21.512s] 10: -- run_test.py: generate result file '/home/leha/navigation2_ws/build/nav2_system_tests/test_results/nav2_system_tests/test_planner_random.xml' with failed test
[21.512s] 10: -- run_test.py: verify result file '/home/leha/navigation2_ws/build/nav2_system_tests/test_results/nav2_system_tests/test_planner_random.xml'
[21.518s] 10/15 Test #10: test_planner_random ...............***Failed    3.66 sec
```

Therefore, we have 3 problems with `test_planner_random_node`/`test_planner_costmaps_node` testcases:
1. Timeouts while waiting for base_link -> map transform
    <- Reproduced in CI, not fixed yet
2. Segfault after planner failed to create a plan
    <- Reproduced locally, fixed by patch above
3. Segfault in the end of testing when `PlannerTester` is starting to destroy all objects and threads
    <- Reproduced locally, not fixed yet

AlexeyMerzlyakov, 2020-04-16:
Therefore, we have 3 flaky problems with `test_planner_random_node`/`test_planner_costmaps_node` testcases:
1. Timeouts while waiting for base_link -> map transform
```
[357.005s] 10: [test_planner_random_node-1] [INFO] [1586732285.265387992] [global_costmap.global_costmap]: Checking transform
[357.005s] 10: [test_planner_random_node-1] [INFO] [1586732285.265437726] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
```
<- Not fixed yet

2. Segfault after planner failed to create a plan:
```
[20.481s] 10: [test_planner_random_node-1] [WARN] [nav2_planner]: GridBased: failed to create plan with tolerance 2.00.
[20.481s] 10: [test_planner_random_node-1] Magick: abort due to signal 11 (SIGSEGV) "Segmentation Fault"...
[21.475s] 10: [ERROR] [test_planner_random_node-1]: process has died [pid 3551, exit code -6, cmd '/home/leha/navigation2_ws/build/nav2_system_tests/src/planning/test_planner_random_node --ros-args -p use_sim_time:=True'].
```
<- Fixed by patch above

3. Segfault in the end of testing when `PlannerTester` is starting to destroy all objects and threads
```
[INFO] [PlannerTester]: Tested with 100 tests. Planner failed on 87. Test time 343 ms
[INFO] [nav2_planner]: Cleaning up
[INFO] [global_costmap.global_costmap]: Cleaning up
Segmentation fault (core dumped)
```
<- Not fixed yet

SteveMacenski, 2020-04-28:
Planner tests were temporarily removed in this commit https://github.com/ros-planning/navigation2/commit/bc6a41d08dbf1c36c0c4e32b0a32393194d12ffd until an issue has been fixed so it can be brought back in

siddhya, 2020-05-03:
I am not getting a timeout but a segfault in the planner_random_node test (when run using colcon test)
```
> [7.354s] 10: [test_planner_random_node-1] [WARN] [PlannerTester]: Failed with start at 63.00, 58.00 and goal at 56.00, 236.00
> [7.362s] 10: [test_planner_random_node-1] Magick: abort due to signal 11 (SIGSEGV) "Segmentation Fault"...
```
However when running stand alone the test completes (./test_planner_random_node)

AlexeyMerzlyakov, 2020-05-04:
@siddhya Unfortunately, I can not reproduce the same issue manually, automatically (via colcon) or even synthetically (setting path as `{63,58} -> {56,236}`). Could you please share `streams.log` and `${HOME}/.ros/log/2020-05-03-...` related log-file if possible? Also, what commit are you on? Did you run `colcon test --packages-select nav2_system_tests` without any addition options or parameters?

siddhya, 2020-05-04:
I installed graphicsmagick and now the planner_random_node test is not failing. However

```
[193.203s] The following tests FAILED:
[193.203s]    7 - uncrustify (Failed)
[193.203s]   12 - test_bt_navigator (Failed)
[193.203s]   13 - test_bt_navigator_with_dijkstra (Failed)
[193.203s]   14 - test_dynamic_obstacle (Failed)
[193.203s]   15 - test_waypoint_follower (Failed)

```
This seems to be the cause. Looking into it.
`[44.613s] 12: [controller_server-7] [INFO] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
`

siddhya, 2020-05-04:
I tried running with the following two commands:
Option 1:
$ colcon test
Option 2:
$ export TEST_MAP=~/git/overlay_ws/src/navigation2/nav2_system_tests/maps/map.pgm
$ ./build/nav2_system_tests/src/planning/test_planner_random_node

I am on [commit id](https://github.com/ros-planning/navigation2/commit/c64ee48b06d3ed24c55b7aaf6089017a080057b2) with eloquent binaries installed. Minor changes to the code to set logging level to Debug.

AlexeyMerzlyakov, 2020-05-04:
Thank you for sharing!

I've also started testing in `while [ "$?" != "0" ]; do ...; done` loop until fail and finally reproduced the problem you've [mentioned](https://github.com/ros-planning/navigation2/issues/1585#issuecomment-623144415). This seems to be an issue No.3 from https://github.com/ros-planning/navigation2/issues/1585#issuecomment-614595069. The log message differs because of printf() I/O buffers. When I've started fflushing them, the messages became to be the same as in issue No.3 I've described.

Regarding [timeouts](https://github.com/ros-planning/navigation2/issues/1585#issuecomment-623379251), this is most likely an issue No.1 from the comment above which occasionally cause failures of different TC-es including test_planner_random_node.

So, there are two kinds of flaky problems remain to fix. Investigation is in progress.

SteveMacenski, 2020-05-04:
@siddhya you shouldn't be getting uncrustify issues since master is passing on that. There was a change on the linting profiles for Eloquent -> Foxy so I wouldn't worry about that warning. 

It is interesting that you see all of those other tests fail - that probably shouldn't be happening. Right now in master at least its only the planner tests and the action test that is failing regularly. 

You can select packages to run specifically with `colcon test --packages-select nav2_system_tests`. Because these take so long to run, I usually go into the CMakeLists and comment out the `add_directories` of the tests I'm not trying to build to just `planner`, for instance.

@AlexeyMerzlyakov any help you can give @siddhya would be great, he's been looking into this recently

siddhya, 2020-05-05:
Below patch seems to have fixed the segfault issue for me. Ran it in a loop for 15 mins. No failures. Let me know if it sounds reasonable and I will raise a PR.

```
diff --git a/nav2_system_tests/src/planning/planner_tester.cpp b/nav2_system_tests/src/planning/planner_tester.cpp
index 7202ba20..a6ef5a13 100644
--- a/nav2_system_tests/src/planning/planner_tester.cpp
+++ b/nav2_system_tests/src/planning/planner_tester.cpp
@@ -86,6 +94,7 @@ void PlannerTester::deactivate()
   spin_thread_.reset();

   auto state = rclcpp_lifecycle::State();
+  planner_tester_->onDeactivate(state);
   planner_tester_->onCleanup(state);

   map_timer_.reset();
diff --git a/nav2_system_tests/src/planning/planner_tester.hpp b/nav2_system_tests/src/planning/planner_tester.hpp
index 5b9c8ac9..438387fb 100644
--- a/nav2_system_tests/src/planning/planner_tester.hpp
+++ b/nav2_system_tests/src/planning/planner_tester.hpp
@@ -105,6 +105,12 @@ public:
     on_activate(state);
   }

+  void onDeactivate(const rclcpp_lifecycle::State & state)
+  {
+    on_deactivate(state);
+  }
+
   void onConfigure(const rclcpp_lifecycle::State & state)
   {
     on_configure(state);
```

SteveMacenski, 2020-05-05:
Looks very reasonable to me, I'd merge that.

AlexeyMerzlyakov, 2020-05-05:
Confirm the change. Calling `on_deactivate()` from parent `nav2_planner::PlannerServer` class will deactivate its members including `costmap_ros_`. `costmap_ros_` deactivation will cause joining `costmap2DROS::mapUpdateLoop()` thread before it will fail in the destructor. So, this is correct fix for the issue No.3.

I am chasing the patch for ~30minutes testing loop and there is no failure. I think, we can merge it. @siddhya thank you for the good job!

SteveMacenski, 2020-05-05:
I think that leaves us with the last issue of the TF timeout issue, after @siddhya submits this patch as a PR. 

Any thoughts on that one?

AlexeyMerzlyakov, 2020-05-05:
Yes, but currently just thoughts. I've analyzed the cause of timeouts for tests `12: test_bt_navigator`, `13: test_bt_navigator_with_dijkstra`, `14: test_dynamic_obstacle` and `15: test_waypoint_follower` which from time to time appears locally; and timeouts for `9: test_planner_costmaps` and `10: test_planner_random` testcases appearing only at CI servers. The nature of timeouts in first and second case is different. For tests 12-15 this appears because of waiting AMCL to correctly start and being localized, but for tests 9-10 there are no AMCL at all. The `/tf` is being published by testcase itself in `publishRobotTransform()` routine. Started costmap2d thread waits until `/tf` will be published and hangs around it. But I can not find any clue why this might happening on CI-server. `/tf` publishing logic in `PlannerTester` is very simple and always triggered locally. It looks like for some reasons on CI-server the published `/tf` messages not being delivered to costmap2d thread, e.g. for ros2 eloquent bug, incorrect environment or something else.

SteveMacenski, 2020-05-05:
the integration tests may fail on rare occasions but CI will run any failed package's tests twice, so I wouldn't be concerned about the bt_navigator / dynamic tests unless its failing > 30%. 

The planner costmap/random tests will regularly fail both the initial test and the retry so its >50% failure rate. I doubt its only in CI, if we ran the same docker environment  I bet can reproduce locally, but I haven't tried. 

When I was trying to debug the planner test issues, I tried shuffling around the TF stuff, putting into a new thread, having some sleeps for DDS to link everything etc, but couldn't figure it out. I had prints at one point making sure that it is theoretically publishing but its not receiving them. 

Maybe its possible to bypass TF here and just give it poses directly? It still highlights so weird issue that none of the other unit tests exhibit.

SteveMacenski, 2020-05-06:
Remaining issue is the TF issue

AlexeyMerzlyakov, 2020-05-06:
> I doubt its only in CI, if we ran the same docker environment I bet can reproduce locally, but I haven't tried.

Great. This might be very helpful for issue reproduction. Is there any information about how or where to get docker image used for testing in CI? Any information about it would be appreciable.

SteveMacenski, 2020-05-06:
The circleCI config should have the setup https://github.com/ros-planning/navigation2/blob/master/.circleci/config.yml

The release dockerfile might be something to look into as well for a simpler look at things: https://github.com/ros-planning/navigation2/blob/master/Dockerfile

siddhya, 2020-05-07:
> Remaining issue is the TF issue

For me the issue seems to be that gzserver takes a long time to come up. Till then the retries in test_InitialPose max out and it returns False leading to the Failed test. Below change fixed it for me.
`-        result = test_InitialPose(robot_tester, timeout=1, retries=10)
+        result = test_InitialPose(robot_tester, timeout=1, retries=100)`
Also increased the TIMEOUT in the CMakeLists.txt just in case.

I think it should wait in a while loop till InitialPose is received. Like it does in wait_for_node_active(). Let me know if that makes sense I can send out a fix.

AlexeyMerzlyakov, 2020-05-07:
> For me the issue seems to be that gzserver takes a long time to come up.

Thank makes sense for tests `test_bt_navigator`, `test_bt_navigator_with_dijkstra` and `test_dynamic_obstacle` which are using amcl+gzserver bundle. However, this does not applied to `test_planner_costmaps` and `test_planner_random` whose don't use amcl or gzserver. TF is being published there by testcase itself.

naiveHobo, 2020-05-08:
A question regarding the ```PlannerTester::updateRobotPosition``` function. Can anyone explain this line: ```base_transform_->header.stamp = now() + rclcpp::Duration(250000000);```.
Why is ```rclcpp::Duration(250000000)``` added to the current time? 

I just wrote a test for a BT node plugin which publishes its own transforms and I didn't find a need for that addition.

Also, I think an update to the ```PlannerTester::publishRobotTransform``` routine might fix the issue. Currently, the base transform header stamp is set only once whenever ```PlannerTester::updateRobotPosition``` is called and this transform is broadcasted at 10hz. This transform might expire by the time a listener actually looks it up. Updating the transform header stamp should do the trick, if I'm not wrong.

```
void TransformHandler::publishRobotTransform()
{
  base_transform_->header.stamp = now();
  tf_broadcaster_->sendTransform(*base_transform_);
}
```

SteveMacenski, 2020-05-08:
`base_transform_->header.stamp = now() + rclcpp::Duration(250000000);` adds 0.25 seconds to the latest timestamp so that there's a valid timestamp when in a few micro/milliseconds in the future when a TF buffer is getting a transform, it has a valid transform to work with and there's no extrapolation in to the future exception. Typically, a global localization system will slightly future date its transforms (e.g. robot localization with GPS or AMCL) so that it leads the system time a little so that any transformation has a valid tree at that time to work with. You could try removing it and see if it help / hurts, but I think it hurts.

> this transform is broadcasted at 10hz

So the latest message should be valid for 0.25 seconds with that future stamp, and if the messages are being published at 10hz (0.1 s) then there should be enough overlap between messages so there's always a valid message in the system. 

I encourage you to try this out. I've spent so much time on this a few months ago when I originally thought I'd fixed it that maybe it could use another set of eyes / strategies.

AlexeyMerzlyakov, 2020-05-08:
> Circle also will let you rerun a session and SSH into it to inspect things. If you go to the web interface for circle, you can select rerun with SSH and it'll give you a set of keys to access the machine to poke around.

Unfortunately, there is no rights to trigger a re-build or use SSH there, so I've chose local docker build:

> The release dockerfile might be something to look into as well for a simpler look at things: https://github.com/ros-planning/navigation2/blob/master/Dockerfile

Thank you a lot! I've started local Docker build to reproduce a problem. It looks like I've got the same as in #1627 bug. The issue is in progress state. Do you know - is there any update on there?

SteveMacenski, 2020-05-08:
You may want to see if you can reproduce using the eloquent docker image as the base then. @AlexeyMerzlyakov I gave you write access to the repo, once you accept I think that should give you the ability to play with circleCI. Let me know if not and I can elevate you more.

AlexeyMerzlyakov, 2020-05-13:
@SteveMacenski thank you for making me an access to CI to reproduce a timeout issue! It works well. I've created testing PR #1723 for the problem No.1 reproduction and tried to reproduce it in different ways. But now timeouts look like disappeared in `test_planner_costmaps` and `test_planner_random` tests. I may suppose, that was docker environment problem which was active before but somehow disappeared now. Anyway, we need to return to timeout problem as soon as it will appear again in the future.

However, while testing the problem in CI I've discovered another new Issue No.4 appeared only in release testing of `test_planner_costmaps` testcase. The problem appears with messages like:
```
[5.842s] 9: Test command: /usr/bin/python3 "-u" "/opt/ros/foxy/share/ament_cmake_test/cmake/run_test.py" "/opt/overlay_ws/build/nav2_system_tests/test_results/nav2_system_tests/test_planner_costmaps.xml" "--package-name" "nav2_system_tests" "--generate-result-on-success" "--env" "TEST_LAUNCH_DIR=" "TEST_EXECUTABLE=/opt/overlay_ws/build/nav2_system_tests/src/planning/test_planner_costmaps_node" "TEST_MAP=/opt/overlay_ws/src/navigation2/nav2_system_tests/maps/map.pgm" "--command" "/opt/overlay_ws/src/navigation2/nav2_system_tests/src/planning/test_planner_costmaps_launch.py"
...
[7.553s] 9: [test_planner_costmaps_node-1] [WARN] [1589375325.287162186] [PlannerTester]: Path has collision at (1.70, 1.70)
[7.553s] 9: [test_planner_costmaps_node-1] [INFO] [1589375325.287213246] [PlannerTester]:    point #0 with x: 1 y: 1
[7.554s] 9: [test_planner_costmaps_node-1]    point #1 with x: 1.7 y: 1.7
[7.554s] 9: [test_planner_costmaps_node-1]    point #2 with x: 2.05 y: 2.05
[7.555s] 9: [test_planner_costmaps_node-1]    point #3 with x: 2.4 y: 2.4
...
[7.560s] 9: [test_planner_costmaps_node-1] /opt/overlay_ws/src/navigation2/nav2_system_tests/src/planning/test_planner_costmaps_node.cpp:50: Failure
[7.560s] 9: [test_planner_costmaps_node-1] Expected equality of these values:
[7.560s] 9: [test_planner_costmaps_node-1]   true
[7.560s] 9: [test_planner_costmaps_node-1]   obj->defaultPlannerTest(result)
[7.560s] 9: [test_planner_costmaps_node-1]     Which is: false
...
[7.710s] 9: [test_planner_costmaps_node-1] [  FAILED  ] testSimpleCostmaps.testSimpleCostmaps (1283 ms)
[7.710s] 9: [test_planner_costmaps_node-1] [----------] 1 test from testSimpleCostmaps (1283 ms total)
```
The problem appears because of costmap is not being updated while changing between cases. Planner considering costmap from previous testcase and producing a path that not correct.
After some digging into a code I've found the part where it actually appears:
```
    unsigned char * costmap_ptr = costmap_ros_->getCostmap()->getCharMap();
    delete[] costmap_ptr;
    costmap_ptr = new unsigned char[prop.size_x * prop.size_y];
    std::copy(cm.data.begin(), cm.data.end(), costmap_ptr);
```
GCC v9 on Focal Fossa considers `costmap_ptr` to be unused and optimizes it out in CI release . Actually costmap_ptr will be a new pointer after deletion and there is no guarantee to point to the same memory area. Making it to be volatile seems to resolve the problem:
```
--- a/nav2_system_tests/src/planning/planner_tester.hpp
+++ b/nav2_system_tests/src/planning/planner_tester.hpp
@@ -67,7 +67,8 @@ public:
     costmap_ros_->getCostmap()->resizeMap(
       prop.size_x, prop.size_y,
       prop.resolution, prop.origin.position.x, prop.origin.position.x);
-    unsigned char * costmap_ptr = costmap_ros_->getCostmap()->getCharMap();
+    // Volatile prevents compiler from treating costmap_ptr as unused or changing its address
+    volatile unsigned char * costmap_ptr = costmap_ros_->getCostmap()->getCharMap();
     delete[] costmap_ptr;
     costmap_ptr = new unsigned char[prop.size_x * prop.size_y];
     std::copy(cm.data.begin(), cm.data.end(), costmap_ptr);
```
But I personally does not like this approach, but it works. Any thoughts on that?

After everything will be OK, I can fix remaining in PR #1723 flake8 problems and submit a new full-fledged PR for that. Then we will finally resolve the problems with nav2_syatem_tests.

SteveMacenski, 2020-05-13:
> But now timeouts look like disappeared in test_planner_costmaps and test_planner_random tests.

That's irritating - that's also how it started. Randomly after the Eloquent release. 

On the costmap pointer thing - think its up to you for what's best. That seems reasonable given its a test and not production code. Some hacks here and there in tests I think is fine as long as you comment them, which you did

closed 2020-05-14 by SteveMacenski