ros-navigation/navigation2 #1589: Website page for configuration guide and setup instructions

SteveMacenski, 2020-03-09:
Configuration guide
- [ ] all parameters listed and their descriptions and defaults
- [ ] configuration guide
- [ ] Namespacing for server maps

Setup (like ROS1 wiki)
- [ ] setup odom sources
- [ ] setup TF
- [ ] setup sensor sources

Marwan99, 2020-05-20:
Will start working on listing the params.

SteveMacenski, 2020-08-19:
@fmrico do you think its worth having a page for the marathon2 experiment on the website? If so, any ideas on what the structure / goal we're trying to leave readers with on that? I could throw something together quickly while I'm making the IROS presentation but not sure what the style of something like that should be.

jackpien, 2020-09-13:
@SteveMacenski - I'm carving out some cycles to address the "setup TF / REP103/105" item above.

Here's my plan:

1. I will add a page in the [navigation.ros.org repo](https://github.com/ros-planning/navigation.ros.org) -> Tutorials called "Setting up your robot using tf2"
1. While recycling most of ROS1 Nav1's [Setting up your robot using tf](http://wiki.ros.org/navigation/Tutorials/RobotSetup/TF), I'm going to "port" the TF tutorial to TF2 for ROS2. 
1. Add code and instructions to show and tell a demonstration using turtlesim.

Questions:

1. Does the plan above sound about right?
1. Despite this "issue" being in the navigation2 repo, there's really nothing to do in this repo right (since the doxygen html from the doc folder doesn't look like it generates anything meaningful)? All the pull requests should happen in the navigation.ros.org repo right?

SteveMacenski, 2020-09-14:
@jackpien you're a little late to the game, a team within Samsung Research has already drafted this work we're doing iterations on before submitting the PR publicly. They're covering TF, REP, and URDF items in these 2 pages. 

If you'd like to apply that same interest to another one of the pages, we can carve that out for you. Right now they're only working on the transforms / URDF pages. Other ones:
- URDF for simulation (adding plugins for sensors / drive)
- R_L setup / odom inputs
- Sensor feed setup / adding sensors to your robot  / costmap
- Lifecycle manager setup 
- Behavior tree setup

jackpien, 2020-09-14:
Thanks for update @SteveMacenski 

>     * R_L setup / odom inputs

Is that basically "porting" the [Nav1 Publishing Odometry Information over ROS](http://wiki.ros.org/navigation/Tutorials/RobotSetup/Odom) page? What else do you envision "R_L setup" to entail?

SteveMacenski, 2020-09-14:
Not really that tutorial because that just goes through the message type, which isn't really important. It really should go over the odometry inputs we accept (anything that makes an `odometry` message; VIO, VO, wheel, IMU, mocap, etc), how to set up to fuse in R_L and the expected frames. 

It might not hurt to introduce the message, but I don't see a point in the tutorial as it exists now.

jackpien, 2020-09-14:
> It really should go over the odometry inputs we accept (anything that makes an `odometry` message; VIO, VO, wheel, IMU, mocap, etc), how to set up to fuse in R_L and the expected frames.

Hi @SteveMacenski - how do I get myself up to speed on the above? 

* Looking at the [Nav1 Robot Setup diagram](http://wiki.ros.org/navigation/Tutorials/RobotSetup#Robot_Setup) are you suggesting we need a step to describe the left-side "odometry source" box that Nav1 amcl and the controllers use?

* A brief web search seems to suggest that ROS1 sensor fusion is done in the [robot_localization](https://github.com/cra-ros-pkg/robot_localization) which sits outside of the Nav1 stack. Is this still the case for Nav2? 

* If Nav2 does do sensor fusion from various odom sources (VIO, wheel, IMU, etc), can you point me to the code / node in Nav2 that does this? 

Thanks!

SteveMacenski, 2020-09-14:
> Is this still the case for Nav2?

Yes, R_L is not in the stack. It's its own repo.

> can you point me to the code / node in Nav2 that does this?

That is R_L.

fmrico, 2020-09-15:
Hi @SteveMacenski 

> @fmrico do you think its worth having a page for the marathon2 experiment on the website? If so, any ideas on what the structure / goal we're trying to leave readers with on that? I could throw [...]

We have the info in this repo: https://github.com/IntelligentRoboticsLabs/marathon_ros2
We can link this repo, or copy these instructions. Any option is ok for me.

SteveMacenski, 2021-01-22:
Thanks to @jd-deleon for taking leadership on this!

SteveMacenski, 2021-12-21:
https://navigation.ros.org/setup_guides/index.html

Done! Thanks to @jd-deleon and team! Thanks for taking the lead on this, it will continue to be helpful for users for years to come!

closed 2021-12-21 by SteveMacenski