ros-navigation/navigation2 #1590: Discussion: use waypoints to represent the high-level global path

orduno, 2020-03-10:
### Problem statement
We're considering [modifying the replanning logic](https://github.com/ros-planning/navigation2/issues/1570) to alleviate issues from constant replanning and update.

### Proposal
Use a sequence of waypoints (route) to represent the high-level global path. Deal with route replanning and next waypoint replanning separately.

### Considerations
- Topological map generation and a planner are needed to obtain the route, adding an additional layer on top of the current planner and controller.
- nav2 waypoint follower feature stops and orients robot at each pose
- ...

SteveMacenski, 2020-03-10:
> Use a sequence of waypoints (route) to represent the high-level global path

That's how a path is defined already. I don't understand

orduno, 2020-03-12:
I've updated the issue description.

SteveMacenski, 2020-03-12:
Can you define what you mean they that? Topographical navigation will mean different things to different people (planning vs perception in particular). 

I think a couple examples would help.

fmrico, 2020-03-12:
Yes, I agree with @SteveMacenski . IMHO, it is more related to [planning](https://www.semanticscholar.org/paper/Planning-Topological-Navigation-for-Complex-Indoor-Mart%C3%ADn-Gines/398748d68d530db5caa954840842016c9d1813a3), and I think this is an application of navigation, not part of general navigation software.

orduno, 2020-03-12:
[Here's](https://www.cpp.edu/~ftang/courses/CS521/notes/topological%20path%20planning.pdf) the interpretation I was considering.

SteveMacenski, 2020-03-12:
This seems related (or the same?) as this ticket https://github.com/ros-planning/navigation2/issues/1595. 

I think what you're describing by that is using a sparse graph for planning (which could be computed by a PRM) where the nodes may or may not represent an area or location of interest. 

Are you referencing the sparse graph of landmarks in your question or are you describing the ability to say "navigate to the kitchen" based on semantic labeling of information? I suppose I think of these are 2 different problems. A sparse graph representation makes it so you can only navigate through topological graph points to get to your semantic label of interest, while navigating to that semantic label of interest just implies knowledge of the topology of the space.

SteveMacenski, 2020-04-30:
@shrijitsingh99 related ticket

SteveMacenski, 2020-05-06:
I think we're getting good discussions on including this type of behavior in the other #1595 ticket. Closing in favor of it.

closed 2020-05-06 by SteveMacenski