ros-navigation/navigation2 #1597: Plan/Control with Dynamic Obstacles

SteveMacenski, 2020-03-16:
Right now, we work only with static data about the state of the world. We should be able to create a generic interface to handle detections to track them and have them included in the planning and control processes.

This could be done with explicit modeling (e.g. planner and controller step forward dt and we project the new time-variant pose of detections and uncertainty into C-space) or implicitly (e.g. we have some object we know is dynamic and we "smear" it in the costmap projecting its pose and uncertainty into the current costmap without being time-dependent)

abylikhsanov, 2020-11-14:
Is this still in development? If so, is the idea to first implement a dynamic obstacle detection (for example as a separate plugin) and then use in any local planner?

SteveMacenski, 2020-11-15:
Its in development now. 

Integrations for dynamic obstacle data into DWB would be an appreciated contribution if you're interested. Basically it would involve creating a dynamic obstacle critic plugin to take in some set of dynamic tracks and using them and the change in time of the trajectory to score the paths.

abylikhsanov, 2020-11-15:
I am actively using TEB local planner and I would like to contribute for that instead if this is still interesting

abylikhsanov, 2020-11-15:
I believe it will not really matter which planner to choose after the first part of dynamic obstacles detection will be acheived as both TEB and DWB can use that information to change their trajectory accordingly. 

I also found this repo: https://github.com/ros-planning/navigation2_dynamic

Any reason why the above repo cannot be used? For me it looks like a legit point to start.

SteveMacenski, 2020-11-16:
In what way?

chiragmakwana0296, 2025-02-01:
I would like to work on adding dynamic obstacle tracking and avoidance to Navigation2. My approach involves:

    Adding the dynamic costmap layer to represent velocity vectors of obstacles, 
    Clearing dynamic obstacles from static layer to prevent global planner to re-plan through dynamic obstacles(if required).
    Smearing dynamic obstacles in the dynamic costmap layer using motion prediction.
    Integrating a tracking framework to estimate future poses of obstacles.
    Creating a dynamic obstacle critic plugin for DWB.

Before proceeding, I’d like to understand the current progress and expected architecture for this feature, if already discussed. Let me know how I can align my contribution with the project's direction.

Looking forward to your thoughts!

SteveMacenski, 2025-02-04:
I think this is worth pulling @alexanderjyuen into. He talked this year about his Radar Layer https://github.com/polymathrobotics/radar_layer work for Nav2 which I know might need some performance work / reworking based on the https://github.com/ros-navigation/navigation2_dynamic stack. It also needs functional validation that this works well enough in practical situations which might involve some reworks or hyperparameter tuning on the decay/smearing functions.  

There's also been a contributor that expressed interest in this in the Nav2 slack (that I'm going to link this comment to).

I think it would be worth getting on a call and all chatting synchronously about this so we can get started and collaborate more efficiently -- or at least start a new Nav2 Slack channel on this topic so we can chat in detail asynchronously. 

Something for instance I know you could work on that's greenfield here is a utility class to take in dynamic [ObstacleArrays](https://github.com/ros-navigation/navigation2_dynamic/blob/master/nav2_dynamic_msgs/msg/ObstacleArray.msg) and a timestamp to find the position of obstacles at some time `t` based on the data. That would be crucial for use in the DWB critics and also MPPI critics (and easily unit testable before integrating into the critics to store at various times). That utility and the controller critics are something you could definitely work on while we wait on Alex to get back from his vacation to chat about the costmap layer portion of things. 

Do you have a platform / way of testing this for functional evaluation?

alexanderjyuen, 2025-02-04:
@chiragmakwana0296, I would be more than happy to support work on this project. At present, we have implemented the 3 of the five things you listed below.
 
> * Adding the dynamic costmap layer to represent velocity vectors of obstacles,
> * Smearing dynamic obstacles in the dynamic costmap layer using motion prediction.
> * Integrating a tracking framework to estimate future poses of obstacles.

As @SteveMacenski mentioned there are a few things we need to take care of in order to push this project forward, to reiterate what @SteveMacenski said I think the tasks included, but is not limited to:
- Functional validation of radar_layer and dynamic obstacle tracking, we still haven't really stressed tested this in a dynamic situation, with a lot of the limitation really coming from the dynamic obstacle detection method
- Performance improvements for the dynamic obstacle layer

Let's arrange a call or Slack group chat to discuss this

chiragmakwana0296, 2025-02-05:
@SteveMacenski Sounds great! I’m currently reviewing the implementation of radar_layer and [navigation2_dynamic](https://github.com/ros-navigation/navigation2_dynamic)  to gain a deeper understanding.

While we coordinate further discussions, I’ll begin development of the utility class in nav2_util pkg 
 for getting obstacle positions at a given timestamp based on ObstacleArray data. 
For this are we using any particular motion model to generate future trajectory of obstacle or for instance we can assume constant velocity model?

In order to do functional testing I do have few options:
- Carla Sim with Detectors and radar sensor. (This setup is already running on my system), this have obstacle detection module which publishes detected and tracked objects array([derived_object_msgs/ObjectArray](https://docs.ros.org/en/melodic/api/derived_object_msgs/html/msg/ObjectArray.html)) and radar publishing tracked PointCloud2. more details given in https://carla.readthedocs.io/projects/ros-bridge/en/latest/ros_sensors/
- Another option is to use open source dataset like KITTI/Waymo and using tracking pipeline like https://github.com/megvii-research/MCTrack.
- I can write a node to publish mock tracked objects msg of type  [ObstacleArray.msg](https://github.com/ros-navigation/navigation2_dynamic/blob/master/nav2_dynamic_msgs/msg/ObstacleArray.msg)
- let me know if you know better way to do function testing and evaluation @SteveMacenski @alexanderjyuen  

@alexanderjyuen i do have few queries on the task you listed, it would be great if we can move to slack/call for further discussion.
cc @SteveMacenski  
> - Functional validation of radar_layer and dynamic obstacle tracking, we still haven't really stressed tested this in a dynamic situation, with a lot of the limitation really coming from the dynamic obstacle detection method
> - Performance improvements for the dynamic obstacle layer w.r.t how fast it performs the necessary smearing calculations

SteveMacenski, 2025-02-06:
> While we coordinate further discussions, I’ll begin development of the utility class in nav2_util pkg for getting obstacle positions at a given timestamp based on ObstacleArray data.

I'd put it in `navigation2_dynamics` as a new `nav2_dynamics_utils` package. We can always include it back into `nav2_util` once that repository is merged into the main of `navigation2`, but its nice to keep this all grouped together while its being worked on in its own sandbox that we can change as needed without API stability guarantees (+ faster iteration cycles having everything in sync intrinsically in the same repo).

>  For this are we using any particular motion model to generate future trajectory of obstacle or for instance we can assume constant velocity model?

I think it might be good to make a motion model interface, but by default constant velocity out of the box is very reasonable!

> This setup is already running on my system

!! This is very beneficial. Should we consolidate https://github.com/ros-navigation/navigation2_dynamic/tree/master/nav2_dynamic_msgs with some of the benefits from [derived_object_msgs/ObjectArray](https://docs.ros.org/en/melodic/api/derived_object_msgs/html/msg/ObjectArray.html))? If there's anything key we're missing, we should consider adding it. The `nav2_dynamic_msgs` will eventually be merged into `nav2_msgs`. If you're already using Carla and have that all set up for your applications, I figure that's the path of least resistance and you can test on your application-space / requirements most naturally. 

> I can write a node to publish mock tracked objects msg of type [ObstacleArray.msg](https://github.com/ros-navigation/navigation2_dynamic/blob/master/nav2_dynamic_msgs/msg/ObstacleArray.msg)

Probably wouldn't hurt. It would be good to have that regardless for some integration testing of the utils. That mock node itself would be a good util! 

---

I'm meeting with Alex tomorrow, let me sync with him on when would be good for a call. I would really like it to be early next week since the week after that I will be out of the office. Tentatively Tuesday night in PST (Wednesday morning in Pune). 

It might not hurt to play with the costmap layer if you have time. I know there is some left to be desired with the current state of it w.r.t. performance and it may be worth working with Alex to rework it into something better performance. Luckily, this is a rich topic with plenty of resources of other dynamic obstacle layers, papers taking about proximity layers with velocity scaling in direction of cost, and so forth.

chiragmakwana0296, 2025-02-08:
> I'd put it in `navigation2_dynamics` as a new `nav2_dynamics_utils` package. We can always include it back into `nav2_util` once that repository is merged into the main of `navigation2`, but its [...]

makes sense, I ll follow this. 

> !! This is very beneficial. Should we consolidate https://github.com/ros-navigation/navigation2_dynamic/tree/master/nav2_dynamic_msgs with some of the benefits from [...]

Yeah, We will need shape and size of the object which is crucial info for the controller. So i am adding it to [ObstacleArray.msg](https://github.com/ros-navigation/navigation2_dynamic/blob/master/nav2_dynamic_msgs/msg/ObstacleArray.msg). I ll raise PR by tomorrow, so we can discuss further on this part.