ros-navigation/navigation2 #1600: Problem with BehaviorTreeEngine reset and halt

facontidavide, 2020-03-18:
This issue is related to the ongoing discussion in #1439 

I noticed that the logic of the methods [resetTree](https://github.com/ros-planning/navigation2/blob/master/nav2_behavior_tree/include/nav2_behavior_tree/behavior_tree_engine.hpp#L58)
and [haltAllActions](https://github.com/ros-planning/navigation2/blob/master/nav2_behavior_tree/include/nav2_behavior_tree/behavior_tree_engine.hpp#L47) is problematic.

A halted node **must** set the status to IDLE and an idled node must always call halt(9 first to do a correct cleanup of RUNNING asynchronous nodes.

A work around, for the time being, would be to have a **single** method like this:

``` C++
    static void haltTree(BT::TreeNode * root_node)
    {
        // the halt should propagate to all the node if the nodes
        // have been implemented correctly
        root_node->halt();
        root_node->setStatus(NodeStatus::IDLE);

        //but, just in case.... 
        auto visitor = [](BT::TreeNode * node) {
            node->halt();
            node->setStatus(BT::NodeStatus::IDLE);
        };
        BT::applyRecursiveVisitor(root_node, visitor);
    }
```

In practice, this same problem might affect many users, therefore I made a breaking change in my API: 
https://github.com/BehaviorTree/BehaviorTree.CPP/commit/4f30e3a42ff0c657eb5711d7ae94318fde93bd60

I know people will hate me, but it is the only way to forbid users from repeating this same error.

What do you think?

SteveMacenski, 2020-03-19:
Do you think that would solve the async issues on reset? @jconn is this something you'd be willing to test for us? It looks like a 5 line change to get it going. 

@facontidavide that sounds reasonable to me if it fixes the issue. Are you certain of that? We're approaching a V1 release and I wouldn't want to revert to include this if there's any doubt of it fixing the problem.

facontidavide, 2020-03-19:
No, I am not certain need to be tested

facontidavide, 2020-03-19:
Something else that we should urgently explore together is why you are using CoroAction.

I will probably have a look to decide if we don't have a better approach during the weekend.

Jconn, 2020-03-19:
I will start testing this locally in the next few days

facontidavide, 2020-03-19:
I will try to work during the weekend on a refactoring that removes CoroAction altogether...

SteveMacenski, 2020-03-19:
@facontidavide I wasn't part of the original work to wrap BT.CPP into navigation use, so I can't speak to the specifics of why that was chosen. Can other action node-types be "yielded" for long running  tick processes? 

Removing it seems to be a pretty nuclear solution if there's not actually a problem with it. It seems like you just need to make sure halted also idles, which  that commit you linked to does.

facontidavide, 2020-03-19:
The big problem with CoroAction is that you are apparently (for what I understood, but I might be wrong) halting from a  different thread.

I am not sure that is a possible. I definitively need to explore the source code and understand better what is going on.

Jconn, 2020-03-19:
@facontidavide we were passing our Behavior tree object and CoroActionNodes between multiple threads. Thread A would halt the CoroAction node, and Thread B would start ticking that node again.

These threads are being spawned by the NavigateToPose action server. Every time a request was made, the action server would spawn a new thread to handle that request. That new thread would be passed the Behavior tree object.

The CoroAction has a [state variable](https://github.com/BehaviorTree/BehaviorTree.CPP/blob/ros2-devel/3rdparty/coroutine/coroutine.h#L275) local to the thread it is being run on, so when we halt from thread A and tick from thread B we created problems.

[Current fix ](https://github.com/ros-planning/navigation2/pull/1515/files)is to recreate the whole tree for every Thread.

facontidavide, 2020-03-19:
This thingy with multiple threads seems to be suspicious to me (code smell), but I guess you have good reasons that I don’t understand.

Why are you halting the node, instead of letting it conclude it’s work?

In any case, If you have good reasons to do that, you must **not** use CoroAction

Jconn, 2020-03-19:
> Why are you halting the node, instead of letting it conclude it’s work?

The Node is halted if a request is cancelled. If a request is allowed to be completed, the Node will tick until it concludes its work. 

I tried to implement a request cancel that would halt nodes, and then tick the node again to try and clean up state, but hit a snag somewhere.

facontidavide, 2020-03-19:
Let me think a good solution for this. I am sure I am are missing something...

facontidavide, 2020-03-19:
@SteveMacenski and @Jconn 
I am not very familiar with ROS2 and Navigation2 (shame of me).
I can look at the source code from the point of view of the BT.CPP author, but it would be nice if someone can assist me with some guidance about how to reproduce the error. 

I have absolutely no idea where to start. Any ready to use turtlebot simulator with Gazebo?
Can any of you spend time with me to finally fix this?

SteveMacenski, 2020-03-19:
>  Thread A would halt the CoroAction node, and Thread B would start ticking that node again.

> This thingy with multiple threads seems to be suspicious to me (code smell), but I guess you have good reasons that I don’t understand.

To close the loop: the ROS action server will call the action, then the callback will create a worker thread. It'll do some stuff, then return when complete. If a BT is reset and then called again, its being processed by a new worker thread from the action server. The BT could be called by N arbitrary threads, but synchronously (e.g. only 1 thread calling at BT at a time).

He's pointing out that there's some variable that attaches to the original thread, which is now 2 worker threads gone away.

Indeed! I've worked hard to populate this website: https://ros-planning.github.io/navigation2/getting_started/index.html

facontidavide, 2020-03-20:
>> The BT could be called by N arbitrary threads, but synchronously (e.g. only 1 thread calling at BT at a time).

This is good thing. If we really need (but probably we don't if we just get rid of CoroAction), it means that we might have a single thread (always the same), responsible for running the BT, trigger by the spawned ones.

But that is not needed probably. 
I have a long weekend in front of me... hopefully I am able to send some interesting PR on Monday ;)

SteveMacenski, 2020-03-20:
> This is good thing. If we really need (but probably we don't if we just get rid of CoroAction), it means that we might have a single thread (always the same), responsible for running the BT, trigger by the spawned ones.

That sounds very not-awesome, because then the triggering thread would just be doing nothing and completely blocked for a result. There may be a fancy C++17 way of doing this, but that may be overkill.

Is there a reason that thread can't be updated on a new call? It's not intuitive to me at all that a coroutine cannot be called by different threads synchronously. 

Can you explain the benefits of the other action types? Your tutorial here: https://www.behaviortree.dev/tutorial_09_coroutines/ lays out a pretty strong argument for coroutines

facontidavide, 2020-03-20:
>> That sounds very not-awesome, because then the triggering thread would just be doing nothing and completely blocked for a result.

From your point of view, you won't notice the difference, not even in terms of latency, but that is not the point.

>> Is there a reason that thread can't be updated on a new call? It's not intuitive to me at all that a coroutine cannot be called by different threads synchronously. 

Precisely because of the fact that it is not untuitive, **I was not aware of this problem myself** :(

I **believed** that coroutines were a good abstraction, but your problems revealed that the coroutine library  I am using (not the concept of coroutines themselves) has serious limitations.

I will give a try to boost::coroutine2, but  I think the marginal benefit are not worth it. 
I will reflect this new recomendations in the documentation of my next release.

SteveMacenski, 2020-03-20:
OK, keep us in the loop. It sounds like my idea of having a pointer swap of BT objects (problem in #1516) would cause issues since the threads are different to help reduce latency. 

For your initial ticket, can you submit a PR with the updates to halting / resetting to be in compliance with your library?

closed 2020-03-27 by SteveMacenski