ros-navigation/navigation2 #1603: CollisionChecker detects a collision when robot footprint hits inscribed radious distance.

gimait, 2020-03-20:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Version or commit hash:
  - master
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue

Place the robot on an area marked as inscribed inflated obstacle.
Trigger spin recovery.

#### Expected behavior
The ```isCollisionFree``` function should return false when there is a collision for a given pose

#### Actual behavior
Currently, the function ```CollisionChecker::isCollisionFree``` returns false when the footprint of the robot enters an area marked as ```INSCRIBED_INFLATED_OBSTACLE```.
This means that the robot would detect a collision in positions like:
[image]
[image]
#### Additional information
This issue was introduced by the changes in #1404.

----
## Feature request

#### Feature description

#### Implementation considerations

SteveMacenski, 2020-03-20:
Try changing the `>= to LETHAL` -- I think that should square you up

gimait, 2020-03-20:
I think the issue comes actually from the costmap, after looking a bit more closely, it looks like the cells inside the robot footprint are marked as inscribed, and then we don't get any collision:
[image]

After checking the code again it looks like this was the original issue. Although that if check was a bit deceptive. I will see if I can find a solution from the costmap side.

SteveMacenski, 2020-03-20:
There's a 'clear footprint' option in the costmap where we clear cells that are under the footprint to allow it to navigate if there's a bad costmap value somewhere. I think that's what you're seeing. 

I'm not sure off hand the best way to handle this situation. I'd also look to see how this is dealt with in ROS1 navigation with the spin recovery with that same thing on.

gimait, 2020-03-20:
Yes, footprint clearing seems to be the source of the problem. However, I do not see any differences between this and the ROS1 version, so I am wondering wether this has always been an issue.

I don't really know what is the best option here, I will need to investigate more how to deal with this situation. We could make the clearing function a bit smarter on some way, to make account of the surroundings (maybe keep obstacles connected to others outside the footprint?) or just set the footprint clearing default to false?

SteveMacenski, 2020-03-20:
I agree from looking at the code, I think this was in issue in ROS1 as well. I don't have an answer to the best course of action right now. The footprint clearing also helps if you happen to already be very close to an obstacle and you're trying to navigate away from it by allowing permissible paths and control efforts to be generated.

By the way, I looked up the `> 0` t hat was there before: https://github.com/ros-planning/navigation/blob/c33d8b19c04d6c8bc156be643a1ef7de346fb4a4/base_local_planner/src/costmap_model.cpp#L47, its because the old Nav1 code encoded information differently for the world model. 

Lets start with changing the code to be correct with `>= LETHAL` so that we're not triggering issues in free space. I could live with having the default be `false` for `footprint_clearing_enabled` for both obstacle and voxel layers -- but that's not really a solution in case someone does flip that switch.

gimait, 2020-03-21:
I just created a pr to change the check to lethal. 
And I just realised, could it be that the turtlebot3 footprint is not properly defined? If the footprint should match the shape of the robot, this might be the reason why we are seeing the problem in this simulation.
I was also thinking that another solution could be keeping the obstacles under the border of the footprint? That would probably be enough to trigger the detection, and would still clear the map from wrong measurements.

gimait, 2020-03-23:
Seems like the issue appears when using the radious of the robot as footprint.
With the current implementation, if the robot footprint is not provided or is wrong and the clear footprint option is enabled, the objects near the robot won't be visible in the map. This is a problem mostly for the spin recovery, other actions can still be performed since the robot wouldn't appear as in collision.
I believe the smartest choice here would be disabling the clear footprint option if no footprint is provided or if it is invalid. This would potentially block the robot in some situations, but at least it won't hit any obstacles appearing in its surroundings.

SteveMacenski, 2020-03-23:
> could it be that the turtlebot3 footprint is not properly defined?

What makes you think that? I'm looking at the green circle in rviz above and it seems fine to me. Can you clarify?

>  If the footprint should match the shape of the robot, this might be the reason why we are seeing the problem in this simulation.

It doesn't have to, circles are easy to check and can be used. The footprint doesn't need to be exact. In general, I add a 5% engineering margin to my footprints. 

> Seems like the issue appears when using the radious of the robot as footprint.

I think you'd see this same issue if the footprint was a set of coordinates, that also were a circle, or a box, or any convex shape. Since the footprint points are going to be cleared with that setting activated. Can you clarify why a circle may be special?

Providing a footprint is strictly a requirement of the navigation stack.

gimait, 2020-03-23:
I believe we might have more than one issue here, that in combination are causing this problem:

First, when it comes to the collision checking itself:
If we use a footprint that does not describe the robot properly, we can hide obstacles in the scene. This is usually not a problem in a static environment where the obstacles are well defined, since the planners will always keep a distance to those obstacles. However, if a new obstacle comes into scene (maybe something fell in front of the robot) and gets within the footprint area, we will automatically hide it.

This is something that cannot be solved on another way than defining a footprint that is appropiate for your robot and then having in mind that things like that can happen and we can't fix them. That's why it is not such a big issue. However, when we put together this issue with a circular footprint and the spin recovery, we have the issue we are seeing in #1404.

In the case of the spin recovery, when the robot rotates around the center of its circular footprint, the calculation of the future footprint is useless. However this happens only in this case, some examples:
[image]

I believe this is quite a corner case as it only affects the spin recovery and mainly when using the default circular shape, and that is probably why it is still an issue. What do you think?
I tried changing the clear footprint to keep the obstacles that touch the footprint, but I don't think that is the best solution. That would basically block the robot from moving if we get into a state similar to the one in the figure: All planners would complain that there is no way you can get out of there without hitting an obstacle since the footprint is already hitting one in the initial position.
I suggested to disable the clear footprint option if no footprint is provided because I couldn't think of a better solution.

SteveMacenski, 2020-03-24:
>  I suggested to disable the clear footprint option if no footprint is provided because I couldn't think of a better solution.

There is no situation where no footprint is provided. This is a strict requirement of the stack to have a configured footprint. It is permissible to make your footprint larger than the actual robot base amount for a multitude of reasons:
- Engineering margin
- Manufacturing tolerances
- Robot is circular-ish so you'd like to use the speed ups for defining a circular footprint 
- etc.

The only problem I see is how to rectify the costmap for the collision checker with the clearing footprint parameter on. Perhaps by representing the costmap clearing footprint differently or having the collision checker receive a more "raw" version. 

Regardless of of how the shape is defined (circle, rectangle, triangle, etc) anything using that collision checker is going to have an issue with the clear footprint enabled.

gimait, 2020-03-24:
When I said no footprint is provided I was looking at this:
https://github.com/ros-planning/navigation2/blob/efc42bd161bb3672ed273f3f4142fc35d88bd190/nav2_costmap_2d/src/costmap_2d_ros.cpp#L329-L344
and I was thinking on doing something with the use_radius_ variable, but I understand that avoiding the clear footprint option might not be the best solution .

> The only problem I see is how to rectify the costmap for the collision checker with the clearing footprint parameter on. Perhaps by representing the costmap clearing footprint differently or having the collision checker receive a more "raw" version.

I don't really know.. the footprint is cleared in the obstacle layer plugin, so there is no more "raw" version to look at, maybe we would need to look at the sensor data directly then.

We could look into some way of deciding when an obstacle is within the footprint or not, but I think anything we add there would be too expensive to be worth it (since ```setConvexPolygonCost``` is called on each costmap calculation cycle).

Something we could do instead would be to clear all cells within a footprint with exception of those touching the footprint itself (so keep the cells on the perimeter of the footprint). This can solve most of the cases, however we would need to look into what would happen to everything using the costmaps to calculate collisions, since it would change a quite basic assumption, blocking the robot movement in multiple cases. 

>That would basically block the robot from moving if we get into a state similar to the one in the figure: All planners would complain that there is no way you can get out of there without hitting an obstacle since the footprint is already hitting one in the initial position.

This might still be a desired behaviour though.

SteveMacenski, 2020-03-24:
Lets be super clear -- the footprint, regardless if its circular, from the radius, or defined by a set of points, will be made into a set of points. These points are then being cleared out by the clear under footprint parameter in the costmap updates for the obstacle and voxel layers. I dont think you should or need to focus on anything regarding the footprint representation itself. Any and all shapes will contain the exact same issue that the collision checker will fail to detect a current collision due to the clearing in its current footprint.

The problem statement we need to resolve is how we can make it so that the collision checker can be made valid when the clearing parameter is enabled. Removing the parameter is not an option. Nor is just making it default to false and leaving a landmine for someone to hit. 

I do wonder, however if you're not in a collision state, and if the backup, spin, or any other predictive behavior is triggered entering you into a collision state, that would work fine. Because the costmap isnt being updated predictively as the collision checker costmap copy is checking it. Therefore, to some degree, this seems to be a "chicken and the egg" problem. You won't get into collisions if you haven't cleared the footprint, but once you are in a collision, you have a cleared footprint. If you don't then clear that footprint, you'll never be able to move again. 

The world isn't a friction-less-vacuum unfortunately, so that rotation may not be perfect or the localization may shift things and put it back into a collision state. For your example 1 of the square, that could be detected with a denser (not just 4 corners) footprint point description. For example 2 that's more of a localization issue but I suppose in theory the robot could actually move itself over if in collision with something outside its kinematics model so that seems valid (and with enough recovery actions, it'll eventually get out). Example 3 is more of an issue since its not yet "in" collision, yet in running the recovery behavior, it'll actually now be in one. However, the system prior thought it was in collision, so then actually colliding is suboptimal but expected behavior since accordingly you thought you already had. I haven't seen folks define "incorrect" footprints like that before in example 3 - its usually additional buffer around the base or a little larger circle if you robot is slightly elongated but rotating in center so the composite of a 360 rotation would sweep out that radius. In those 2 cases, I think situations 1 and 2 hold.

As such, I wonder if this is a problem at all. Once in collision, you need to find a way out, and spin may not be the right way for some cases (example 1) so you want your collision checker to reject them. Rather than the backup recovery should be used if valid (which it would be if backing straight out). 

Thoughts?

gimait, 2020-03-24:
> Any and all shapes will contain the exact same issue that the collision checker will fail to detect a current collision due to the clearing in its current footprint.

I completely agree with you there. However, all collision checks will be done on future positions of the robot, and almost any robot movement towards the object would potentially be prevented by the planners. That's why I think is a corner case happening only with the spin at the moment, and only with a specific setting.

> As such, I wonder if this is a problem at all.

I am also not sure wether this is something we should fix. 
At the end there is one use case that would see this issue as a problem, please correct me if I'm wrong:
1. The robot is collision free.
2. An obstacle appears in scene, close enough to enter the footprint area.
3. We trigger a spin, but since we do not have a full picture of the colliding object, the robot hits it.

What is the expected behaviour in this case? Are we supposed to be able to prevent this collision?

I can think on two simple solutions that could allow keeping the clear footprint option but with a slightly different definition:

- We could keep the cells that touch the footprint if there is an object detected. 
The change would be from this:
[image]
To this:
[image]
This change would fix the issue with the spin. Since we can see there is a colliding object, the spin can stop. Then another behaviour can be used to try to exit that position. However, we would need to look into what happens then, because the new behaviour might still see a collision in future positions that are close to the robot (in the first iteration, maybe), and fail to find a trajectory getting the robot out of there.

- Another option would be fixing this in the spin, maybe expanding the area of interest for the collision check ( more or less like checking a bigger footprint).
This would only affect the behaviour of the spin recovery, preventing collisions in this corner cases without affecting other behaviours.
I guess this would be a better option since it won't affect other modules.

What should we do about it?

SteveMacenski, 2020-03-25:
I think we should leave as is. It seems to me to be working as intended -- adding the inflated cost would stop planning from occurring and result in an infinite loop. 

```
1. The robot is collision free.
2.  An obstacle appears in scene, close enough to enter the footprint area.
3.  We trigger a spin, but since we do not have a full picture of the colliding object, the robot hits it.
```

An object that would appear would likely be of width of > 1 pixel. So any detected stuff behind it at any angle would give you more information to not drive forward. For the issue of spinning, that isnt an issue with circular robots since it can spin in place without issue, for your example a square or other shape would have the corners move out and hit that other non-1-pixel part of the obstacle as part of the collision checking. 

I'm not sure there's anything here to fix -- I think the logic is sound but worth of some notes in documentation

gimait, 2020-03-25:
> For the issue of spinning, that isnt an issue with circular robots since it can spin in place without issue, for your example a square or other shape would have the corners move out and hit that other non-1-pixel part of the obstacle as part of the collision checking.

I agree, this is not a problem for circular robots, or robots with other shapes, if their footprint matches the robot shape (plus a safety width). This issue can only happen when you have a non circular robot with a circular footprint, or in any case where the footprint of the robot does not describe correctly the movement of the robot in a spin. This was the source of the issue #1404.

Maybe we can close this issue and see if someone finds it to be a problem. If it is an actual issue people will find it sooner or later and then we will have more information about the changes needed.

SteveMacenski, 2020-03-25:
I'm not sure there's an issue to be solved - if someone uses the wrong footprint they're going to get weird behavior. I don't necessarily think that using a circular footprint and having the behavior we talk about is a problem because it will still be collision free. If you have a square robot with a circular footprint with the rotation axel in the center, you're still not going to hit anything if the footprint extends to the diagonal length. It might just be more trigger happy. 

With that -- I think we  can close this

closed 2020-03-25 by SteveMacenski