ros-navigation/navigation2 #1604: Investigate if there's a problem clearing free space with track unknown space is true

SteveMacenski, 2020-03-20:
I noticed in testing some motion planners that when I set track unknown space to true or false, the background default value will change, but it doesn't appear that the measurements from the laser and depth camera "free" unknown space. In ROS1, you'd see this by rays casted from the robot the measurement being marked "Free" rather than "Unknown", which is especially useful in SLAM and navigating in environments that are partially mapped and not going off-map. 

We need to investigate whether the clearing free space functionality is working properly. If its not, fixing it, and if it is, figuring out why its not properly visualizing in RVIZ.

Marwan99, 2020-05-04:
I will try to explore this.

Marwan99, 2020-05-06:
Did a couple of tests by manually moving the robot while mapping its surrounding with track_unknown_space set to true. Unknown spaces seemed to change to Free/Occupied after getting new sensor data as expected (see example below). As far as I can tell, clearing free space functionality seems to be working fine. Tests have been done on both static and rolling window costmaps.

@SteveMacenski Any pointers on how re-create the errors you have seen? Do you have any particular tests in mind?

[image]
*Robot at starting pose with Unknown cells(grey)*
[image]
*Robot moved forward, previously Unknown space now marked as Free (white)*

SteveMacenski, 2020-05-06:
OK, I think it works then. That looks exactly right. I don't know why I was having issues then trying to do this myself....

Awesome, well thanks for letting me know I'm slowly loosing my sanity :rofl:

closed 2020-05-06 by SteveMacenski