ros-navigation/navigation2 #1611: CMake error during building navigation2 for Crystal

kurshakuz, 2020-03-25:
## Bug report

**Required Info:**

- Operating System:
  -  Ubuntu 18.04

#### Steps to reproduce issue

``` 
mkdir -p ~/navigation2_ws/src
cd ~/navigation2_ws/src
git clone https://github.com/ros-planning/navigation2.git --branch crystal-devel
cd ~/navigation2_ws
rosdep install -y -r -q --from-paths src --ignore-src --rosdistro crystal
colcon build --symlink-install

``` 

#### Expected behavior
Successful build
#### Actual behavior
```
--- stderr: nav2_tasks                                                   
CMake Error at CMakeLists.txt:11 (find_package):
  By not providing "Findbehaviortree_cpp.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "behaviortree_cpp", but CMake did not find one.

  Could not find a package configuration file provided by "behaviortree_cpp"
  with any of the following names:

    behaviortree_cppConfig.cmake
    behaviortree_cpp-config.cmake

  Add the installation prefix of "behaviortree_cpp" to CMAKE_PREFIX_PATH or
  set "behaviortree_cpp_DIR" to a directory containing one of the above
  files.  If "behaviortree_cpp" provides a separate development package or
  SDK, be sure it has been installed.

---
Failed   <<< nav2_tasks	[ Exited with code 1 ]
Aborted  <<< nav2_util                           
Aborted  <<< dwb_msgs                             
Aborted  <<< nav2_map_server                                          

Summary: 6 packages finished [11.9s]
  1 package failed: nav2_tasks
  3 packages aborted: dwb_msgs nav2_map_server nav2_util
  1 package had stderr output: nav2_tasks
  17 packages not processed

```
#### Additional information

shivaang12, 2020-03-25:
Navigation2 depends on several packages. This error points to package called behavior tree which you might not have it prior to building Navigation 2. Behavior tree isn't the only dependency. Here is the link to Navigation 2 dependencies

https://github.com/ros-planning/navigation2/blob/master/tools/ros2_dependencies.repos

Make sure first build the dependencies and source them prior to building Nav 2.

SteveMacenski, 2020-03-25:
@shivaang12 is correct. Additionally, crystal is EOL and will not receive any new development or support

kurshakuz, 2020-03-25:
Got it, thanks! I will use Dashing then as it is supported by Gazebo

closed 2020-03-25 by kurshakuz