ros-navigation/navigation2 #1613: Reimplement passing of action_name to BtActionNode

atb033, 2020-03-27:
## Feature request

#### Feature description
- Currently, the `action_server_name` is passed as an argument to the constructor, and the action_client is initialized in the constructor.
- In this [implementation](https://github.com/ros-planning/navigation2/blob/2214a04a61b1dadc2f2518b29a2237bb3c6f190c/nav2_behavior_tree/plugins/action/compute_path_to_pose_action.cpp#L38-L41) the action client is reinitialized with a `remapped_action_name`. However, this wouldn't work unless the default action server is available. Otherwise, the code will get blocked at the constructor of `BtActionNode`
#### Implementation considerations
- Ideally, I would like it to be implemented just as how `service_name` is passed in `BtServiceNode`
- However, this could change the signature of `BtActionNode`, and could lead to errors to the ones already using it.

SteveMacenski, 2020-03-27:
A PR would be appreciated. One reason we didnt do that is we didnt want to force users to define the default server address if they didnt have multiple. If you can find a way around that, very open to that. 

I think potentially if you add the default address in the constructor being registered, it can be set as the input default and give you that behavior

SteveMacenski, 2020-03-27:
The goal here was so that be "default" it goes to the default server name, and then if provided it uses the provided server address. 

The mechanism in which you think is most suitable to do that, I'm ok with. I'm not particularly picky on this one. I think there's probably a half dozen ways it could be done.

atb033, 2020-03-27:
Sure. I will do a PR soon.

closed 2020-03-29 by atb033