ros-navigation/navigation2 #1617: Investigate RADAR support and where it may fit in

SteveMacenski, 2020-03-29:
Radars are getting cheaper and cheaper, they make for excellent sensors for static mounted applications for detecting and tracking moving obstacles. Removes the need for ML for detecting things of interest in the scene and give ranging / velocity profiles of blobs.

Investigate if this works well (and computationally efficiently) on mobile robots as autonomous cars. Integrate into the navigation stack after identifying potential straights and weaknesses. In particular, I think this will be valuable for a radar dynamic obstacle tracking costmap-ish layer, or correlating information from visual or lidar detections, or filter tracking/behavior prediction.

SteveMacenski, 2020-05-29:
Additionally, cover the general dynamic obstacle case, RADAR support is just an example of one particular source of dynamic obstacle information. Complex YOLO and other machine learning techniques can also provide positions and velocities of dynamic obstacles for use. 

This ticket covers how to use dynamic obstacle information in the navigation stack (e.g. how to integrate into costmaps, how to integrate into planners and controllers)

elladuffy17, 2023-06-13:
Hi there, I need help configuring a radar so it can be used (alongside a lidar) in the collision monitor node. How might I properly go about doing this? I have tried adding to the turtlebot_waffle.urdf within nav2_bringup, but haven't had any luck being able to see the sensor simulated in gazebo when I run the getting started example. l asked a similar question at the follow site: https://answers.ros.org/question/416414/navigation2-collision-monitor-with-multiple-data-sources/  Any help is very very appreciated!! Thank you.

SteveMacenski, 2026-09-02:
I would love to continue to explore this, but so far there doesn't seem to be alot of interest in radar in AMR/Nav2 users. I know a few that do use it, but use the radar's discrete tracked returns directly or with minor filtering, so we could treat this no differently than any other dynamic obstacle detection method (https://github.com/ros-navigation/navigation2/issues/1597) 

As such, closing

closed 2026-09-02 by SteveMacenski