ros-navigation/navigation2 #1627: Night builds blocked | No definition of libgazebo9-dev for Ubuntu Focal

ruffsl, 2020-04-07:
Looks like the docker hub builds have stalled on a rosdep install step,
given that libgazebo9-dev has not yet been released for Ubuntu Focal:

```
Step 19/31 : RUN . /opt/ros/$ROS_DISTRO/setup.sh && apt-get update && rosdep install -q -y --from-paths src --ignore-src && rm -rf /var/lib/apt/lists/*
---> Running in b1dad8c0128a
Get:1 http://packages.ros.org/ros2-testing/ubuntu focal InRelease [3910 B]
Get:2 http://security.ubuntu.com/ubuntu focal-security InRelease [97.9 kB]
Get:3 http://archive.ubuntu.com/ubuntu focal InRelease [265 kB]
Get:4 http://archive.ubuntu.com/ubuntu focal-updates InRelease [89.1 kB]
Get:5 http://packages.ros.org/ros2-testing/ubuntu focal/main amd64 Packages [16.3 kB]
Get:6 http://archive.ubuntu.com/ubuntu focal-backports InRelease [89.2 kB]
Get:7 http://archive.ubuntu.com/ubuntu focal/universe amd64 Packages [11.3 MB]
Get:8 http://archive.ubuntu.com/ubuntu focal/restricted amd64 Packages [33.5 kB]
Get:9 http://archive.ubuntu.com/ubuntu focal/main amd64 Packages [1281 kB]
Get:10 http://archive.ubuntu.com/ubuntu focal/multiverse amd64 Packages [177 kB]
Fetched 13.4 MB in 7s (2032 kB/s)
Reading package lists...
[91mERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
gazebo_dev: No definition of [libgazebo9-dev] for OS version [focal]
[0mThe command '/bin/sh -c . /opt/ros/$ROS_DISTRO/setup.sh && apt-get update && rosdep install -q -y --from-paths src --ignore-src && rm -rf /var/lib/apt/lists/*' returned a non-zero code: 1
```

How would we like to proceed to get the nightly build up again?
COLCON_IGNORE any packages bringing in gazebo dependencies?

SteveMacenski, 2020-04-07:
I saw that but expected it to resolve itself relatively soon, which evidently didn't happen. Is there a different version of gazebo or just not in rosdistro yet?

The nightly issues are unrelated I think - those are just failing on tear down.

ruffsl, 2020-04-07:
> Is there a different version of gazebo or just not in rosdistro yet?

Looks like we'll be jumping to Gazebo 11 soon:

https://answers.gazebosim.org/question/24720/does-gazebo-11-go-with-both-ros-melodic-and-neotic/?answer=24723#post-id-24723

SteveMacenski, 2020-04-07:
The only place I can see gazebo in the package.xml deps tree is the nav2_system_tests calls out ros_gazebo_pkgs which has gazebo_dev key. The foxy distribution.yaml has no gazebo yet but gazebo9-dev looks to be an entry in bionic https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml#L2175

SteveMacenski, 2020-04-08:
@ruffsl I haven't had the cycles to look into this yet, but please correct me if I'm wrong that this is something we need to change to get the builds up and running long term. It appears to me like a temporary problem while waiting on gazebo / ros pkgs to get released. If you don't believe that's the case, we should add this to the active queue to look into

ruffsl, 2020-04-08:
> It appears to me like a temporary problem while waiting on gazebo / ros pkgs to get released.

That's just about it here. There are a few other unrelated issues upstream where something in the master ros2 .repos file is broken, blocking fresh builds of the nightly archives, that the nightly docker tag installs, leading to stale nightly images. But that's out of scope for this ticket.

Anyhow, @chapulina is there a ticket we can subscribe to about ros_gazebo_pkgs for foxy?

closed 2020-05-09 by SteveMacenski