ros-navigation/navigation2 #1642: Allow generated velocities to be zero `(x= 0.0, y=0.0, theta=0.0)` to stop robot at goal.

Bitfroest, 2020-04-19:
## Feature request
Allow generated velocities to be zero `(x= 0.0, y=0.0, theta=0.0)`.

#### Feature description

In package `dwb_plugins` we handle trajectory (velocity) generation and check if our generated velocities are valid to kinematic constraints.

Validating a generated velocity is based on `min` and `max` speeds including:
```
min_speed_xy: 0.0
min_speed_theta: 0.0
```
If both values are zero I expect that my trajectory generator provides also trajectories with both speeds zero if this is allowed by acceleration and deacceleration limits. However, this is especially useful when waiting or for stopping a robot at the end. Instead, a velocity  `(x= 0.0, y=0.0, theta=0.0)` is handled as invalid. Thus, a robot will never be truly stopped at a Goal (e.g. the _StoppedGoalChecker_ will never trigger).

#### Implementation considerations

In `dwb_plugins/src/kinematic_parameters.cpp`:  

Remove sanity check of generated velocity `(x= 0.0, y=0.0, theta=0.0)` in `KinematicParameters::isValidSpeed()`, see line 61

[dwb_plugins/src/kinematic_parameters.cpp#L61](https://github.com/ros-planning/navigation2/blob/master/nav2_dwb_controller/dwb_plugins/src/kinematic_parameters.cpp#L61)

```diff
bool KinematicParameters::isValidSpeed(double x, double y, double theta)
{
  double vmag_sq = x * x + y * y;
  if (max_speed_xy_ >= 0.0 && vmag_sq > max_speed_xy_sq_ + EPSILON) {return false;}
  if (min_speed_xy_ >= 0.0 && vmag_sq + EPSILON < min_speed_xy_sq_ &&
    min_speed_theta_ >= 0.0 && fabs(theta) + EPSILON < min_speed_theta_) {return false;}
-  if (vmag_sq == 0.0 && theta == 0.0) {return false;}
  return true;
}
```

SteveMacenski, 2020-04-19:
I believe that would also make invalid trajectories appear valid. 

I believe once we’re in the pose tolerance of the goal we publish a set of zeros to stop the robot instead.

shrijitsingh99, 2020-04-21:
> Thus, a robot will never be truly stopped at a Goal (e.g. the _StoppedGoalChecker_ will never trigger).

The _StoppedGoalChecker_ does get triggered if the velocity is  within a set threshold (see L77-L79). By default the tolerance is set to 0.25 for linear as well angular velocity.

https://github.com/ros-planning/navigation2/blob/db4193fd87dbd9ffc12237f47814704793ec405b/nav2_dwb_controller/dwb_plugins/src/stopped_goal_checker.cpp#L69-L80

> I believe once we’re in the pose tolerance of the goal we publish a set of zeros to stop the robot instead.

Yup, it happens over here https://github.com/ros-planning/navigation2/blob/e15d8f228d8c9814506099e22aacce5ce6d7400a/nav2_controller/src/nav2_controller.cpp#L284

Bitfroest, 2020-04-22:
Thank you for your answers.

I was thinking about setting the velocity thresholds of the _StoppedGoalChecker_ to zero. Ideally it has implemented an Epsilon and gets triggered as the robot truly stopped (Odometry is zero + Epsilon). 
This would increase (intermediate) goal accuracy and precision. -> You can write a critic which controls the robot by millimeters close to the goal.
Furthermore, braking to zero is not done "blind" and we don't slide to the finish. 

Next: Waiting with critics. 
Imagine you face a dynamic obstacle on a straight path. So, you go straight. Then, your _Obstacle*Critic_ eliminates all trajectories (because of the dynamic obstacle). We will also slip a minimal distance, because we `publishZeroVelocity()`. Controller will cancel the path with an error before the dynamic obstacle is gone, right?

Instead, a critic could prefer zero velocity until the dynamic obstacle is gone or we exceed a waiting time. In addition to that, we can control the robot to zero instead of eliminating all trajectories and waiting for `publishZeroVelocity()`.

To sum up: I am convinced, that the check of zero velocity should be removed of the `KinematicParameters` because if I set the kinematic `min_*` and `min_speed_*` parameters to zero, theoretically it could be possible that the robot can drive zero velocities. To guarantee that the "outlier" zero in my sampled velocities generated of any _TrajectoryGenerator_ is not _KinematicParameters_ task in my point of view. 

But then we might provide the current velocity to `isValidSpeed()` to actually check if the velocity tested is within our kinematic (de)acceleration (which is not used by now). Thus, outliers will be gone and we allow zero.

SteveMacenski, 2020-04-22:
I think that's a fully valid situation, however to date there aren't any critics that can allow for a "waiting" maneuver in DWB. I do not believe that is the result of any intentional design decision, but just how things are right now and the lack of someone with a need for it implementing it. I'd more than welcome a PR to allow the robot the wait (no velocity) in the presence of a dynamic obstacle because of its interference.

shrijitsingh99, 2020-04-28:
Cannot this waiting functionality be currently implemented using the `Wait` recovery plugin?

SteveMacenski, 2020-04-28:
It could be, but that's a little bit of a hack. You'd like to be able to have the waiting modeled in the controller as a valid temporary state and not rely on the specific format of the behavior tree that the first recovery is waiting outside the controller BT node

shrijitsingh99, 2020-04-28:
True, it pretty much is a hack.

One thing I am unsure of is what will trigger such a waiting state? 
The dynamic obstacle should actually completely block all the possible paths for it to trigger and should not be activated when the robot is actually stuck.

SteveMacenski, 2020-04-28:
I think there's something that needs to be created to discern the difference between an intentional pause and a failure to generate. I think perhaps that would be a dynamic obstacle if the controller can tell that it is specifically dynamic.

shrijitsingh99, 2020-04-29:
Currently, I don't see an external trigger for this waiting state that seems like the only way as there is no way do differentiate between static and dynamic obstacles in costmap.
As you mentioned, if there existed a way, later on, we could have a check to see if trajectory was discarded because it collided with a dynamic obstacle then enter the waiting state if this condition is true.

SteveMacenski, 2020-04-29:
> Instead, a critic could prefer zero velocity until the dynamic obstacle is gone or we exceed a waiting time. In addition to that, we can control the robot to zero instead of eliminating all trajectories and waiting for publishZeroVelocity().

@Bitfroest totally fine in concept, I assume then you have a way of differentiating between the static and dynamic obstacle? 

> I am convinced, that the check of zero velocity should be removed of the KinematicParameters 

@Bitfroest I can get behind that is there's a replacement for checking for invalid trajectories due to failure. What would you suggest? I would agree either way, this check for validity doesn't belong in the parameters class but in the iterator where its used (which itself is a plugin to override). 

If you're willing to share more about your work that's led you to this thinking, I'd be super open to hearing about it (also merging a waiting-like behavior). That could be really valuable in dynamic environments like retail. I've implemented this type of thing before in other controllers but not in the DWB/DWA framework. 

Either way, we can shuffle around the interfaces so you can implement your plugin.

SteveMacenski, 2020-05-15:
@Bitfroest any update?

closed 2020-06-23 by Bitfroest