ros-navigation/navigation2 #1645: [DWB] StoppedGoalChecker does not consider xy_speed for checking if robot has stopped

shrijitsingh99, 2020-04-21:
Currently, _StoppedGoalChecker_ checks if the linear X & Y velocity are less then the threshold set to consider the robot as stopped. Instead of checking linear X and Y separately it is more intuitive to check if the speed (i.e. combined X & Y velocity) is less than the threshold. 

#### Expected behavior
The robot should be consider as stopped if the speed is within the set threshold and not the individual X & Y velocity.

#### Actual behavior

The robot is considered as stopped if the either of the linear X or Y velocity is within the threshold.

#### Additional information

https://github.com/ros-planning/navigation2/blob/db4193fd87dbd9ffc12237f47814704793ec405b/nav2_dwb_controller/dwb_plugins/src/stopped_goal_checker.cpp#L77-L79

SteveMacenski, 2020-04-21:
> The robot is considered as stopped if the either of the linear X or Y velocity is within the threshold.

Both would have to be, since these are all and statements. Y couldn't be 100m/s with X 0.001m/s to be considered stopped.

However, I understand the point you're trying to make, this should be the `hypot()` speed of them. Feel free to submit a PR.

shrijitsingh99, 2020-04-21:
> Both would have to be, since these are all and statements. Y couldn't be 100m/s with X 0.001m/s to be considered stopped.

Yeah, I missed that :sweat_smile: 

> However, I understand the point you're trying to make, this should be the `hypot()` speed of them. Feel free to submit a PR.

Great, will go ahead with it.

closed 2020-04-23 by SteveMacenski