ros-navigation/navigation2 #1648: Build fails for master branch

p-vega, 2020-04-23:
## Bug report

**Required Info:**

- Operating System:
  -  Ubuntu 20.04 
-Installation type:
  - From source, following these steps: https://ros-planning.github.io/navigation2/build_instructions/index.html#build-navigation2-master

- Version or commit hash:
  - Master branch (26d776b4b1e3b39c01bd2f3a32fc924fff04ddf5)
- DDS implementation:
  - 

#### Steps to reproduce issue

` git clone https://github.com/ros-planning/navigation2.git --branch master `
` colcon build --symlink-install`

#### Expected behavior
Build without errors 

#### Actual behavior
`--- stderr: dwb_core                                                                                 
In file included from /home/dev/ros2_foxy/install/rclcpp/include/rclcpp/node_options.hpp:25,
                 from /home/dev/ros2_foxy/install/rclcpp_lifecycle/include/rclcpp_lifecycle/lifecycle_node.hpp:40,
                 from /home/dev/navigation2_ws/install/nav2_costmap_2d/include/nav2_costmap_2d/costmap_2d_publisher.hpp:46,
                 from /home/dev/navigation2_ws/install/nav2_costmap_2d/include/nav2_costmap_2d/costmap_2d_ros.hpp:47,
                 from /home/dev/navigation2_ws/install/nav2_core/include/nav2_core/controller.hpp:42,
                 from /home/dev/navigation2_ws/src/navigation2/nav2_dwb_controller/dwb_core/include/dwb_core/dwb_local_planner.hpp:42,
                 from /home/dev/navigation2_ws/src/navigation2/nav2_dwb_controller/dwb_core/src/dwb_local_planner.cpp:41:
/home/dev/ros2_foxy/install/rclcpp/include/rclcpp/parameter.hpp: In function ‘void nav_2d_utils::moveParameter(const SharedPtr&, std::string, std::string, param_t, bool) [with param_t = double]’:
/home/dev/ros2_foxy/install/rclcpp/include/rclcpp/parameter.hpp:70:42: error: ‘value’ is used uninitialized in this function [-Werror=uninitialized]
   70 |   : Parameter(name, ParameterValue(value))
      |                                          ^
In file included from /home/dev/navigation2_ws/src/navigation2/nav2_dwb_controller/dwb_core/src/dwb_local_planner.cpp:47:
/home/dev/navigation2_ws/install/nav_2d_utils/include/nav_2d_utils/parameters.hpp:150:11: note: ‘value’ was declared here
  150 |   param_t value;
      |           ^~~~~
cc1plus: all warnings being treated as errors
make[2]: *** [CMakeFiles/dwb_core.dir/build.make:63: CMakeFiles/dwb_core.dir/src/dwb_local_planner.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:119: CMakeFiles/dwb_core.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< dwb_core	[ Exited with code 2 ]
Aborted  <<< nav2_controller                                                

Summary: 17 packages finished [1min 14s]
  1 package failed: dwb_core
  1 package aborted: nav2_controller
  2 packages had stderr output: dwb_core nav2_controller
  10 packages not processed
`

#### Additional information
There was a similar problem for the lifecycle branch #754

SteveMacenski, 2020-04-23:
I believe this may be in your set up, I'm see it build fine in CI which uses ROS2 master branch, last CI run: https://circleci.com/gh/ros-planning/navigation2/10384

While it is true I think both CI and I run on 18.04 right now so this may be a change from clang defaults. 

Feel free to submit a PR to have this have a default value and resolve the error @p-vega

p-vega, 2020-04-25:
Sure!

closed 2020-04-25 by SteveMacenski