ros-navigation/navigation2 #1651: More complex task demo using behavior trees

SteveMacenski, 2020-04-23:
A  demo to show the ways you can conduct an application using behavior trees. Either (or both for demo) pushing logic into the navigation stack behavior tree or using the NavigateToPose node as a component of an application behavior tree engine.

SteveMacenski, 2020-05-27:
This demo could also open the framework to be a "navigation-task framework" rather than soley a point-to-point navigation task. The behavior tree can bake in entire tasks that may need to be repeated. If you had a robot with an arm and you wanted to command it to go around, pick something up and drop it off, you could formulate that entire problem in the navigation2 BT and have 1 instance of that go->pick->go->drop->return as a single task (rather than thinking of each of the "go" as a navigation point-to-point request and the stuff between as other tasks). It would be certainly reasonable for many use cases to have the Nav2 BT be a sub-tree of a larger one to implement these features, but it really depends on the scope and consistency of the task requests.

SteveMacenski, 2021-08-23:
Covered by navigation.ros.org tickets for tutorials.

closed 2021-08-23 by SteveMacenski

tonynajjar, 2022-03-31:
Hi! I'm very interested in a demo demonstrating Nav2 being used as a subtree. I can't seem to find the corresponding ticket in https://github.com/ros-planning/navigation.ros.org/issues. Was this ever done? If not, do you know of some open-source project I could get inspired from? Thanks!

SteveMacenski, 2022-03-31:
That is it, but not yet completed. I know some folks are using it, but nothing that's been open sourced or public facing that I'm aware of