ros-navigation/navigation2 #1659: No deinition for OS [debian]

salmagro, 2020-04-28:
I am trying to install Navigation2, but I am getting the following error. 

I am sure that the setup.bash is sourced in the ROS 2 installation, but can not finish the installation.

## Bug report

**Required Info:**
- Operating System:
  Distributor ID:	Debian
  Description:	Debian GNU/Linux 9.12 (stretch)
  Release:	9.12
  Codename:	stretch

- Version or commit hash:
 https://github.com/ros-planning/navigation2.git --branch dashing-devel

#### Steps to reproduce issue
mkdir -p ~/navigation2_ws/src
cd ~/navigation2_ws/src
git clone https://github.com/ros-planning/navigation2.git --branch <ros2-distro>-devel
cd ~/navigation2_ws
rosdep install -y -r -q --from-paths src --ignore-src --rosdistro <ros2-distro>
#### Expected behavior
Install librarries
#### Actual behavior

ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
dwb_critics: No definition of [angles] for OS [debian]
nav2_bt_navigator: No definition of [behaviortree_cpp] for OS [debian]
nav2_behavior_tree: No definition of [behaviortree_cpp] for OS [debian]
nav2_system_tests: No definition of [gazebo_ros_pkgs] for OS [debian]
dwb_plugins: No definition of [angles] for OS [debian]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully

shivaang12, 2020-04-28:
Hi @salmagro,

I am assuming that you are trying to build from source. You might find bash script under tools directory useful. 

https://raw.githubusercontent.com/ros-planning/navigation2/dashing-devel/tools/initial_ros_setup.sh

Let me know this solves the issue.

SteveMacenski, 2020-04-28:
Debian is a tier 2 supported OS. Not all of the packages have been released for that OS so you’ll have to install from source.

salmagro, 2020-04-28:
@shivaang12 and @SteveMacenski ,  thank you very much for your help.

I am new on ROS, and not familiar with the terminology. What does it mean "installing from source"?

I was following the steps:
1. For installing ROS: https://index.ros.org/doc/ros2/Installation/Dashing/Linux-Install-Debians/
2. For installing Navigation 2: https://ros-planning.github.io/navigation2/build_instructions/index.html

I successfully followed the step for ROS. But for Navigation 2,  I have the problem as shown in this ticket.

Is there any tutorial for installing Navigation2 for Dashing from source?

shivaang12, 2020-04-28:
@salmagro "Installing from source" is building package on your machine rather than installing available (platform specific) binaries.

Sorry documentation are still work in progress. But please refer to the bash script which I referenced above.

Let me know if you need help.

SteveMacenski, 2020-04-28:
It doesn't look like you used the manually build instructions. Because there are dependencies that aren't released as binaries in Debian, you will need to build a bunch of stuff yourself that would ordinarily come packaged as binaries, since Debian is not a Tier 1 supported operating system in ROS2.

salmagro, 2020-04-28:
So far I understand for the link given by @shivaang12 and explanation of @SteveMacenski, what I need to do is run:
`echo "2. or run 'make' from navigation2/build/<project> folder"`
In every package that I need for Navigation 2.

Where I could find the list of all necessary packages that I should install? all the subfolders in the install directory?

I really appreciate your guide.

shivaang12, 2020-04-28:
@salmagro Sorry but you cannot use make to build as Navigation2 packages (Meaning all ROS2 packages) are build using Colcon build tool.

shivaang12, 2020-04-28:
@salmagro These are the dependent packages aside from ROS2. 

https://github.com/ros-planning/navigation2/blob/master/tools/ros2_dependencies.repos

salmagro, 2020-04-29:
While trying to install from source Nav2, I am following the steps mention here:
https://ros-planning.github.io/navigation2/build_instructions/index.html#build-navigation2-dependencies 

**Build Navigation2 Dependencies**
```
source ros2_ws/install/setup.bash
mkdir -p ~/nav2_depend_ws/src
cd ~/nav2_depend_ws
wget https://raw.githubusercontent.com/ros-planning/navigation2/master/tools/ros2_dependencies.repos
vcs import src < ros2_dependencies.repos
colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release
```
However I am getting the following error:
```
~/workspace_local/nav2_depend_ws$ colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release
Starting >>> camera_calibration_parsers
Starting >>> cv_bridge
Starting >>> image_transport
Starting >>> gazebo_dev
Starting >>> gazebo_msgs
Starting >>> image_geometry                                                        
Starting >>> angles
Starting >>> behaviortree_cpp_v3
Finished <<< angles [5.50s]                                                                   
Finished <<< gazebo_dev [5.62s]
--- stderr: camera_calibration_parsers                                                
/home/salmagro/workspace_local/nav2_depend_ws/src/image_common/camera_calibration_parsers/src/parse_ini.cpp:52:43: fatal error: rcpputils/filesystem_helper.hpp: No such file or directory
 #include "rcpputils/filesystem_helper.hpp"
                                           ^
compilation terminated.
make[2]: *** [CMakeFiles/camera_calibration_parsers.dir/src/parse_ini.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/home/salmagro/workspace_local/nav2_depend_ws/src/image_common/camera_calibration_parsers/src/parse_yml.cpp:45:43: fatal error: rcpputils/filesystem_helper.hpp: No such file or directory
 #include "rcpputils/filesystem_helper.hpp"
                                           ^
compilation terminated.
make[2]: *** [CMakeFiles/camera_calibration_parsers.dir/src/parse_yml.cpp.o] Error 1
/home/salmagro/workspace_local/nav2_depend_ws/src/image_common/camera_calibration_parsers/src/parse.cpp:43:43: fatal error: rcpputils/filesystem_helper.hpp: No such file or directory
 #include "rcpputils/filesystem_helper.hpp"
                                           ^
compilation terminated.
make[2]: *** [CMakeFiles/camera_calibration_parsers.dir/src/parse.cpp.o] Error 1
make[1]: *** [CMakeFiles/camera_calibration_parsers.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
---
Failed   <<< camera_calibration_parsers	[ Exited with code 2 ]
Aborted  <<< cv_bridge                                                        
Aborted  <<< image_geometry                                                   
Aborted  <<< gazebo_msgs                                                      
Aborted  <<< behaviortree_cpp_v3                                                      
Aborted  <<< image_transport                                           

Summary: 2 packages finished [37.6s]
  1 package failed: camera_calibration_parsers
  5 packages aborted: behaviortree_cpp_v3 cv_bridge gazebo_msgs image_geometry image_transport
  2 packages had stderr output: behaviortree_cpp_v3 camera_calibration_parsers
  7 packages not processed
```

I have read in #1369 some steps but I am not sure with this steps:
```
try fresh build

    make sure .bashrc is clean
    get the lastest ROS 2 packages (master branch)
    build ROS 2
    source setup.bash in ROS 2 workspace
    run rosdep install -y -r -q --from-paths src --ignore-src --rosdistro eloquent in both navigation2_ws and nav2_dependencies_ws.
    build nav2_dependencies
    source nav2_dependencies_ws/install/setup.bash
    build navigation2
```
Means that I have to re-install ROS2? in a master branch, instead of Dashing branch?
Should I erase my previous ros2_dashing installation?

Thanks in advance.

shivaang12, 2020-04-29:
@salmagro I think you are missing `rcpputils` package. You can clone it from [here](https://github.com/ros2/rcpputils). If you are using dashing make sure you are cloning dashing branch.

SteveMacenski, 2020-04-29:
> Means that I have to re-install ROS2? in a master branch, instead of Dashing branch?
Should I erase my previous ros2_dashing installation?

Yes. If you run master, you need to have master.

SteveMacenski, 2020-05-01:
@salmagro figure out your issues?

salmagro, 2020-05-02:
Actually now, but I have a better understanding of how the installation works after the explanations. I am having issues due different versions of python installed when trying to build Nav2 manually. I will start with a new env and close this one. Thank you very much four your guide.

closed 2020-05-02 by salmagro