ros-navigation/navigation2 #1660: Following a dynamic point at a certain distance

fmrico, 2020-04-28:
## Feature request

#### Feature description

I am interested in navigating to a dynamic point at a certain distance. Of course, I compromise to be working on this next week, but I would like to know your opinion about the best way to get it.

Using this feature, we could follow a person. Using Nav2 for following a person lets you avoid obstacles during the following.

#### Implementation considerations

- I am not very sure how to activate this behavior. Maybe extending bt_navigator to manage another action that activates this operation mode. 
- The (dynamic) goal position could be received using a subscription to a topic of `geometry_msgs/msg/Vector3Stamped`
- It should use another behavior tree that would compute the path to the pose and would follow the path. If the distance is lower than a configured distance (it could be set when activating this mode) the robot stop translation and only rotates to the goal, if it changes.
- How to detect the object to follow and what to do if it is lost is out of the scope of this feature, as this situation will be managed by the application (moving to the last know position, for example).

SteveMacenski, 2020-04-30:
@fmrico apologies, this got lost in my queue. 

I think the best way to approach this is actually a controller plugin (one of the reasons I fought so hard against the name “follow path”). If you have a thing you’d like to follow (as is perhaps a person or a carrot, whatever) you do not require any global planner at all. This is a suitable use and intent behind a controller / local planner. 

Now a question is: what to do about a planner? You could either remove the planner from the behavior tree altogether, or we could come up with a dumby global planner plugin that just returns success so that you can run in a controller only mode. I can see arguments for both.

fmrico, 2020-04-30:
As you said, Controller is very "path-oriented". It assumes that all the controller plugins must follow a path, and maybe it is very reasonable.

Maybe an alternative is a planner plugin for following a point. This planner could subscribe to a topic with an updated goal point, and it could generate a path to the dynamic point (at a configured distance that could be zero).

SteveMacenski, 2020-04-30:
> Controller is very "path-oriented". It assumes that all the controller plugins must follow a path

I disagree, you can implement what you propose as a controller most easily if you’re just driving towards a point and doing collision avoidance. I think the planner only makes sense if you don’t want to drive towards the point but plan some continuous path to it. Though you’ll need to increase the planning rate to be _much_ faster because its a dynamic point, which will pretty much turn into a controller.

fmrico, 2020-04-30:
Ok, I will do as you say.

fmrico, 2020-04-30:
I have been working today in an approximation on the planner level (https://github.com/fmrico/navigation2/tree/carrot_planner), but it's still incomplete. I could try to solve this at both levels (maybe someone is interested in pushing at controller level too) and we could discuss then the advantages on both approaches...

shrijitsingh99, 2020-05-03:
> I also somewhat wonder if you just had a pose of the robot and the pose of the thing you want to follow (minus the distance to follow by) as a path and you directly fed that into DWB, lowering the weights on path distance and raising it on goal distance if that would work out for you. Maybe you don't even need to do anything special and DWB's critics will deal with basic collision avoidance if something is in the way between your robot and its tracked object

Just to chip in my experience with this, I actually have tried this by using only ObstacleFoorprint and GoalDist critic for navigation. It works quite well as long as the environment is not too complicated.

fmrico, 2020-05-03:
Thanks @shrijitsingh99 . I will test your solution too ;)

I have implemented the first version something like a "carrot", both in the planner and in the controller, independently, of course. I am trying to reuse DWB and NavFn. My plugins inherit from them and just modify their behavior slightly.

I am increasingly convinced that the best solution is a planner instead of a controller. Setting a path as a straight line to the goal has problems when there are obstacles in the trajectory. A planner can avoid the obstacle better...

I continue with this...

SteveMacenski, 2020-05-05:
@fmrico got it. Feel free to also submit a PR to add this planner to navigation2. I think it has sufficient general reuse. I think we just need to make the target to track a standard interface so it can be used with any number of detectors. A reference detector that you used for your application as an example would be awesome (though probably not in this repo but linked to your specific application stack).

I agree the planner makes most sense if there's going to be things regularly in the way that are large enough the controller can't trivially get around them. Are you setting the new goal poses based on the dynamic obstacle through the behavior tree or through a topic your new planner listens to? The BT probably makes more sense from a reliability standpoint (just increase the replanning rate). That probably wouldn't then involve any new navigation code but still a very useful demo to show-case. I think this is a common application.

SteveMacenski, 2020-07-23:
@fmrico please update your PR to navigation.ros.org for the new params / locations / migration guide / adding that feature in 1 sentence to the landing page

closed 2020-08-05 by SteveMacenski