ros-navigation/navigation2 #1663: Update docs

adamkrawczyk, 2020-04-29:
## Bug report 

**Required Info:**

- Operating System:
  - ubuntu 18 
- Version or commit hash:
  -  website 
- DDS implementation:
  - cyclone dds

#### Steps to reproduce the issue
Just follow steps at the website to launch demo navigation with turtlebot3 

```
https://ros-planning.github.io/navigation2/getting_started/index.html
```

#### Expected behavior

Turtlebot driving autonomously 

#### Actual behavior

No launch file. Launch file provided with the package doesn't work.

#### Additional information

After launching simulation `ros2 launch nav2_bringup nav2_simulation_launch.py` there is no map. 
----

shivaang12, 2020-04-29:
Hi @adamkrawczyk I was wondering where are you looking for maps? In Nav2's repo or Turtlebot's repo?

SteveMacenski, 2020-04-29:
Nav2 simulation launch is circa dashing. Where did you find instructions to use that? 

The directions should tell you to use tb3_simulation.py.launch.

adamkrawczyk, 2020-04-29:
I'm just saying that the provided command doesn't work. Rest of the demo works fine.

SteveMacenski, 2020-04-29:
Where did you find that command, specifically?

shivaang12, 2020-04-29:
@adamkrawczyk Sorry, I think I miss understood you before. `tb3_simulation_launch.py` was introduced later after Dashing release.

@SteveMacenski Can we cherrypick this thing? or not?

SteveMacenski, 2020-04-29:
That page is for Eloquent and newer, we changed that launch file right after the dashing release to better describe its function. Sorry if that caused you trouble; we've been recommending users to use Eloquent or ROS2 master since Dashing is a little buggy and it was **massively** improved in Eloquent. 

I'd welcome a PR from you though at https://github.com/ros-planning/navigation.ros.org to add a "Note:" about if using dashing or older, use the `nav2_simulation_launch.py` file. You're now the second new user to mention that issue and we should address it.

adamkrawczyk, 2020-04-29:
Okay, thank you. I’m closing now

closed 2020-04-29 by adamkrawczyk