ros-navigation/navigation2 #1676: Legged robot demo

SteveMacenski, 2020-05-04:
Create a demonstration using Navigation2 and open-source tools to enable autonomous navigation of a quadruped, biped, or other non-wheeled robot to show that nav2 is capable of handling it.

This may require creating a new controller plugin to interface to the robot's stability / gait / footstep controller. Though if the controller is setup to accept a target velocity, we may be able to directly utilize DWB for computing control targets and the robot's specific controllers will use the sensors and velocity command to achieve those goals itself.

hyunseok-yang, 2021-08-12:
Here are quick question in terms for quadruped controller.

I tried to find the package in ROS2 version. but I couldn't. 
There is one of open project for that. https://github.com/chvmp/champ
But, as far as I know, it's ROS1 based.

Is there anyone who tried ROS2 based quadruped controller ? :)

SteveMacenski, 2021-08-12:
Look at the MIT cheetah repos, those are much better from a controls perspective. 

This ticket is more focused on the demonstration of quadruped _navigation_ using Nav2, in an applied deployment, and algorithms meant for the specific constraints of quad nav. It assumes theres some step planner (for vision based) and reactive walking gait controllers to track a given path / commanded velocity.

hyunseok-yang, 2021-08-12:
@SteveMacenski  thank you for your comment :)

So, for more clarification, if my understanding is correct, only 'odom' pose will be used for quadruped navigation, right?
And All quadruped controlling is low level side and it's depends on each commercial platform.

I'm not building a new one, just trying to bring-up any one of those, for example, SPOT(boston dynamics) model!

corot, 2025-03-27:
@SteveMacenski would not be enough just to pass Twist msgs as provided by Nav2 and let the robot controller to handle all the complexities of legged motion?  In principle a quadruped is an holonomic robot (I guess with extra sluggishness when advancing and turning at the same time)
🤔 

@hyunseok-yang , did you try?

SteveMacenski, 2025-03-27:
This is true! I just haven’t had the time or hardware to do this yet with. If using blind control, we can just pass in the twist and let the MPC / RL handle the terrain. If it’s a vision based system, then use the environmental model the hardware provides to do the gait and leg placement planning. In the original ticket description, I had larger ambitions, but the legged robotics space has grown so much since then. It makes much more sense now to leverage the robot’s internal leg controllers / placement planners and leave Nav2 to the higher levels. 

I’m confident that Nav2 works well for this case since we already have on the order of 10 companies I know using Nav2 with various quadruped robots. I still would like some kind of public demo, but it’s all ready to go!

SteveMacenski, 2025-05-07:
This was opened in the earlier days of legged robots. At this point, robot vendors provide pretty good solutions for lower level gait and full body control that I think we can exploit rather than relying on a generalized open-source framework (which can be useful for many doing research or developing their own systems, but I think most people buying legged robots for applications aren't trying to reinvent the wheel). 

This demonstration has been completed before on Unitree, Ghost, and Boston Dynamics robots, but unfortunately none so far as public demos with configurations and videos that I can show. But, many companies use Nav2 with legged robots successfully in a broad range of applications and environments.

Thus, still leaving this ticket open, I'd still like to be able to show a demo and tutorial for how to setup and work with Nav2 and legged robotics to make it easier for future developers.

CHAMP uses Nav2 / Navigation for mobility but I believe it tries to solve the problem of an open-source legged controller for leg planning / gaits

hyunseok-yang, 2025-06-05:
> [@SteveMacenski](https://github.com/SteveMacenski) would not be enough just to pass Twist msgs as provided by Nav2 and let the robot controller to handle all the complexities of legged motion? In [...]

Sorry for late response.

@corot  I've tried to manipulate CLOi Q9 of LG electronics which is two-legged wheel drive using twist message.  (models are not opend yet)
(refer to https://www.kedglobal.com/korean-innovators-at-ces-2025/newsView/ked202501090006_)

Just move go forward/backward and turn left/right by keyboard input. And other mechanical movements are calculated virtual robot controller(like micom).
(refer to https://github.com/lge-ros2/cloisim/tree/main/Assets/Scripts/Devices/Modules/Motor/SelfBalanceControl)