ros-navigation/navigation2 #1686: Migrate from deprecated BehaviorTree.CPP APIs

ruffsl, 2020-05-05:
Looks like navigation2 is using a few deprecated APIs from BehaviorTree.CPP. Looks like there where some important changes from v3.3.0. The latest release as of writing is v3.4.0.
 
<details>

Building https://github.com/ros-planning/navigation2/commit/b2c1661743f3407025ad0acf050fcb6c288537b3 agents https://github.com/BehaviorTree/BehaviorTree.CPP/releases/tag/3.4.0 

```Step 33/36 : RUN . $UNDERLAY_WS/install/setup.sh &&     colcon build       --symlink-install       --mixin $OVERLAY_MIXINS     || touch build_failed &&     if [ -f build_failed ] && [ -n "$FAIL_ON_BUILD_FAILURE" ]; then       exit 1;     fi
 ---> Running in 0dd8bbd22e7f
[connext_cmake_module] Warning: The location at which Connext was found when the workspace was built [[/opt/rti.com/rti_connext_dds-5.3.1]] does not point to a valid directory, and the NDDSHOME environment variable has not been set. Support for Connext will not be available.
Starting >>> nav2_common
Starting >>> nav_2d_msgs
Starting >>> nav2_gazebo_spawner
Finished <<< nav2_gazebo_spawner [0.91s]
Finished <<< nav2_common [1.30s]
Starting >>> nav2_msgs
Starting >>> nav2_voxel_grid
Finished <<< nav_2d_msgs [14.3s]
Starting >>> dwb_msgs
Finished <<< nav2_voxel_grid [16.2s]
Finished <<< dwb_msgs [23.4s]
Finished <<< nav2_msgs [53.8s]
Starting >>> nav2_util
Finished <<< nav2_util [23.1s]
Starting >>> nav_2d_utils
Starting >>> nav2_behavior_tree
Starting >>> nav2_lifecycle_manager
Starting >>> nav2_map_server
Starting >>> nav2_amcl
Starting >>> nav2_waypoint_follower
Finished <<< nav_2d_utils [8.92s]
Finished <<< nav2_lifecycle_manager [19.3s]
Starting >>> nav2_rviz_plugins
Finished <<< nav2_waypoint_follower [19.9s]
--- stderr: nav2_behavior_tree
/opt/overlay_ws/src/navigation2/nav2_behavior_tree/plugins/decorator/rate_controller.cpp: In member function ‘virtual BT::NodeStatus nav2_behavior_tree::RateController::tick()’:
/opt/overlay_ws/src/navigation2/nav2_behavior_tree/plugins/decorator/rate_controller.cpp:89:52: error: ‘void BT::TreeNode::setStatus(BT::NodeStatus)’ is protected within this context
   89 |         child_node_->setStatus(BT::NodeStatus::IDLE);
      |                                                    ^
In file included from /opt/underlay_ws/install/behaviortree_cpp_v3/include/behaviortree_cpp_v3/decorator_node.h:4,
                 from /opt/overlay_ws/src/navigation2/nav2_behavior_tree/plugins/decorator/rate_controller.cpp:21:
/opt/underlay_ws/install/behaviortree_cpp_v3/include/behaviortree_cpp_v3/tree_node.h:166:10: note: declared protected here
  166 |     void setStatus(NodeStatus new_status);
      |          ^~~~~~~~~
/opt/overlay_ws/src/navigation2/nav2_behavior_tree/plugins/decorator/rate_controller.cpp:95:52: error: ‘void BT::TreeNode::setStatus(BT::NodeStatus)’ is protected within this context
   95 |         child_node_->setStatus(BT::NodeStatus::IDLE);
      |                                                    ^
In file included from /opt/underlay_ws/install/behaviortree_cpp_v3/include/behaviortree_cpp_v3/decorator_node.h:4,
                 from /opt/overlay_ws/src/navigation2/nav2_behavior_tree/plugins/decorator/rate_controller.cpp:21:
/opt/underlay_ws/install/behaviortree_cpp_v3/include/behaviortree_cpp_v3/tree_node.h:166:10: note: declared protected here
  166 |     void setStatus(NodeStatus new_status);
      |          ^~~~~~~~~
make[2]: *** [CMakeFiles/nav2_rate_controller_bt_node.dir/build.make:63: CMakeFiles/nav2_rate_controller_bt_node.dir/plugins/decorator/rate_controller.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:193: CMakeFiles/nav2_rate_controller_bt_node.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/opt/overlay_ws/src/navigation2/nav2_behavior_tree/plugins/control/round_robin_node.cpp: In member function ‘virtual BT::NodeStatus nav2_behavior_tree::RoundRobinNode::tick()’:
/opt/overlay_ws/src/navigation2/nav2_behavior_tree/plugins/control/round_robin_node.cpp:49:23: error: ‘void BT::ControlNode::haltChildren(size_t)’ is deprecated: deprecated: please use explicitly haltChildren() or haltChild(i) [-Werror=deprecated-declarations]
   49 |         haltChildren(0);
      |                       ^
In file included from /opt/overlay_ws/src/navigation2/nav2_behavior_tree/plugins/control/round_robin_node.cpp:17:
/opt/underlay_ws/install/behaviortree_cpp_v3/include/behaviortree_cpp_v3/control_node.h:49:10: note: declared here
   49 |     void haltChildren(size_t first);
      |          ^~~~~~~~~~~~
/opt/overlay_ws/src/navigation2/nav2_behavior_tree/plugins/control/round_robin_node.cpp:53:23: error: ‘void BT::ControlNode::haltChildren(size_t)’ is deprecated: deprecated: please use explicitly haltChildren() or haltChild(i) [-Werror=deprecated-declarations]
   53 |         haltChildren(0);
      |                       ^
In file included from /opt/overlay_ws/src/navigation2/nav2_behavior_tree/plugins/control/round_robin_node.cpp:17:
/opt/underlay_ws/install/behaviortree_cpp_v3/include/behaviortree_cpp_v3/control_node.h:49:10: note: declared here
   49 |     void haltChildren(size_t first);
      |          ^~~~~~~~~~~~
cc1plus: all warnings being treated as errors
[... 100 lines cut ...]
                 from /opt/ros/foxy/include/rclcpp/executors.hpp:21,
                 from /opt/ros/foxy/include/rclcpp/rclcpp.hpp:145,
                 from /opt/overlay_ws/install/nav2_util/include/nav2_util/node_utils.hpp:19,
                 from /opt/overlay_ws/src/navigation2/nav2_behavior_tree/include/nav2_behavior_tree/bt_service_node.hpp:22,
                 from /opt/overlay_ws/src/navigation2/nav2_behavior_tree/plugins/action/clear_costmap_service.cpp:22:
/opt/ros/foxy/include/rclcpp/future_return_code.hpp:48:7: note: declared here
   48 | using FutureReturnCode [[deprecated("use rclcpp::FutureReturnCode instead")]] = FutureReturnCode;
      |       ^~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
make[2]: *** [CMakeFiles/nav2_clear_costmap_service_bt_node.dir/build.make:63: CMakeFiles/nav2_clear_costmap_service_bt_node.dir/plugins/action/clear_costmap_service.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:598: CMakeFiles/nav2_clear_costmap_service_bt_node.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< nav2_behavior_tree	[ Exited with code 2 ]
Aborted  <<< nav2_rviz_plugins
Aborted  <<< nav2_map_server
Aborted  <<< nav2_amcl

Summary: 10 packages finished [1min 52s]
  1 package failed: nav2_behavior_tree
  3 packages aborted: nav2_amcl nav2_map_server nav2_rviz_plugins
  2 packages had stderr output: nav2_behavior_tree nav2_rviz_plugins
  15 packages not processed
The command '/bin/sh -c . $UNDERLAY_WS/install/setup.sh &&     colcon build       --symlink-install       --mixin $OVERLAY_MIXINS     || touch build_failed &&     if [ -f build_failed ] && [ -n "$FAIL_ON_BUILD_FAILURE" ]; then       exit 1;     fi' returned a non-zero code: 1
```

SteveMacenski, 2020-05-05:
@facontidavide any thoughts on how to remove these `setStatus` calls that you made protected in 3.3.0 (https://github.com/BehaviorTree/BehaviorTree.CPP/releases)? I think that really messes up some of our custom control flow nodes and haltAllAction logic with the recursive visitor.

facontidavide, 2020-05-06:
**First**: your decorator does NOT need to set the status of the children.

https://github.com/BehaviorTree/BehaviorTree.CPP/blob/0f51631dadd7186a415f6c86f01066fadbabdb7c/src/decorator_node.cpp#L69-L78

Just remove `child_node_->setStatus(BT::NodeStatus::IDLE)`

**Second**: `haltChildren(0)` is a very bad API that makes me ashamed as a developer. Bad, bad Davide.

Use instead: `void ControlNode::haltChildren();`

**Third**: ...and this is apparent in the new [tutorials](https://github.com/BehaviorTree/BehaviorTree.CPP/tree/0f51631dadd7186a415f6c86f01066fadbabdb7c/examples).

`result = tree->root_node->executeTick();` 
becomes 
`result = tree->tickRoot();`

You can (should) also remove the line
`root_node->setStatus(BT::NodeStatus::IDLE);`
because it is done [under the hood](https://github.com/BehaviorTree/BehaviorTree.CPP/blob/1fb010263a6731ec332a386c983e89b409f75241/include/behaviortree_cpp_v3/bt_factory.h#L187-L190).

**Finally**:

`tree->root_node->halt()`
becomes
`tree->rootNode()->halt()`

facontidavide, 2020-05-06:
>  I think that really messes up some of our custom control flow nodes and haltAllAction logic with the recursive visitor.

I do not believe you will notice any difference in the control flow. If you do, please let me know.

SteveMacenski, 2020-05-06:
Awesome. Its become clear to me that someone really needs to take a really long, deep look at our use of BT.CPP, this isn't the first time our implementation is abusing things or making things redundant. I'm dissatisfied to say that I don't know this section of the code particularly well. A refactor / redesign may be in order to clean house of these problems and try to simplify.

deepanshubansal01, 2020-05-08:
I am working on this behavior tree task.

SteveMacenski, 2020-05-09:
@deepanshubansal01 I had to do this to unblock CI, apologies

closed 2020-05-09 by SteveMacenski

deepanshubansal01, 2020-05-09:
@SteveMacenski No worries.