ros-navigation/navigation2 #1696: ROS2 Navigation Defects

Aldis-K, 2020-05-06:
Required Info:
•	Operating System:
     o	ubuntu 18
•	Version or commit hash:
    o	Website

1. **Some nodes of ROS2 Navigation stack can’t be correctly killed by CTRL-C**

#### Steps to reproduce the issue:
```
ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py                                                
ros2 launch turtlebot3_navigation2 navigation2.launch.py use_sim_time:=True map:=${WORK_DIR}/install/turtlebot3_navigation2/share/turtlebot3_navigation2/map/map.yaml
```  
CTRL-C Kill process.

#### Expected behavior:
  All nodes shut down properly

#### Actual behavior:
  Some nodes of ROS2 Navigation stack can’t be correctly killed 

#### Additional information:
```
[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[rviz2-9] [INFO] [rclcpp]: signal_handler(signal_value=2)
[INFO] [lifecycle_manager-1]: process has finished cleanly [pid 2077]
[INFO] [amcl-3]: process has finished cleanly [pid 2081]
[INFO] [navfn_planner-6]: process has finished cleanly [pid 2092]
[INFO] [recoveries_node-7]: process has finished cleanly [pid 2093]
[navfn_planner-6] [INFO] [rclcpp]: signal_handler(signal_value=2)
[navfn_planner-6] [INFO] [navfn_planner]: Destroying
[bt_navigator-8] [INFO] [rclcpp]: signal_handler(signal_value=2)
[bt_navigator-8] [INFO] [bt_navigator]: Destroying
[world_model-4] [INFO] [rclcpp]: signal_handler(signal_value=2)
[world_model-4] [INFO] [world_model]: Destroying
[dwb_controller-5] [INFO] [rclcpp]: signal_handler(signal_value=2)
[amcl-3] [INFO] [rclcpp]: signal_handler(signal_value=2)
[amcl-3] [INFO] [amcl]: Destroying
[lifecycle_manager-1] [INFO] [rclcpp]: signal_handler(signal_value=2)
[lifecycle_manager-1] [INFO] [lifecycle_manager]: Destroying
[recoveries_node-7] [INFO] [rclcpp]: signal_handler(signal_value=2)
[map_server-2] [INFO] [rclcpp]: signal_handler(signal_value=2)
[dwb_controller-5] double free or corruption (fasttop)
[INFO] [bt_navigator-8]: process has finished cleanly [pid 2095]
[INFO] [rviz2-9]: process has finished cleanly [pid 2098]
[world_model-4] double free or corruption (fasttop)
[map_server-2] Magick: abort due to signal 11 (SIGSEGV) "Segmentation Fault"...
[ERROR] [dwb_controller-5]: process[dwb_controller-5] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[INFO] [dwb_controller-5]: sending signal 'SIGTERM' to process[dwb_controller-5]
[ERROR] [world_model-4]: process[world_model-4] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[ERROR] [map_server-2]: process[map_server-2] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[INFO] [world_model-4]: sending signal 'SIGTERM' to process[world_model-4]
[INFO] [map_server-2]: sending signal 'SIGTERM' to process[map_server-2]
[ERROR] [dwb_controller-5]: process has died [pid 2091, exit code -6, cmd '/opt/ros/dashing/lib/dwb_controller/dwb_controller __params:=/tmp/tmprcntpikh'].
[ERROR] [world_model-4]: process[world_model-4] failed to terminate '10.0' seconds after receiving 'SIGTERM', escalating to 'SIGKILL'
[ERROR] [map_server-2]: process[map_server-2] failed to terminate '10.0' seconds after receiving 'SIGTERM', escalating to 'SIGKILL'
[INFO] [world_model-4]: sending signal 'SIGKILL' to process[world_model-4]
[INFO] [map_server-2]: sending signal 'SIGKILL' to process[map_server-2]
[ERROR] [map_server-2]: process has died [pid 2078, exit code -9, cmd '/opt/ros/dashing/lib/nav2_map_server/map_server __node:=map_server __params:=/tmp/tmpcmv5eorn'].
[ERROR] [world_model-4]: process has died [pid 2090, exit code -9, cmd '/opt/ros/dashing/lib/nav2_world_model/world_model __params:=/tmp/tmp50nirk8l'].
```

2. **Some nodes of ROS2 navigation stack crashes without warning or exception**

#### Steps to reproduce the issue:
```
ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py   
ls /var/crash
```

#### Actual behavior:
  Some nodes of ROS2 navigation stack crashes without warning or exception

#### Additional information:
```
_opt_ros_dashing_lib_dwb_controller_dwb_controller.1000.crash             _opt_ros_dashing_lib_nav2_world_model_world_model.1000.crash
_opt_ros_dashing_lib_nav2_lifecycle_manager_lifecycle_manager.1000.crash  
_usr_bin_gzserver-9.0.0.1000.crash
_opt_ros_dashing_lib_nav2_map_server_map_server.1000.crash
```

SteveMacenski, 2020-05-06:
`turtlebot3_gazebo` please follow up with turtlebot3 repos. This is not our launch file. Please see if you can reproduce with our launch files to see if we can help.

Further, it looks like you're using a very old version of this software if `world_model` exists. I'd recommend upgrading to at minimum Eloquent (Foxy being released here in a matter of days) for support. So much has changed since then as we are continuing to develop the stack. If I were to guess, if this issue is due to software in this stack, its long been resolved.

SteveMacenski, 2020-05-06:
Also please update your ticket name to describe your issue. A general title like that doesn’t provide any context

SteveMacenski, 2020-05-08:
Pinging @Aldis-K

closed 2020-05-11 by SteveMacenski