ros-navigation/navigation2 #1706: Subtree are not able to share blackboard resources

naiveHobo, 2020-05-08:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Version or commit hash:
  - master

#### Steps to reproduce issue
Spin up a robot in the simulation with the following tree:
```xml
<root main_tree_to_execute="MainTree">
  <BehaviorTree ID="ReplanningTree">
    <RateController hz="1.0">
      <ComputePathToPose goal="{goal}" path="{path}" planner_id="GridBased"/>
    </RateController>
  </BehaviorTree>
  <BehaviorTree ID="MainTree">
    <PipelineSequence name="NavigateWithReplanning">
      <SubTree ID="ReplanningTree"/>
      <FollowPath path="{path}"  controller_id="FollowPath"/>
    </PipelineSequence>
  </BehaviorTree>
</root>
```

#### Expected behavior
```ReplanningTree``` is able to get blackboard resources and executions continues as expected.

#### Actual behavior
Throws a missing key error for resources on the blackboard andaborts bt_navigator configuration.

#### Additional information
```
[lifecycle_manager-13] [INFO] [1588963214.003140012] [lifecycle_manager_navigation]: Configuring bt_navigator
[bt_navigator-11] [INFO] [1588963214.003796274] [bt_navigator]: Configuring
[bt_navigator-11] [ERROR] [1588963214.033044873] []: Caught exception in callback for transition 10
[bt_navigator-11] [ERROR] [1588963214.033069811] []: Original error: Blackboard::get() error. Missing key [node]
[bt_navigator-11] [WARN] [1588963214.033102801] []: Error occurred while doing error handling.
[bt_navigator-11] [FATAL] [1588963214.033121137] [bt_navigator]: Lifecycle node entered error state
[lifecycle_manager-13] [ERROR] [1588963214.034978660] [lifecycle_manager_navigation]: Failed to change state for node: bt_navigator
[lifecycle_manager-13] [ERROR] [1588963214.035045085] [lifecycle_manager_navigation]: Failed to bring up all requested nodes. Aborting bringup.
```

SteveMacenski, 2020-05-08:
@naiveHobo I think this is a ticket for https://github.com/BehaviorTree/BehaviorTree.CPP,  I don't think there's anything about navigation that creates this problem.

closed 2020-05-08 by naiveHobo

SteveMacenski, 2020-05-10:
@naiveHobo please file a ticket on BT.CPP

m-morelli, 2020-12-09:
Hello, I've just now run into this issue as well. @naiveHobo / @SteveMacenski  have you opened an issue on BT.CPP, or have you found a workaround for this? Thanks

SteveMacenski, 2020-12-09:
https://github.com/BehaviorTree/BehaviorTree.CPP/issues/189 it was filed and closed. Davide's thoughts (that I disagree with):

> As clearly explained in the documentation, one of the MAIN purpose of subtree is isolating the blackboards. That is not a bug, it is the desired behaviour.

But he later caved to our pressure and added a `__shared_blackboard` port

cmraaron, 2022-08-03:
So subtrees are _required_ to use `__shared_blackboard`?  because someone might be grabbing out `{node}` in their constructor.
even `<Subtree ID="xx" node="{node}" />` doesn't work.
I wonder what is the answer here, grab `node` every time at `tick` time, or is some init time context required.
Or perhaps changes to that the blackboard attributes are copied at init time too.

SteveMacenski, 2022-08-17:
This would be a question for BT.CPP library that we use https://github.com/BehaviorTree/BehaviorTree.CPP

cmraaron, 2022-08-17:
So for those who may also face the same issue, I was able to get this to work.  `Subtree` has slightly different syntax in that its attributes are always blackboard pointers.
So to pass all the required bits to subtrees with nav2 components:
```
<Subtree ID="xx" node="node" tf_buffer="tf_buffer" bt_loop_duration="bt_loop_duration" server_timeout="server_timeout"/>
```
seems to do the trick.

If you want to pass additional information, use `<SubtreePlus>` eg
```
<SubtreePlus ID="xx" additional_info="blah" node="{node}" tf_buffer="{tf_buffer}" bt_loop_duration="{bt_loop_duration}" server_timeout="{server_timeout}"/>
```
nb the curly braces on this one

Perhaps to followup with BehaviourTree.CPP may be to see if information required at construction time might be better to come in some context object, leaving the blackboard for tick time use.