ros-navigation/navigation2 #1710: Working list of potential algorithms to include

SteveMacenski, 2020-05-09:
Hi all,

I have a working list of planners / controllers that I've been compiling and I wanted to share them here for potential ideas if anyone's interested in developing / adapting one in particular. Feel free to ping me about this and I'll help however I can

Planners
- [x] A* - holonomic robots or robots small relative to environment
- [x] Dij. - holonomic robots or robots small relative to environment
- [ ] OMPL - non-holonomic and non-circular footprint robots, potential formulation to include dynamic obstacles [IN PROGRESS]
- [ ] Hybrid-A* - non-holonomic and non-circular robots with smoothed paths [IN PROGRESS]
- [ ] D* variants - {Field, Lite, etc} holonomic robot or robot small relative to environment. Not sure value over A* or Dij. in practacal terms, low priority. A*/Dij. are sufficiently fast for any practical use
- [ ] Vornoi planner - for a robot to stay far, far away from things as its primary goal
- [ ] Graph / lane - planning in a predefined directional graph rather than free space [IN PROGRESS]
- [ ] State Lattice - non-holonomic and non-circular robots, useful in analog situations as Hybrid-A*

Controllers
- [x] DWA - compute trajectories based on weighted heuristics
- [x] TEB - compute trajectories based on optimal bands 
- [ ] MPC - {Linear, nonlinear} compute trajectories by model simulation 
- [ ] LQR/iLQR/CiLQR - "MPC-lite" is the easiest way to describe it
- [ ] Pure Pursuit - following a carrot [IN PROGRESS]
- [ ] Splines - {Bezier, B-spline, etc} follow a path using admissible curves

Recoveries
- [x] Spin - spin in place
- [x] Backup - back up from pose
- [ ] Alerts - sound, lights, etc to alert obstacle to move
- [x] Wait - wait out some duration or until event occurs
- [ ] Push - if you know its safe, push a thing out of the way
- [ ] Call for help - Slack, msgs, email, operations center request for assistance 
- [x] Clear - clear state, plans, and costmaps
- [ ] "Get out of here" - find some permissible path using a dynamically feasible trajectory to just get out of this area some fixed distance, if any way of motion is possible

feel free to comment to add more

maxlein, 2020-05-11:
Is there some issue/repo to see the progress of these tasks here?  
Like Graph / lane or Pure Pursuit.

SteveMacenski, 2020-05-11:
Pure Pursuit PR should land this week. Graph/lane is not likely to be ready anytime particularly soon. If someone else is interested in working on that, very much so welcome to do so.

soldierofhell, 2020-07-10:
Hi @SteveMacenski, as for Hybrid-A*, is there any draft on this planner? I'll need it soon, so I'm gonna do this, but it would be good to have some starting point (if there's any)

SteveMacenski, 2020-07-10:
Its in progress, if you're interested in collaborating with the 3 of us to finish it, I'm sure we can find a place for you to help out! 

I had intended to finish it months ago but had to drop it to do more community management tasks and unblocking other people.

soldierofhell, 2020-08-13:
I did my first approch, but seems like A* is too slow on large grid. Does anyone know any efficient parallization of A* on CPU?

robomoco, 2020-09-03:
> Its in progress, if you're interested in collaborating with the 3 of us to finish it, I'm sure we can find a place for you to help out!
> 
> I had intended to finish it months ago but had to drop it to do more community management tasks and unblocking other people. Starting next week I should be working on it again in a limited capacity that will grow as other existing commitments complete.
> 
> I'd love to hear about your requirements, needs, and uses to make sure it meets your needs.

Hi @SteveMacenski 
A colleague and I have some spare cycles to help with Hybrid-A*. What is the best way to get involved?

SteveMacenski, 2020-09-03:
Hi @robomoco please join the slack with this link https://join.slack.com/t/navigation2/shared_invite/zt-gnsw70li-tUsWwB1C9h3FLIabNVvB~A and introduce yourself on the on-boarding channel. You'll probably want to join the hybrid A* channel and also give me a PM. I'm in the middle of working on that this week too so its a really good time to jump in since my head is in it.

We also have a bi-weekly working group meeting on Thursdays. If you go to the TSC ROS2 page, at the bottom there's a calendar. You can add yourself to the navigation2 meetings and there's a link there for the Google Meet meeting.

SteveMacenski, 2020-10-10:
@soldierofhell the hybrid A* planner has been merged and ready for use!

SteveMacenski, 2021-03-03:
Pure Pursuit now also merged and complete!

dkuenster, 2021-03-04:
Porting mpc_local_planner to ROS2/Navigation 2 here:
 https://github.com/rst-tu-dortmund/mpc_local_planner/pull/35

SteveMacenski, 2021-03-04:
Sounds good! I thought it was still experimental, so I didn't add that to my queue to work on, thanks for the great work!

mkolodziejczyk-piap, 2021-03-05:
Thanks @dkuenster for putting it all together. I actually did it my port few months ago and just couldn't find time to clean to public and in the meantime it grew with some additional features. I'll try to review your work. We've been using it already with LGSVL, Carla and custom Gazebo simulator and now starting more testing on real vehicles. In general I'm happy with results and the similarity to teb interface.
BTW I'm using acceleration control and thinking about implementing AckermannDrive message replacement in nav2, because now I have to put acceleration, jerk and steering angle into Twist fields

dkuenster, 2021-03-15:
@mkolodziejczyk-piap A review would be much appreciated. The new AckermannDrive msgs also sound interesting.

PauloFavero, 2021-04-16:
Hi @SteveMacenski,

I am researching comparing planners applied to a holonomic robot. Asides from A* and Dijkstra, I would like to also use and compare algorithms such as RRT, RRT*, or PRM, for example. 

I found this [issue]( https://github.com/ros-planning/navigation2/issues/1646) that is a work to incorporate OMPL as a planner plugin. I didn't see this implementation integrated on nav2 or anything related to this on the roadmap. is the implementation of such planner algorithms planned soon? As I read on the issue, the focus of the project changed due to other priorities.

If there is need, I will be interested in implementing an RRT planner plugin.

SteveMacenski, 2021-04-16:
Hi,

There are no plans on creating an RRT or similar planner in Nav2 currently. We had that work going on and did a great deal of prototyping but were unable to get a planner of sufficient performance worth using in Nav2. At best, we were able to get an RRT* planner using Dubin motion models to work relatively well, but then Hybrid-A* is a better choice for mobile robot navigation and uses the same models (and faster). 

If there was a legitimate need for an RRT (or similar) planner, I'd like to hear it and if it fills a niche that another algorithm we already have or is on the roadmap doesn't solve, then we can add it and I'd definitely appreciate your help in implementing it

PauloFavero, 2021-04-18:
Thanks for your feedback @SteveMacenski. 

What performance requirement the RRT* prototype didn't fill? Convergence time, time to cover a region... Talking about performance issues, do you have a benchmark comparing these methods when implemented on Nav2? Like RRT* vs Hybrid-A*?

soldierofhell, 2021-04-18:
I'd compare probabilistic methods (like RRT*) vs search based to Monte Carlo integration vs regularną quadratures. Probabilistic methods are for higher dimension spaces, to overcome curse of dimensionality. This is their strength, not in SE2 space.

Parv-Maheshwari, 2021-07-06:
@SteveMacenski continuing our discussion from #2042 , me and my team have a **sampling based parallelized local planner in Frenet frame**. Our code is available [here](https://github.com/anime-sh/Frenet_Planner_AGV).

We have applied **parallelisation** on the **entire trajectory generation step** - sampling trajectories in Frenet frame, trajectories converted from Frenet frame to global frame and collision checking. We also have **used openmp in our costmap_callback function** which accesses the costmap and updates the obstacle coordinates.

You can find these in two files - ./frenet_planner_agv/src/frenetROS_obst.cpp and frenet_planner_agv/src/frenet_optimal_trajectory.cpp

We have already started working on porting our code to ROS2.

P.S. - We are also cleaning the code so apologies for that.

SteveMacenski, 2021-07-06:
I've taken a glance at the codebase, but clearly any analysis I do without really diving into it isn't going to be complete. So it might be helpful if you started from the beginning and provide us context and a common understanding of what you've done to start off a discussion with. For example answering the questions
- What was your rationale for this particular algorithm? What environments / robot types / drivetypes is this meant for and tested on?
- What is the benefit or detractors of this approach when compared to DWB, TEB/MPC, or a Pure Pursuit algorithm? 
- Has this been significantly tested / validated on hardware and is this something you actually are using on some product / advanced research on a regular basis?
- What group would benefit from this being in Nav2 (e.g. what problem does this solve that is sufficiently reusable that it is worth us maintaining in Nav2 itself)?

Parv-Maheshwari, 2021-07-08:
>     * What was your rationale for this particular algorithm? What environments / robot types / drivetypes is this meant for and tested on?

The planner in its current state, is deployed & tested on a non-holonomic electric vehicle. The planner samples in both spatial & temporal with special focus being given to minimization of jerk for passenger comfort. Previous works focused solely on finding the shortest path only which given the use case of autonomous driving isn't the most suited factor for consideration.

>     * What is the benefit or detractors of this approach when compared to DWB, TEB/MPC, or a Pure Pursuit algorithm?

TEB works very well for holonomic vehicles. But in cases of non-holonomic vehicles, there is a lot of parameter tuning that needs to be done to make it functional on an actual vehicle. Even then the algorithm has a large overhead being a heuristic based one. Comparisons on an actual test vehicle (Mahindra e2O) showed that while TEB ran at <1Hz planning frequency, we were able to attain 3-5Hz on frenet without any kind of optimizations.

Apart from that the algorithm is a local planning algorithm & still requires running MPC and/or pure pursuit for controller and tracking purpose for actual vehicle (as can be observed in our package)

>     * Has this been significantly tested / validated on hardware and is this something you actually are using on some product / advanced research on a regular basis?

Before the onset of COVID, the above package has been successfully tested on an actual test vehicle Mahindra e2O along with successful Gazebo simulations.

>     * What group would benefit from this being in Nav2 (e.g. what problem does this solve that is sufficiently reusable that it is worth us maintaining in Nav2 itself)?

 The planner is specially useful for non-holonomic scenarios. But sampling on both spatial & temporal values is the key component of the algorithm and thus makes it applicable in various other fields specially but not restricted to where lane like scenarios are applicable.
> 
> At least one concern I have from looking over it is that I see a parameter for "road width" which makes me question this approach's usefulness to a broad class of mobile robotics problems, since that is not typically going to be usable for free space planning and for some specific mobile robotics applications where you do have lanes, they will probably vary at different locations. I'm fine with having some specialized (e.g. non freespace) planners if there are sufficient reusability, but I'm not sure what that scope is here.

The road_width value can be considered as a parameter which limits the range of sampling of lateral offset at the final point of the trajectory.

This can be changed according to the lane width that is supposed to be input from the laserscan/semantic_segmentation module.

SteveMacenski, 2021-07-08:
Hi,

I understand better now. I don't think this plugin would make sense for us to add to Nav2 directly, but I'd be happy to include it in the list of Nav2 plugins available for users to use on our documentation https://navigation.ros.org/plugins/index.html if/when it is made available in ROS2. The reason is that this controller is really more suited to AV situations and this is an autonomous mobile robot framework. While you could use this on an AV, it would NOT be recommended and none of our design requirements are based on that intention. I would be open to reconsidering this if you could provide examples of other independent users of this algorithm in conjunction with Nav2 and significant hardware validation testing / deployment. 

Does that make sense?

Parv-Maheshwari, 2021-07-08:
Sure. We will work on ROS2 port as of now and would also try to implement it in non AV scenarios to expand the userbase and functionality. Thank you. Would update here when significant progress has been made.

salmagro, 2021-11-02:
>     * MPC /Nonlinear MPC- compute trajectories by model simulation supporting high speeds. Exact path follower / dynamic obstacles in formulation

Is there any source for this implementation (repository/paper)? There are several options for this, but I would like to know which specific approach are you analysing.

vinnnyr, 2022-03-17:
Any feedback on thinking of e_star style nav2 planner for online coverage? https://linkslab.uconn.edu/wp-content/uploads/sites/246/2018/11/e_star_Preprint.pdf

SteveMacenski, 2022-03-17:
There are time / space optimal coverage algorithms that depend on what your application is for which makes most sense (e.g. randomly bouncing around is a  bad idea for a lawn mower; and row by row coverage is probably bad for a security robot) so I don't think there's going to be 1 coverage algorithm to rule them all. 

I do wonder thusly if a coverage algorithm makes sense to add to Nav2 itself, but we certainly can use more of them in the ecosystem and link to it to Nav2 documentation so that others can be aware of them and use them. 

Some others that exist for ROS 1
- https://github.com/Greenzie/boustrophedon_planner
- https://github.com/ethz-asl/polygon_coverage_planning
- https://github.com/nobleo/full_coverage_path_planner

Which I have explored in the past and for various reasons I don't recall for each I decided not to continue to explore for Nav2 inclusion. I believe at least one wasn't actually being used and more by the company and at least one more it wasn't actually used ever for more than a paper

LotfiZ, 2022-06-29:
Hi,

I dont know if its relevant here but i'm wondering if the controllers listed above takes into account the localization uncertainties ? We experienced with TEB for example that even if the trajectory is optimal, the execution is jerky. First i thought that the very simple control law implemented there caused this, but i'm not sure that it's the only reason ! Do you think that with MPC ( and its variants) the trajectory tracking will be smooth even using an accurate models  ? 

Thanks

SteveMacenski, 2022-06-29:
All trajectory planners work in the odometry frame which is continuous and not subject to localization uncertainty / jumping

LotfiZ, 2022-06-30:
Thanks Steve for your answer. Correct me if i'm wrong but if for example the localization stack (e.g AMCL ) jumps for X reason while performing local planning, the first pose of the TEB will change as its the pose of the robot on the map  so the TEB will try continuously  "compensate" for this kind of drift. In another word : if i take same robot in simulation with same scenario, in first place i run TEB with a perfect localization system and after that  i run the same configuration with putting some uncertainties in localization. The output commands will be more jerky than in the case of perfect localization. Even in reality, a part of hardware limitation, the worse my localization, the more jerk i have. 

Concerning the control law, dont you think that an improvement in the actual one for TEB ( takes two first poses of the optimized trajectory within the time difference between them and compute the velocity commands ) could be interesting to revaluate ? Because at each frequency cycle both of optimizer and computation of the velocity commands are executed. Is it a good approach for example to separate this two execution with different frequencies ? I read something pointing that here : [https://github.com/BadgerTechnologies/teb_local_planner/commit/460d73b21be3a1f8f20b0535190ac319d073005a](url)

I'm very interested about contributing  to limit the jerk in the TEB without using an external module like a velocity smoother. we have the opportunity to work with real industrial AGVs/Forklifts so we are able to tests/log a lot of things in real life scenarios ! 

Thank you very much !

SteveMacenski, 2022-07-01:
I can't answer you how TEB works internally too much, that's not my codebase. What I'm saying though is that the trajectory planner operates in the odometry frame which is smooth and continuous, not subject to localization updates. The global plan is probably transformed into the odometry frame for use, as we do in Nav2's internal trajectory planners, which could then be subject to some of those discontinuities in the position of the path, but the trajectory outputs should still not jerk if the smoothness terms are evaluated. But I'm can't get into the specific details of TEB, I really haven't looked that closely. 

 TEB is not in this project, if you have concerns, you should file tickets with TEB directly.

SteveMacenski, 2023-01-04:
GPS covered by another ticket and MPPI/MPC is imminent to be added in January. I think that closes this out for the cross-section of capabilities I wanted to have added! It only took me 3 years :laughing:

closed 2023-01-04 by SteveMacenski

chenwu-cc, 2023-02-28:
Hello, is MPPI online?

chenwu-cc, 2023-02-28:
hello，I can't find the MPPI，

salmagro, 2023-02-28:
> hello，I can't find the MPPI，

Please give a look at https://github.com/artofnothingness/mppic 
The MR is still open: https://github.com/ros-planning/navigation2/pull/3350

chenwu-cc, 2023-02-28:
ok,thank you

chenwu-cc, 2023-02-28:
> > hello，I can't find the MPPI，
> 
> @chenwu-cc Please give a look at https://github.com/artofnothingness/mppic The MR is still open: #3350

I also want to ask, do you know if there is a ROS2 implementation of mpc

SteveMacenski, 2023-03-01:
MPPI merged here: https://github.com/ros-planning/navigation2/pull/3350

MPPI is an MPC controller. We selected this over another variant of MPC because MPPI lets us set up less restrictive penalty functions which are necessary to do "realistic" mobile robotics things beyond just chase a carrot and minimize cross-track error.

chenwu-cc, 2023-03-02:
> MPPI merged here: #3350
> 
> MPPI is an MPC controller. We selected this over another variant of MPC because MPPI lets us set up less restrictive penalty functions which are necessary to do "realistic" mobile robotics things beyond just chase a carrot and minimize cross-track error.

I used the Humble version to compile the mppi, but an error occurred. What is your working environment?

SteveMacenski, 2023-03-02:
Its in `main`, so Rolling and newer. It may be backported to Humble, but not for several weeks at the earliest.

aladin6060, 2024-02-29:
I need a controller that follows a given path with an exact speed. Can I realize this with MPPI? How could I do that?

I have a working implementation in a custom MPC controller, with a cost function for the velocity, but I would like to move to the NAV2 stack.