ros-navigation/navigation2 #1711: navigation2 source build error

gokul-gokz, 2020-05-09:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Version or commit hash:
  - 
- DDS implementation:
  - 

#### Steps to reproduce issue

```
Followed the quick installation steps given in navigation2:
mkdir ros2
cd ros2
wget https://raw.githubusercontent.com/ros-planning/navigation2/master/tools/initial_ros_setup.sh
chmod a+x initial_ros_setup.sh
./initial_ros_setup.sh
```
#### Error Message
```
--- stderr: nav2_behavior_tree                                                
In file included from /home/gokul/ros2/navigation2_ws/src/navigation2/nav2_behavior_tree/plugins/action/reinitialize_global_localization_service.cpp:22:0:
/home/gokul/ros2/navigation2_ws/src/navigation2/nav2_behavior_tree/include/nav2_behavior_tree/bt_service_node.hpp: In member function ‘virtual BT::NodeStatus nav2_behavior_tree::BtServiceNode<ServiceT>::check_future(std::shared_future<typename ServiceT::Response::SharedPtr>)’:
/home/gokul/ros2/navigation2_ws/src/navigation2/nav2_behavior_tree/include/nav2_behavior_tree/bt_service_node.hpp:101:40: error: ‘using FutureReturnCode = enum class rclcpp::FutureReturnCode’ is deprecated: use rclcpp::FutureReturnCode instead [-Werror=deprecated-declarations]
     rclcpp::executor::FutureReturnCode rc;
                                        ^~
In file included from /home/gokul/ros2/ros2_ws/install/rclcpp/include/rclcpp/executor.hpp:34:0,
                 from /home/gokul/ros2/ros2_ws/install/rclcpp/include/rclcpp/executors/multi_threaded_executor.hpp:25,
                 from /home/gokul/ros2/ros2_ws/install/rclcpp/include/rclcpp/executors.hpp:21,
                 from /home/gokul/ros2/ros2_ws/install/rclcpp/include/rclcpp/rclcpp.hpp:146,
                 from /home/gokul/ros2/navigation2_ws/install/nav2_util/include/nav2_util/node_utils.hpp:19,
                 from /home/gokul/ros2/navigation2_ws/src/navigation2/nav2_behavior_tree/include/nav2_behavior_tree/bt_service_node.hpp:22,
                 from /home/gokul/ros2/navigation2_ws/src/navigation2/nav2_behavior_tree/plugins/action/reinitialize_global_localization_service.cpp:22:
/home/gokul/ros2/ros2_ws/install/rclcpp/include/rclcpp/future_return_code.hpp:48:97: note: declared here
 using FutureReturnCode [[deprecated("use rclcpp::FutureReturnCode instead")]] = FutureReturnCode;
                                                                                                 ^
In file included from /home/gokul/ros2/navigation2_ws/src/navigation2/nav2_behavior_tree/plugins/action/clear_costmap_service.cpp:22:0:
/home/gokul/ros2/navigation2_ws/src/navigation2/nav2_behavior_tree/include/nav2_behavior_tree/bt_service_node.hpp: In member function ‘virtual BT::NodeStatus nav2_behavior_tree::BtServiceNode<ServiceT>::check_future(std::shared_future<typename ServiceT::Response::SharedPtr>)’:
/home/gokul/ros2/navigation2_ws/src/navigation2/nav2_behavior_tree/include/nav2_behavior_tree/bt_service_node.hpp:101:40: error: ‘using FutureReturnCode = enum class rclcpp::FutureReturnCode’ is deprecated: use rclcpp::FutureReturnCode instead [-Werror=deprecated-declarations]
     rclcpp::executor::FutureReturnCode rc;
                                        ^~
In file included from /home/gokul/ros2/ros2_ws/install/rclcpp/include/rclcpp/executor.hpp:34:0,
                 from /home/gokul/ros2/ros2_ws/install/rclcpp/include/rclcpp/executors/multi_threaded_executor.hpp:25,
                 from /home/gokul/ros2/ros2_ws/install/rclcpp/include/rclcpp/executors.hpp:21,
                 from /home/gokul/ros2/ros2_ws/install/rclcpp/include/rclcpp/rclcpp.hpp:146,
                 from /home/gokul/ros2/navigation2_ws/install/nav2_util/include/nav2_util/node_utils.hpp:19,
                 from /home/gokul/ros2/navigation2_ws/src/navigation2/nav2_behavior_tree/include/nav2_behavior_tree/bt_service_node.hpp:22,
                 from /home/gokul/ros2/navigation2_ws/src/navigation2/nav2_behavior_tree/plugins/action/clear_costmap_service.cpp:22:
/home/gokul/ros2/ros2_ws/install/rclcpp/include/rclcpp/future_return_code.hpp:48:97: note: declared here
 using FutureReturnCode [[deprecated("use rclcpp::FutureReturnCode instead")]] = FutureReturnCode;
         

#### Things done so far
1. Installation through binaries work but the source installation fails.
2. Tried the manual installation steps as well but received the same error.

SteveMacenski, 2020-05-09:
This PR resolves: https://github.com/ros-planning/navigation2/pull/1702 but our CI for some reason hasn't picked it up yet when it was submitted a few days ago. Trying again now.

SteveMacenski, 2020-05-09:
Hang tight - what I recommend right now is to install Eloquent and use those binaries to develop. That will work fine for you.

closed 2020-05-10 by SteveMacenski