ros-navigation/navigation2 #1715: Missing test dependencies in master

SteveMacenski, 2020-05-09:
Reports of missing test dependencies we need to add to the package.xml files. Can be reproduced by 

```
cd tools
docker build -t nav2:release -f release.Dockerfile .
```

after reenabling https://github.com/ros-planning/navigation2/blob/master/tools/release.Dockerfile#L91-L102

siddhya, 2020-05-10:
How do I get the build logs out from docker? Will need those to find out what dependencies need to be added to each package.xml

`#21 3.429 build/costmap_queue/Testing/20200510-0904/Test.xml: 7 tests, 0 errors, 7 failures, 0 skipped                                           #21 3.429 build/dwb_core/Testing/20200510-0904/Test.xml: 5 tests, 0 errors, 5 failures, 0 skipped                                                #21 3.429 build/dwb_critics/Testing/20200510-0904/Test.xml: 5 tests, 0 errors, 5 failures, 0 skipped`

SteveMacenski, 2020-05-11:
#1718 covers @siddhya I see that I have a PM on Slack from you, I will get to that today promise :smile: 

In the future, if you go into the docker container, you should be able to see the logs with `colcon test-result --verbose`. I agree, I would really like that to be the output ( or an option of the output ) of the test call itself.

SteveMacenski, 2020-05-12:
@siddhya did that get you unblocked?

siddhya, 2020-05-13:
Yes but when I comment out the vcs import and uncomment COPY to use my local navigation2 code in the docker I get the below error. Local code is on latest master.

```
#17 3.472 ERROR: Rosdep experienced an error: Multiple packages found with the same name "nav2_gazebo_spawner":
#17 3.472 - ros-planning/navigation2/install/nav2_gazebo_spawner/share/nav2_gazebo_spawner
#17 3.472 - ros-planning/navigation2/nav2_bringup/nav2_gazebo_spawner
```

SteveMacenski, 2020-05-13:
Did you run one build with symlink and other without it?

siddhya, 2020-05-14:
This is the only change I made to the docker file
```
diff --git a/tools/release.Dockerfile b/tools/release.Dockerfile
index 2b89b534..3f188b2e 100644
--- a/tools/release.Dockerfile
+++ b/tools/release.Dockerfile
@@ -29,9 +29,9 @@ repositories: \n\
     url: https://github.com/ros-planning/navigation2.git \n\
     version: ${ROS_DISTRO}-devel \n\
 " > ../overlay.repos
-RUN vcs import ./ < ../overlay.repos && \
-    find ./ -name ".git" | xargs rm -rf
-# COPY ./ ./ros-planning/navigation2
+#RUN vcs import ./ < ../overlay.repos && \
+#    find ./ -name ".git" | xargs rm -rf
+COPY ./ ./ros-planning/navigation2
```

And ran it using
$ docker build -t nav2:release -f tools/release.Dockerfile . |& tee docker.log

siddhya, 2020-05-14:
Tests are failing with the following error:
```
#22 3.609 build/costmap_queue/Testing/20200514-1437/Test.xml: 7 tests, 0 errors, 7 failures, 0 skipped
#22 3.609 - mbq_test
#22 3.609   <<< failure message
#22 3.609     Traceback (most recent call last):
#22 3.609       File "/opt/ros/eloquent/share/ament_cmake_test/cmake/run_test.py", line 19, in <module>
#22 3.609         import ament_cmake_test
#22 3.609     ModuleNotFoundError: No module named 'ament_cmake_test'
```

ros-eloquent-ament-cmake is not installed in the docker image from ros:eloquent

siddhya, 2020-05-14:
Adding the middle line fixed the dependency issues for me. Will raise a PR if it makes sense.

But this file uses ros:eloquent as base and that has stopped working since  #1714 
```
+RUN if [ -z "$RUN_TESTS" ]; then \
+        . /opt/ros/$ROS_DISTRO/setup.sh && \
+        colcon test \
```

SteveMacenski, 2020-05-14:
@ruffsl does that sound right to you? I was thinking maybe not the best since that implies potentially packages from the docker image and not through rosdep necessarily, however, I do see that we do that for building the overlay workspace https://github.com/ros-planning/navigation2/blob/master/tools/release.Dockerfile#L76. So that seems reasonable to me. Since you reported this issue, would this be a good solution?

ruffsl, 2020-05-14:
```
+        . /opt/ros/$ROS_DISTRO/setup.sh && \
```

Derp. Never mind this issues then. It'll be included in the refresh of the Dockerfiles: https://github.com/ros-planning/navigation2/pull/1731/commits/df955b219c02a573bcb9e4877374dafb81625693

closed 2020-05-14 by ruffsl

SteveMacenski, 2020-05-14:
Thanks @siddhya for taking a look! It also appears the planner test issue wasn't actually fixed... mind working with Alexey on that? Its come back on every PR so far.