ros-navigation/navigation2 #1719: Node namespace arg is not forwarded to node in launch files.

mvoellmy, 2020-05-11:
#854  Bug report

I am not a hundred percent sure if this is a bug or intended, but in case it is a bug I can create a pull request and fix it. I just would like confirmation if my thinking is correct.

**Required Info:**

- Operating System:
Ubuntu 18.04
- Version or commit hash:
 from source: 9615c2ef5d3c89db901f916b136845367e5fcabb
- DDS implementation:
  - Fast-RTPS (i think)

#### Steps to reproduce issue
Run
```
ros2 launch nav2_localization_launch.py namespace:=foo
```
check nodes namespace w/

```
ros2 node list
```

#### Expected behavior
Launching `ros2 launch nav2_localization_launch.py namespace:=foo` should result in the nodes being launched as `foo/map_server` etc.

#### Actual behavior
Instead the nodes are launched without the namespace which causes an error as the config file is parsed with the namespace in mind.

By adding
`node_namespace=namespace`
to the node definitions this can be easily fixed and the node is now started in the correct namespace which allows it to read the parameters correctly.

This affects all launch files in:
`nav2_brinup/bringup/launch`

SteveMacenski, 2020-05-11:
Please review this ROS2 tutorial https://index.ros.org/doc/ros2/Tutorials/Node-arguments/

mvoellmy, 2020-05-12:
Hi @SteveMacenski,
This tutorial uses `ros2 run` only, while I'm using `ros2 launch`. I anyway tried using `--ros-args` but it seems to only work with `ros2 run`. Am I missing something?

I was trying to set the namespace either in the command line as described above or as the default value in the launch file. With the modification proposed, both of these methods work however.

SteveMacenski, 2020-05-12:
@mvoellmy can you try if you add a `DeclareLaunchArgument` in the launch file for namespace? Maybe we haven't enabled that. https://answers.ros.org/question/322874/ros2-what-is-different-between-declarelaunchargument-and-launchconfiguration/ this answer makes me think that none of our launch files will properly accept just those args without the `--ros-args` prefix.

mvoellmy, 2020-05-13:
@SteveMacenski
I just noticed, that my fork was based on the eloquent branch (since I was doing the turtlebot tutorial) and that the namespacing problem was already addressed in the master branch. My bad. I guess this can be closed here.

It was solved using a `GroupAction` and `PushRosNamespace`.
See [bringup_launch.py](https://github.com/ros-planning/navigation2/blob/master/nav2_bringup/bringup/launch/bringup_launch.py)
```
    bringup_cmd_group = GroupAction([
        PushRosNamespace(
            condition=IfCondition(use_namespace),
            namespace=namespace),

        IncludeLaunchDescription(
            PythonLaunchDescriptionSource(os.path.join(launch_dir,
                                                       'localization_launch.py')),
            launch_arguments={'namespace': namespace,
                              'map': map_yaml_file,
                              'use_sim_time': use_sim_time,
                              'autostart': autostart,
                              'params_file': params_file,
                              'use_lifecycle_mgr': 'false'}.items()),

        IncludeLaunchDescription(
            PythonLaunchDescriptionSource(os.path.join(launch_dir, 'navigation_launch.py')),
            launch_arguments={'namespace': namespace,
                              'use_sim_time': use_sim_time,
                              'autostart': autostart,
                              'params_file': params_file,
                              'bt_xml_file': bt_xml_file,
                              'use_lifecycle_mgr': 'false',
                              'map_subscribe_transient_local': 'true'}.items()),
    ])

```

closed 2020-05-13 by mvoellmy