ros-navigation/navigation2 #1726: Use Parameterized frame_id

ymd-stella, 2020-05-14:
## Feature request

#### Feature description
Use parameterized frame_id.

#### Implementation considerations
Modification is needed for
- https://github.com/ros-planning/navigation2/blob/13633cd3efdaf2a206bb9df0f76e4f5e8e996d2b/nav2_recoveries/src/recovery_server.cpp#L75
  - Using odom and base_link
- https://github.com/ros-planning/navigation2/blob/13633cd3efdaf2a206bb9df0f76e4f5e8e996d2b/nav2_recoveries/plugins/back_up.cpp#L56-L70
  - Using odom and base_link
- https://github.com/ros-planning/navigation2/blob/13633cd3efdaf2a206bb9df0f76e4f5e8e996d2b/nav2_recoveries/plugins/spin.cpp#L73-L94
  - Using odom and base_link
- https://github.com/ros-planning/navigation2/blob/13633cd3efdaf2a206bb9df0f76e4f5e8e996d2b/nav2_costmap_2d/src/collision_checker.cpp#L176
  - Using map and base_link
  - I think that costmap_.getRobotPose should be used.
- https://github.com/ros-planning/navigation2/blob/13633cd3efdaf2a206bb9df0f76e4f5e8e996d2b/nav2_behavior_tree/plugins/condition/goal_reached_condition.cpp#L73
  - Using map and base_link

SteveMacenski, 2020-05-14:
Can you be explicit about what you want to have changed? I don't understand what the ask / issue is. 

The backup/spin should be in odom frame since we want continuous local coordinates as they're basic controllers. The rest that you mention either override the relevant names and if you check the source of `getCurrentPose`, it has implied frames that can be overridden if they're not the default frames.

ymd-stella, 2020-05-14:
For example, collision_checker.cpp uses a parameterized global frame `global_frame_`. I think `base_link`/`odom` should be parameterized in the same way and set as a parameter of ROS. This is essential if you are handling multiple robots.

SteveMacenski, 2020-05-14:
I'd consider a PR adding that capability, are you willing to make those changes?

ymd-stella, 2020-05-14:
I'm sorry, I'm not familiar with the ROS2 parameter settings, so I'd like to ask you to change them. If you are busy, I will consider a change.

SteveMacenski, 2020-05-14:
I will probably not get to this in any reasonable timeframe. I would appreciate you taking a stab at least some of them.

ymd-stella, 2020-05-14:
I will try to implement it.

ymd-stella, 2020-05-16:
I just realized that tf has been renamed in https://github.com/ros-planning/navigation2/blob/fc762f659d98ce4ca2ebc5d87e6fc556742ffc9a/nav2_bringup/nav2_gazebo_spawner/nav2_gazebo_spawner/nav2_gazebo_spawner.py#L87  
I'm curious as to which of the following approaches is more practical or preferable.
- A. Rename the topic from `/tf` to a robot-specific name, as in the above code.
- B. Rename frame_id from `odom` to `robot/odom` and from `base_link` to `robot/base_link`.

ymd-stella, 2020-05-16:
For instance, I think the inability to lookupTransform from robot1 to robot2 is a drawback of A.

SteveMacenski, 2020-05-16:
I think that is very application dependent and I can't give general advise on it.

ymd-stella, 2020-05-16:
Isn't there any particular reason why the navigation2 demo uses the A approach?

PR #1742 (2020-05-19): Parameterize frame IDs

---

## Basic Info

Original PR:  #1729

| Info | Please fill out this column |
| ------ | ----------- |
| Ticket(s) this addresses   | closes  #1726 |
| Primary OS tested on | Ubuntu 18.04 |
closed 2020-05-18 by SteveMacenski