ros-navigation/navigation2 #1727: Controller_server and planner_server dies during execution

charlotteheggem, 2020-05-14:
- Operating System: Ubuntu 18.04
- Version or commit hash:  master downloaded from source 08.04.2020
- ROS distro: Eloquent
- DDS implementation: Fast-RTPS (default)

Hi. When launching Navigation2 with the attached parameter file, the controller_server and planner_server dies during the execution due to the following errors: 

Error1:

`
[controller_server-4] double free or corruption (fasttop)
[ERROR] [controller_server-4]: process has died [pid 19044, exit code -6, cmd '/home/ninamwa/navigation2/install/nav2_controller/lib/nav2_controller/controller_server --ros-args --params-file /tmp/tmprvckenvo -r /tf:=tf -r /tf_static:=tf_static'].
`
Error2:

`
[controller_server-4] [INFO] [controller_server]: Passing new path to controller.
[controller_server-4] [INFO] [controller_server]: Passing new path to controller.
[controller_server-4] [INFO] [controller_server]: Passing new path to controller.
[controller_server-4] [INFO] [controller_server]: Passing new path to controller.
[controller_server-4] [INFO] [controller_server]: Passing new path to controller.
[controller_server-4] [INFO] [controller_server]: Passing new path to controller.
[controller_server-4] [INFO] [controller_server]: Passing new path to controller.
[controller_server-4] [INFO] [controller_server]: Passing new path to controller.
[controller_server-4] [INFO] [controller_server]: Passing new path to controller.
[controller_server-4] [INFO] [controller_server]: Passing new path to controller.
[controller_server-4] [INFO] [controller_server]: Passing new path to controller.
[controller_server-4] malloc_consolidate(): invalid chunk size
[ERROR] [controller_server-4]: process has died [pid 21673, exit code -6, cmd '/home/ninamwa/navigation2/install/nav2_controller/lib/nav2_controller/controller_server --ros-args --params-file /tmp/tmpqtefwy86 -r /tf:=tf -r /tf_static:=tf_static'].
`

Error 3:

`
[planner_server-5] double free or corruption (fasttop)
[ERROR] [planner_server-5]: process has died [pid 25477, exit code -6, cmd '/home/ninamwa/navigation2/install/nav2_planner/lib/nav2_planner/planner_server --ros-args -r __node:=planner_server --params-file /tmp/tmpagidbqgn -r /tf:=tf -r /tf_static:=tf_static'].
`

This happens about every time, and the robot almost never reaches the goal because one of these errors occur (not the same every time). It happens both with the real robot and in simulation (gazebo).
Is there anything with the parameters which overload the system or what could cause the problem? It is really frustrating having to restart Navigation2 all the time. I tried downloading the master branch from source yesterday, but got a build error.
Thanks. 

`
# Parameters:
	amcl:
	  ros__parameters:
	    use_sim_time: False
	    alpha1: 0.2
	    alpha2: 0.2
	    alpha3: 0.2
	    alpha4: 0.2
	    alpha5: 0.2
	    base_frame_id: "base_footprint"
	    beam_skip_distance: 0.5
	    beam_skip_error_threshold: 0.9
	    beam_skip_threshold: 0.3
	    do_beamskip: false
	    global_frame_id: "map"
	    lambda_short: 0.1
	    laser_likelihood_max_dist: 2.0
	    laser_max_range: 100.0
	    laser_min_range: -1.0
	    laser_model_type: "likelihood_field"
	    max_beams: 60
	    max_particles: 2000
	    min_particles: 500
	    odom_frame_id: "odom"
	    pf_err: 0.05
	    pf_z: 0.99
	    recovery_alpha_fast: 0.0
[... 229 lines cut ...]
	    use_sim_time: False

	robot_state_publisher:
	  ros__parameters:
	    use_sim_time: False

	recoveries_server:
	  ros__parameters:
	    costmap_topic: local_costmap/costmap_raw
	    cycle_frequency: 10.0
	    footprint_topic: local_costmap/published_footprint
	    max_rotational_vel: 0.1
	    min_rotational_vel: 0.05
	    plugin_names:
	    - Spin
	    - BackUp
	    - Wait
	    plugin_types:
	    - nav2_recoveries/Spin
	    - nav2_recoveries/BackUp
	    - nav2_recoveries/Wait
	    rotational_acc_lim: 0.2
	    simulate_ahead_time: 2.0
	    use_sim_time: False
`

SteveMacenski, 2020-05-14:
Please provide an actual commit hash of your last build. 

The ROS2 master branch is what master is built on. If you’re using eloquent binaries, you should use the eloquent branch but you will lose certain new functionality.

SteveMacenski, 2020-05-14:
Also, how soon after starting did these happen for you?

charlotteheggem, 2020-05-15:
Hi, thank you for the follow up. Due to the covid19 situation we have limited access to the robot labratorium, and have not been able to test any further. We will continue the work on monday and get back to you with the full terminal output.

SteveMacenski, 2020-05-15:
Got it. I suspect it has to do with your install / mismatched versions / using a really old version / error on startup you're just not noticing. If this were happening, we'd be having CI crash and other users reporting it (we run 10 minutes full simulation in CI and we have about 20 developers running master).

There also might be a real issue, in that case we should figure out where this is happening with a traceback

daisukes, 2020-05-18:
Hi, I got the same "double free or corruption" error (1 and 3) with eloquent and not with foxy.
I think this is related to the rclcpp bug which is already fixed but unfortunately is not included in the eloquent release.
https://github.com/ros2/rclcpp/issues/938

I have never got the error 2 but it could be related to
https://github.com/ros2/rosidl/issues/490

I hope my experience could help.

SteveMacenski, 2020-05-18:
I run eloquent binaries locally (or rather, I did until a couple weeks ago when API breaking changes happened) and never saw this with any of my local development - so I'd be surprised if it was only due to that. 

Checking with master/foxy would be a good second step, but running GDB with it and figuring out who's crashing is a good starting point, to know if its something we can change in ROS2-ecosystem-land or something in ROS2-land.

charlotteheggem, 2020-05-25:
Hi, sorry for the delay. The problem occur after an initial pose and goal pose is set, and the robot is navigating towards the goal. It doesn't happen at every execution, but often enough. The errors are observed both in simulation and with real execution. We tried, as you suggested, to install the Eloquent binaries, but the problems still occur. Below are three terminal outputs of the commonly observed errors. The errors from the master-branch installation are shown in 1. Output and 2. Output. The terminal output for the Eloquent branch is shown in 3. Output (next comment)

<details>
  <summary>1. Output:</summary>
  
> ninamwa@ninamwa-X550JX:~$ ros2 launch kmr_navigation2 navigation2.launch.py
> [INFO] [launch]: All log files can be found below /home/ninamwa/.ros/log/2020-05-25-14-12-42-067135-ninamwa-X550JX-13739
> [INFO] [launch]: Default logging verbosity is set to INFO
> [INFO] [map_server-1]: process started with pid [13757]
> [INFO] [amcl-2]: process started with pid [13758]
> [INFO] [lifecycle_manager-3]: process started with pid [13759]
> [INFO] [controller_server-4]: process started with pid [13760]
> [INFO] [planner_server-5]: process started with pid [13761]
> [INFO] [recoveries_server-6]: process started with pid [13762]
> [INFO] [bt_navigator-7]: process started with pid [13765]
> [INFO] [waypoint_follower-8]: process started with pid [13775]
> [INFO] [lifecycle_manager-9]: process started with pid [13783]
> [INFO] [static_transform_publisher-10]: process started with pid [13791]
> [INFO] [robot_state_publisher-11]: process started with pid [13808]
> [INFO] [rviz2-12]: process started with pid [13816]
> [map_server-1] [INFO] [map_server]: 
> [map_server-1] 	map_server lifecycle node launched. 
> [map_server-1] 	Waiting on external lifecycle transitions to activate
> [map_server-1] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
> [map_server-1] [INFO] [map_server]: Creating
> [map_server-1] [INFO] [map_server]: Configuring
> [map_server-1] [INFO] [map_server]: OccGridLoader: Creating
> [map_server-1] [INFO] [map_server]: OccGridLoader: Configuring
> [map_server-1] [INFO] [map_server]: Loading yaml file: /home/ninamwa/kuka-ELO/install/kmr_navigation2/share/kmr_navigation2/map/carto_map.yaml
> [map_server-1] [INFO] [map_server]: Loading image_file: /home/ninamwa/kuka-ELO/src/kmr_navigation2/map/carto_map.pgm
> [amcl-2] [INFO] [amcl]: 
> [amcl-2] 	amcl lifecycle node launched. 
> [amcl-2] 	Waiting on external lifecycle transitions to activate
> [amcl-2] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
> [amcl-2] [INFO] [amcl]: Creating
> [lifecycle_manager-3] [INFO] [lifecycle_manager_localization]: Creating
> [lifecycle_manager-3] [INFO] [lifecycle_manager_localization]: Creating and initializing lifecycle service clients
> [lifecycle_manager-3] [INFO] [lifecycle_manager_localization]: Starting managed nodes bringup...
> [lifecycle_manager-3] [INFO] [lifecycle_manager_localization]: Configuring map_server
> [controller_server-4] [INFO] [controller_server]: 
> [controller_server-4] 	controller_server lifecycle node launched. 
> [controller_server-4] 	Waiting on external lifecycle transitions to activate
> [controller_server-4] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
> [recoveries_server-6] [INFO] [recoveries_server]: 
> [recoveries_server-6] 	recoveries_server lifecycle node launched. 
> [recoveries_server-6] 	Waiting on external lifecycle transitions to activate
> [recoveries_server-6] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
> [lifecycle_manager-3] [INFO] [lifecycle_manager_localization]: Configuring amcl
> [amcl-2] [INFO] [amcl]: Configuring
> [amcl-2] [INFO] [amcl]: initTransforms
> [static_transform_publisher-10] [INFO] [static_transform_publisher]: Spinning until killed publishing base_footprint to base_link
> [controller_server-4] [INFO] [controller_server]: Creating controller server
> [lifecycle_manager-9] [INFO] [lifecycle_manager_navigation]: Creating
> [robot_state_publisher-11] Initialize urdf model from file: /home/ninamwa/kuka-ELO/install/kmr_bringup/share/kmr_bringup/urdf/kmriiwa_withcameraframes.urdf
> [robot_state_publisher-11] Parsing robot urdf xml string.
> [amcl-2] [INFO] [amcl]: initPubSub
> [robot_state_publisher-11] Link base_link had 7 children
> [amcl-2] [INFO] [amcl]: Subscribed to map topic.
> [robot_state_publisher-11] Link back_left_wheel_link had 0 children
> [robot_state_publisher-11] Link back_right_wheel_link had 0 children
> [robot_state_publisher-11] Link base_link_middle had 1 children
> [robot_state_publisher-11] Link base_link_top had 4 children
> [robot_state_publisher-11] Link base_iiwa had 1 children
> [robot_state_publisher-11] Link link_1 had 1 children
> [robot_state_publisher-11] Link link_2 had 1 children
> [robot_state_publisher-11] Link link_3 had 1 children
> [robot_state_publisher-11] Link link_4 had 1 children
> [robot_state_publisher-11] Link link_5 had 1 children
> [robot_state_publisher-11] Link link_6 had 1 children
> [robot_state_publisher-11] Link link_7 had 2 children
[... 739 lines cut ...]
> [amcl-2] [INFO] [amcl_rclcpp_node]: Message Filter dropping message: frame 'scan' at time 1590408310.633 for reason 'Unknown'
> [amcl-2] [INFO] [amcl_rclcpp_node]: Message Filter dropping message: frame 'scan' at time 1590408310.635 for reason 'Unknown'
[... cut ...]
> [controller_server-4] [INFO] [controller_server]: Passing new path to controller.
> [planner_server-5] malloc_consolidate(): invalid chunk size
> [controller_server-4] [INFO] [controller_server]: Passing new path to controller.
> [ERROR] [planner_server-5]: process has died [pid 12051, exit code -6, cmd '/home/ninamwa/navigation2/install/nav2_planner/lib/nav2_planner/planner_server --ros-args -r __node:=planner_server --params-file /tmp/tmpwvrrrcwo -r /tf:=tf -r /tf_static:=tf_static'].
> 
</details>

charlotteheggem, 2020-05-25:
<details>
  <summary>3. Output: </summary>

> kmriiwa@kmriiwa-NUC8i7HNKQC:~$ ros2 launch kmr_navigation2 navigation2.launch.py[INFO] [launch]: All log files can be found below /home/kmriiwa/.ros/log/2020-05-25-18-38-37-888820-kmriiwa-NUC8i7HNKQC-9816
> [INFO] [launch]: Default logging verbosity is set to INFO
> [INFO] [lifecycle_manager-1]: process started with pid [9834]
> [INFO] [map_server-2]: process started with pid [9835]
> [INFO] [amcl-3]: process started with pid [9836]
> [INFO] [controller_server-4]: process started with pid [9837]
> [INFO] [planner_server-5]: process started with pid [9838]
> [INFO] [recoveries_server-6]: process started with pid [9839]
> [INFO] [bt_navigator-7]: process started with pid [9848]
> [INFO] [waypoint_follower-8]: process started with pid [9877]
> [INFO] [static_transform_publisher-9]: process started with pid [9878]
> [INFO] [robot_state_publisher-10]: process started with pid [9882]
> [INFO] [rviz2-11]: process started with pid [9894]
> [lifecycle_manager-1] [INFO] [lifecycle_manager]: Creating
> [lifecycle_manager-1] [INFO] [lifecycle_manager]: Creating and initializing lifecycle service clients
> [lifecycle_manager-1] [INFO] [lifecycle_manager]: Starting managed nodes bringup...
> [lifecycle_manager-1] [INFO] [lifecycle_manager]: Configuring map_server
> [map_server-2] [INFO] [map_server]: Creating
> [amcl-3] [INFO] [amcl]: Creating
> [planner_server-5] [INFO] [planner_server]: Creating
> [controller_server-4] [INFO] [controller_server]: Creating controller server
> [planner_server-5] [INFO] [global_costmap.global_costmap]: Creating Costmap
> [robot_state_publisher-10] Initialize urdf model from file: /home/kmriiwa/kmriiwa_ws/install/kmr_bringup/share/kmr_bringup/urdf/kmriiwa_withcameraframes.urdf
> [robot_state_publisher-10] Parsing robot urdf xml string.
> [map_server-2] [INFO] [map_server]: Configuring
> [map_server-2] [INFO] [map_server]: OccGridLoader: Creating
> [map_server-2] [INFO] [map_server]: OccGridLoader: Configuring
> [map_server-2] [INFO] [map_server]: Loading yaml file: /home/kmriiwa/kmriiwa_ws/install/kmr_navigation2/share/kmr_navigation2/map/playpen_map.yaml
> [map_server-2] [INFO] [map_server]: Loading image_file: /home/kmriiwa/kmriiwa_ws/src/kmr_navigation2/map/playpen_map.pgm
> [robot_state_publisher-10] Link base_link had 7 children
> [robot_state_publisher-10] Link back_left_wheel_link had 0 children
> [robot_state_publisher-10] Link back_right_wheel_link had 0 children
> [robot_state_publisher-10] Link base_link_middle had 1 children
> [robot_state_publisher-10] Link base_link_top had 4 children
> [robot_state_publisher-10] Link base_iiwa had 1 children
> [robot_state_publisher-10] Link link_1 had 1 children
> [robot_state_publisher-10] Link link_2 had 1 children
> [robot_state_publisher-10] Link link_3 had 1 children
> [robot_state_publisher-10] Link link_4 had 1 children
> [robot_state_publisher-10] Link link_5 had 1 children
> [robot_state_publisher-10] Link link_6 had 1 children
> [robot_state_publisher-10] Link link_7 had 2 children
> [robot_state_publisher-10] Link adapter had 2 children
> [robot_state_publisher-10] Link d435_manipulator_link had 0 children
> [robot_state_publisher-10] Link gripper_base_link had 5 children
> [robot_state_publisher-10] Link left_outer_knuckle had 1 children
> [robot_state_publisher-10] Link left_outer_finger had 1 children
> [robot_state_publisher-10] Link left_inner_finger had 1 children
> [robot_state_publisher-10] Link left_inner_finger_pad had 0 children
> [robot_state_publisher-10] Link gripper_middle_point had 0 children
> [robot_state_publisher-10] Link left_inner_knuckle had 0 children
> [robot_state_publisher-10] Link right_inner_knuckle had 0 children
> [robot_state_publisher-10] Link right_outer_knuckle had 1 children
> [robot_state_publisher-10] Link right_outer_finger had 1 children
> [robot_state_publisher-10] Link right_inner_finger had 1 children
> [robot_state_publisher-10] Link right_inner_finger_pad had 0 children
> [robot_state_publisher-10] Link tool0 had 0 children
> [robot_state_publisher-10] Link d435_base_front_link had 2 children
> [robot_state_publisher-10] Link camera1_color_optical_frame_link had 0 children
> [robot_state_publisher-10] Link camera1_depth_optical_frame_link had 0 children
> [robot_state_publisher-10] Link metall_plate_left_link had 1 children
> [robot_state_publisher-10] Link d435_base_left_link had 2 children
> [robot_state_publisher-10] Link camera3_color_optical_frame_link had 0 children
> [robot_state_publisher-10] Link camera3_depth_optical_frame_link had 0 children
> [robot_state_publisher-10] Link metall_plate_right_link had 1 children
> [robot_state_publisher-10] Link d435_base_right_link had 2 children
> [robot_state_publisher-10] Link camera2_color_optical_frame_link had 0 children
[... 1657 lines cut ...]
> [planner_server-5] [INFO] [global_costmap.global_costmap_rclcpp_node]: Message Filter dropping message: frame 'laser_B4_link' at time 3907.080 for reason 'Unknown'
> [controller_server-4] [INFO] [controller_server]: Preempting the goal. Passing the new path to the planner.
> [controller_server-4] [INFO] [controller_server]: Preempting the goal. Passing the new path to the planner.
> [controller_server-4] [INFO] [controller_server]: Preempting the goal. Passing the new path to the planner.
> [controller_server-4] [INFO] [controller_server]: Preempting the goal. Passing the new path to the planner.
[... cut ...]
> [controller_server-4] [INFO] [controller_server]: Preempting the goal. Passing the new path to the planner.
> [controller_server-4] [INFO] [controller_server]: Preempting the goal. Passing the new path to the planner.
> [controller_server-4] double free or corruption (out)
> [ERROR] [controller_server-4]: process has died [pid 9837, exit code -6, cmd '/home/kmriiwa/ros2/install/nav2_controller/lib/nav2_controller/controller_server --ros-args --params-file /tmp/tmp8tb0wiy7        '].
>
</details>

SteveMacenski, 2020-05-26:
@charlotteheggem we need a traceback, I don't see anything in logs that hits me immediately as a problem. Given that you see 3 different crashes with the multiple of the same setups as I (and all of the developers on this project) test with, I think the issue is likely to be in your system rather than this codebase, but lets work to figure out why just to make sure. The first step is to run GDB with this so that when it crashes you can get a traceback of what happened and where. Once we have that, we can go from there.

I suspect that you'll find that its something unrelated to Navigation2, I suspect TF, since the only thing I can think of that links those 3 servers together is TF. It may be that your master builds are old and you need to pull in new ones. In fact, if you're working with our current master branch, you couldn't compile it without updates because of some `rclcpp` namespace updates.

charlotteheggem, 2020-05-27:
Hi,
The outputs when running with GDB is shown below. Both outputs are obtained with the Eloquent installation, but was obtained at two different computers.

1. Output:
PC: Intel® NUC8i7HNKQC (https://www.newegg.com/intel-nuc-8-business-boxnuc8i7hnkqc1/p/3C6-008J-00004)

- Memory : 15.6 GiB RAM
- Processor: Intel® Core™ i7-8705G CPU @ 3.10GHz × 8 
- Ubuntu 18.04.4 LTS 64-bit 

2. Output:
PC: ASUS X550JX
- Memory: 8 GB RAM
- Processor: Intel Core i7-4720HQ CPU: @ 2.6 GHz x 4
- Ubuntu 18.04.4 LTS 64-bit 

Both the ROS computer and the mobile robot is running over WIFI on a ASUS 5G network. (https://www.asus.com/Networking/RTAC51U/)

<details>
  <summary>GDB output 1</summary>
> 
> kmriiwa@kmriiwa-NUC8i7HNKQC:~$ ros2 launch kmr_navigation2 navigation2.launch.py launch-prefix:="gdb -ex run --args" 
> [INFO] [launch]: All log files can be found below /home/kmriiwa/.ros/log/2020-05-27-08-53-12-158193-kmriiwa-NUC8i7HNKQC-32115
> [INFO] [launch]: Default logging verbosity is set to INFO
> [INFO] [lifecycle_manager-1]: process started with pid [32133]
> [INFO] [map_server-2]: process started with pid [32134]
> [INFO] [amcl-3]: process started with pid [32135]
> [INFO] [controller_server-4]: process started with pid [32136]
> [INFO] [planner_server-5]: process started with pid [32137]
> [INFO] [recoveries_server-6]: process started with pid [32138]
> [INFO] [bt_navigator-7]: process started with pid [32139]
> [INFO] [waypoint_follower-8]: process started with pid [32140]
> [INFO] [static_transform_publisher-9]: process started with pid [32141]
> [INFO] [robot_state_publisher-10]: process started with pid [32142]
> [INFO] [rviz2-11]: process started with pid [32143]
> [map_server-2] GNU gdb (Ubuntu 8.1-0ubuntu3.2) 8.1.0.20180409-git
> [map_server-2] Copyright (C) 2018 Free Software Foundation, Inc.
> [map_server-2] License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
> [map_server-2] This is free software: you are free to change and redistribute it.
> [map_server-2] There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
> [map_server-2] and "show warranty" for details.
> [map_server-2] This GDB was configured as "x86_64-linux-gnu".
> [map_server-2] Type "show configuration" for configuration details.
> [map_server-2] For bug reporting instructions, please see:
> [map_server-2] <http://www.gnu.org/software/gdb/bugs/>.
> [map_server-2] Find the GDB manual and other documentation resources online at:
> [map_server-2] <http://www.gnu.org/software/gdb/documentation/>.
> [map_server-2] For help, type "help".
> [map_server-2] Type "apropos word" to search for commands related to "word"...
> [lifecycle_manager-1] GNU gdb (Ubuntu 8.1-0ubuntu3.2) 8.1.0.20180409-git
> [lifecycle_manager-1] Copyright (C) 2018 Free Software Foundation, Inc.
> [lifecycle_manager-1] License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
> [lifecycle_manager-1] This is free software: you are free to change and redistribute it.
> [lifecycle_manager-1] There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
> [lifecycle_manager-1] and "show warranty" for details.
> [lifecycle_manager-1] This GDB was configured as "x86_64-linux-gnu".
> [lifecycle_manager-1] Type "show configuration" for configuration details.
> [lifecycle_manager-1] For bug reporting instructions, please see:
> [lifecycle_manager-1] <http://www.gnu.org/software/gdb/bugs/>.
> [lifecycle_manager-1] Find the GDB manual and other documentation resources online at:
> [lifecycle_manager-1] <http://www.gnu.org/software/gdb/documentation/>.
> [lifecycle_manager-1] For help, type "help".
> [lifecycle_manager-1] Type "apropos word" to search for commands related to "word"...
> [lifecycle_manager-1] Reading symbols from /home/kmriiwa/ros2/install/nav2_lifecycle_manager/lib/nav2_lifecycle_manager/lifecycle_manager...done.
> [map_server-2] Reading symbols from /home/kmriiwa/ros2/install/nav2_map_server/lib/nav2_map_server/map_server...done.
> [map_server-2] Starting program: /home/kmriiwa/ros2/install/nav2_map_server/lib/nav2_map_server/map_server --ros-args -r __node:=map_server --params-file /tmp/tmprlfe6z5z '' '' '' '' '' '' '' ''
> [amcl-3] GNU gdb (Ubuntu 8.1-0ubuntu3.2) 8.1.0.20180409-git
> [amcl-3] Copyright (C) 2018 Free Software Foundation, Inc.
> [amcl-3] License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
> [amcl-3] This is free software: you are free to change and redistribute it.
[... 1587 lines cut ...]
> [controller_server-4] [INFO] [local_costmap.local_costmap_rclcpp_node]: Message Filter dropping message: frame 'scan_2' at time 1590562533.496 for reason 'Unknown'
> [controller_server-4] [INFO] [local_costmap.local_costmap_rclcpp_node]: Message Filter dropping message: frame 'scan' at time 1590562533.496 for reason 'Unknown'
> [controller_server-4] [INFO] [local_costmap.local_costmap_rclcpp_node]: Message Filter dropping message: frame 'scan_2' at time 1590562533.497 for reason 'Unknown'
> [controller_server-4] [INFO] [local_costmap.local_costmap_rclcpp_node]: Message Filter dropping message: frame 'scan' at time 1590562533.497 for reason 'Unknown'
> [controller_server-4] [INFO] [local_costmap.local_costmap_rclcpp_node]: Message Filter dropping message: frame 'scan_2' at time 1590562533.497 for reason 'Unknown'
[... cut ...]
> [controller_server-4] [INFO] [local_costmap.local_costmap_rclcpp_node]: Message Filter dropping message: frame 'scan_2' at time 1590562535.581 for reason 'Unknown'
> [controller_server-4] [INFO] [local_costmap.local_costmap_rclcpp_node]: Message Filter dropping message: frame 'scan' at time 1590562538.983 for reason 'Unknown'
> [controller_server-4] [INFO] [local_costmap.local_costmap_rclcpp_node]: Message Filter dropping message: frame 'scan_2' at time 1590562536.373 for reason 'Unknown'
>
</details>

charlotteheggem, 2020-05-27:
<details>
  <summary>GDB output 2</summary>

> ninamwa@ninamwa-X550JX:~$ ros2 launch kmr_navigation2 navigation2.launch.py
> [INFO] [launch]: All log files can be found below /home/ninamwa/.ros/log/2020-05-27-09-59-53-135339-ninamwa-X550JX-15806
> [INFO] [launch]: Default logging verbosity is set to INFO
> [INFO] [lifecycle_manager-1]: process started with pid [15824]
> [INFO] [map_server-2]: process started with pid [15825]
> [INFO] [amcl-3]: process started with pid [15826]
> [INFO] [controller_server-4]: process started with pid [15827]
> [INFO] [planner_server-5]: process started with pid [15828]
> [INFO] [recoveries_server-6]: process started with pid [15830]
> [INFO] [bt_navigator-7]: process started with pid [15838]
> [INFO] [waypoint_follower-8]: process started with pid [15850]
> [INFO] [static_transform_publisher-9]: process started with pid [15858]
> [INFO] [robot_state_publisher-10]: process started with pid [15863]
> [INFO] [rviz2-11]: process started with pid [15864]
> [lifecycle_manager-1] [INFO] [lifecycle_manager]: Creating
> [lifecycle_manager-1] [INFO] [lifecycle_manager]: Creating and initializing lifecycle service clients
> [lifecycle_manager-1] [INFO] [lifecycle_manager]: Starting managed nodes bringup...
> [lifecycle_manager-1] [INFO] [lifecycle_manager]: Configuring map_server
> [map_server-2] [INFO] [map_server]: Creating
> [map_server-2] [INFO] [map_server]: Configuring
> [map_server-2] [INFO] [map_server]: OccGridLoader: Creating
> [map_server-2] [INFO] [map_server]: OccGridLoader: Configuring
> [map_server-2] [INFO] [map_server]: Loading yaml file: /home/ninamwa/kuka-ELO/install/kmr_navigation2/share/kmr_navigation2/map/carto_map.yaml
> [map_server-2] [INFO] [map_server]: Loading image_file: /home/ninamwa/kuka-ELO/install/kmr_navigation2/share/kmr_navigation2/map/carto_map.pgm
> [lifecycle_manager-1] [INFO] [lifecycle_manager]: Configuring amcl
> [robot_state_publisher-10] Initialize urdf model from file: /home/ninamwa/kuka-ELO/install/kmr_bringup/share/kmr_bringup/urdf/kmriiwa.urdf
> [robot_state_publisher-10] Parsing robot urdf xml string.
> [robot_state_publisher-10] Link base_link had 7 children
> [robot_state_publisher-10] Link back_left_wheel_link had 0 children
> [robot_state_publisher-10] Link back_right_wheel_link had 0 children
> [robot_state_publisher-10] Link base_link_middle had 1 children
> [robot_state_publisher-10] Link base_link_top had 4 children
> [robot_state_publisher-10] Link base_iiwa had 1 children
> [robot_state_publisher-10] Link link_1 had 1 children
> [robot_state_publisher-10] Link link_2 had 1 children
> [robot_state_publisher-10] Link link_3 had 1 children
> [robot_state_publisher-10] Link link_4 had 1 children
> [robot_state_publisher-10] Link link_5 had 1 children
> [robot_state_publisher-10] Link link_6 had 1 children
> [robot_state_publisher-10] Link link_7 had 2 children
> [robot_state_publisher-10] Link adapter had 2 children
> [robot_state_publisher-10] Link d435_manipulator_link had 0 children
> [robot_state_publisher-10] Link gripper_base_link had 5 children
> [robot_state_publisher-10] Link left_outer_knuckle had 1 children
> [robot_state_publisher-10] Link left_outer_finger had 1 children
> [robot_state_publisher-10] Link left_inner_finger had 1 children
> [robot_state_publisher-10] Link left_inner_finger_pad had 0 children
> [robot_state_publisher-10] Link gripper_middle_point had 0 children
> [robot_state_publisher-10] Link left_inner_knuckle had 0 children
> [robot_state_publisher-10] Link right_inner_knuckle had 0 children
> [robot_state_publisher-10] Link right_outer_knuckle had 1 children
> [robot_state_publisher-10] Link right_outer_finger had 1 children
> [robot_state_publisher-10] Link right_inner_finger had 1 children
> [robot_state_publisher-10] Link right_inner_finger_pad had 0 children
> [robot_state_publisher-10] Link tool0 had 0 children
> [robot_state_publisher-10] Link d435_base_front_link had 1 children
> [robot_state_publisher-10] Link d435_base_front_depth_frame had 1 children
> [robot_state_publisher-10] Link d435_base_front_depth_optical_frame had 0 children
> [robot_state_publisher-10] Link metall_plate_left_link had 1 children
> [robot_state_publisher-10] Link d435_base_left_link had 0 children
> [robot_state_publisher-10] Link metall_plate_right_link had 1 children
> [robot_state_publisher-10] Link d435_base_right_link had 0 children
> [robot_state_publisher-10] Link front_left_wheel_link had 0 children
> [robot_state_publisher-10] Link front_right_wheel_link had 0 children
> [robot_state_publisher-10] Link laser_B1_link had 0 children
> [robot_state_publisher-10] Link laser_B4_link had 0 children
> [robot_state_publisher-10] got segment adapter
[... 2043 lines cut ...]
> [planner_server-5] [INFO] [global_costmap.global_costmap_rclcpp_node]: Message Filter dropping message: frame 'scan' at time 1590566655.365 for reason 'Unknown'
> [planner_server-5] [INFO] [global_costmap.global_costmap_rclcpp_node]: Message Filter dropping message: frame 'scan_2' at time 1590566655.372 for reason 'Unknown'
> [planner_server-5] [INFO] [global_costmap.global_costmap_rclcpp_node]: Message Filter dropping message: frame 'scan' at time 1590566655.385 for reason 'Unknown'
> [planner_server-5] [INFO] [global_costmap.global_costmap_rclcpp_node]: Message Filter dropping message: frame 'scan_2' at time 1590566655.412 for reason 'Unknown'
[... cut ...]
> [planner_server-5] [INFO] [global_costmap.global_costmap_rclcpp_node]: Message Filter dropping message: frame 'scan_2' at time 1590566655.534 for reason 'Unknown'
> [planner_server-5] [INFO] [global_costmap.global_costmap_rclcpp_node]: Message Filter dropping message: frame 'scan_2' at time 1590566655.567 for reason 'Unknown'
> [planner_server-5] [INFO] [global_costmap.global_costmap_rclcpp_node]: Message Filter dropping message: frame 'scan_2' at time 1590566655.605 for reason 'Unknown'
> [planner_server-5] [INFO] [global_costmap.global_

SteveMacenski, 2020-05-27:
What was the traceback though? Please see https://ftp.gnu.org/old-gnu/Manuals/gdb/html_node/gdb_42.html. In the GDB session, what I'm looking for is the actual traceback of the error to see where its coming from.

From your second console output: 

```
[controller_server-4]
[controller_server-4] >>> [rcutils|error_handling.c:107] rcutils_set_error_state()
[controller_server-4] This error state is being overwritten:
[controller_server-4]
[controller_server-4] 'jump callback was not found, at /tmp/binarydeb/ros-eloquent-rcl-0.8.4/src/rcl/time.c:445'
[controller_server-4]
[controller_server-4] with this new error message:
[controller_server-4]
[controller_server-4] 'callback/user_data are already added to this clock, at /tmp/binarydeb/ros-eloquent-rcl-0.8.4/src/rcl/time.c:403'
[controller_server-4]
[controller_server-4] rcutils_reset_error() should be called after error handling to avoid this.
[controller_server-4] <<<
[controller_server-4] [ERROR] [rclcpp]: Couldn't initialize rcl timer handle: callback/user_data are already added to this clock, at /tmp/binarydeb/ros-eloquent-rcl-0.8.4/src/rcl/time.c:403
[controller_server-4]
[controller_server-4]
[controller_server-4] Thread 24 "controller_serv" received signal SIGSEGV, Segmentation fault.
```

Its relatively clear, though we can't know 100% for sure without the traceback, your issue is deeper than navigation2.  Looks like issues with your clock jumping in rcl https://github.com/ros2/rcl/issues/293. Maybe you need to update eloquent? Are you resetting clocks / simulations or something?

charlotteheggem, 2020-05-28:
Hi, here are some of the tracebacks from today:

<details>
  <summary>traceback 1:</summary>

> Thread 10 "planner_server" received signal SIGSEGV, Segmentation fault.
> [Switching to Thread 0x7fffe6ffd700 (LWP 10114)]
> tcache_get (tc_idx=0) at malloc.c:2943
> 2943    malloc.c: No such file or directory.
> (gdb) backtrace
> #0  tcache_get (tc_idx=0) at malloc.c:2943
> #1  __GI___libc_malloc (bytes=17) at malloc.c:3050
> #2  0x00007ffff6ea0298 in operator new(unsigned long) ()
>    from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
> #3  0x00007ffff1914417 in tf2::displayTimePoint[abi:cxx11](std::chrono::time_point<std::chrono::_V2::system_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > > const&) () from /opt/ros/eloquent/lib/libtf2.so
> #4  0x00007ffff1903d9f in tf2::cache::createExtrapolationException2(std::chrono::time_point<std::chrono::_V2::system_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > >, std::chrono::time_point<std::chrono::_V2::system_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*) ()
>    from /opt/ros/eloquent/lib/libtf2.so
> #5  0x00007ffff19053ab in tf2::TimeCache::getParent(std::chrono::time_point<std::chrono::_V2::system_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*) () from /opt/ros/eloquent/lib/libtf2.so
> #6  0x00007ffff1912a54 in tf2::TF2Error tf2::BufferCore::walkToTopParent<tf2::CanTransformAccum>(tf2::CanTransformAccum&, std::chrono::time_point<std::chrono::_V2::system_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > >, unsigned int, unsigned int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*, std::vector<unsigned int, std::allocator<unsigned int> >*) const () from /opt/ros/eloquent/lib/libtf2.so
> 
</details>

<details>
  <summary>traceback 2:</summary>

> [INFO] [local_costmap.local_costmap_rclcpp_node]: Message Filter dropping message: frame 'scan_2' at time 1590649370.291 for reason 'Unknown'
> double free or corruption (fasttop)
> Thread 12 "controller_serv" received signal SIGABRT, Aborted.
> [Switching to Thread 0x7fffe5be8700 (LWP 11290)]
> __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
> 51      ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
> (gdb) backtrace
> #0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
> #1  0x00007ffff492c801 in __GI_abort () at abort.c:79
> #2  0x00007ffff4975897 in __libc_message (action=action@entry=do_abort, 
>     fmt=fmt@entry=0x7ffff4aa2b9a "%s\n") at ../sysdeps/posix/libc_fatal.c:181
> #3  0x00007ffff497c90a in malloc_printerr (
>     str=str@entry=0x7ffff4aa4828 "double free or corruption (fasttop)")
>     at malloc.c:5350
> #4  0x00007ffff497f1d5 in _int_free (av=0x7ffff4cd7c40 <main_arena>, 
>     p=0x555555c311d0, have_lock=<optimized out>) at malloc.c:4230
> #5  0x00007ffff498179e in _int_realloc (
>     av=av@entry=0x7ffff4cd7c40 <main_arena>, oldp=oldp@entry=0x555555c2c010, 
>     oldsize=oldsize@entry=20976, nb=nb@entry=20928) at malloc.c:4650
> #6  0x00007ffff4984f9b in __GI___libc_realloc (oldmem=0x555555c2c020, 
>     bytes=20920) at malloc.c:3230
> #7  0x00007ffff649e359 in rcl_clock_remove_jump_callback ()
>    from /opt/ros/eloquent/lib/librcl.so
> #8  0x00007ffff649fd72 in rcl_timer_fini ()
>    from /opt/ros/eloquent/lib/librcl.so
> #9  0x00007ffff796aac2 in ?? () from /opt/ros/eloquent/lib/librclcpp.so
> #10 0x00007ffff796ae59 in rclcpp::TimerBase::~TimerBase() ()
>    from /opt/ros/eloquent/lib/librclcpp.so
> #11 0x00007ffff31f8259 in ?? () from /opt/ros/eloquent/lib/libtf2_ros.so
> #12 0x00007ffff31f8722 in tf2_ros::CreateTimerROS::remove(unsigned long const&)
>
</details>

<details>
  <summary>traceback 3:</summary>

> [Thread 0x7fff79f98700 (LWP 17801) exited]
> double free or corruption (fasttop)
> Thread 36 "planner_server" received signal SIGABRT, Aborted.
> [Switching to Thread 0x7fff7b7fe700 (LWP 17768)]
> __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
> 51      ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
> (gdb) backtrace
> #0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
> #1  0x00007ffff6844801 in __GI_abort () at abort.c:79
[... 78 lines cut ...]
>    from /opt/ros/eloquent/lib/librcl.so
> (gdb) backtrace
> #0  0x00007ffff5243eb1 in rcl_timer_is_canceled ()
>    from /opt/ros/eloquent/lib/librcl.so
> #1  0x00007ffff52460ca in rcl_wait () from /opt/ros/eloquent/lib/librcl.so
> #2  0x00007ffff7662331 in rclcpp::executor::Executor::wait_for_work (
>     this=this@entry=0x555555d28108, timeout=timeout@entry=...)
>     at ./src/rclcpp/executor.cpp:470
> #3  0x00007ffff76627a5 in rclcpp::executor::Executor::get_next_executable (
>     this=this@entry=0x555555d28108, any_executable=..., 
>     timeout=timeout@entry=...) at ./src/rclcpp/executor.cpp:573
> #4  0x00007ffff7666710 in rclcpp::executors::SingleThreadedExecutor::spin (
>     this=0x555555d28108)
>     at ./src/rclcpp/executors/single_threaded_executor.cpp:35
> #5  0x00007ffff62fd43a in std::thread::_State_impl<std::thread::_Invoker<std::tuple<nav2_util::NodeThread::NodeThread(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>)::{lambda()#1}> > >::_M_run() ()
>    from /home/ninamwa/navigation2/install/nav2_util/lib/libnav2_util_core.so
> #6  0x00007ffff6eca6df in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
> #7  0x00007ffff73af6db in start_thread (arg=0x7fffa27fc700)
>     at pthread_create.c:463
> #8  0x00007ffff692588f in clone ()
>     at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
>
</details>

We agree that this does not seem to be due to Navigation2. We suspect that the TF data is published too slow compared to the sensor data. What rates should the scan data and odometry data ideally be published at?

SteveMacenski, 2020-05-28:
You have a bunch of issues with TF2 and clocks - my guess is your issue is actually with the clocks and its just being manifested in TF2 since it uses timing heavily. Publishing TF too slow would not cause a crash or a segfault. You'd get navigation level warnings about it but that's all. 

I think you should file tickets in the appropriate places from these tracebacks and see what they say. I don't know how you've installed ROS2, but you might have something out of sync and you need to pull in compliant versions.

charlotteheggem, 2020-05-28:
Thank you so much for using your time to look into this, it is really appreciated! :) We will hopefully sort this out soon. I am closing this issue as it is not related to Navigation2.

closed 2020-05-28 by charlotteheggem

ninamwa, 2020-06-04:
If someone else happens to run into the same problem, we solved it by using a different branch of rclcpp (https://github.com/ros2/rclcpp/issues/1144) :-)