ros-navigation/navigation2 #1737: Add developer process to getting started / getting involved sections

SteveMacenski, 2020-05-14:
We should add some developer FYIs to the website, since this seems to be coming up alot
- [ ] how to use docker containers for development
- [ ] how to run tests and display results (and interpret them)
- [ ] how to run linters
- [ ] Process when you start working on a ticket (comment that you're doing it / assign it) 
- [ ] meetings / schedule

Shru, 2020-05-14:
These are the starting points with regards to available documentation, correct? 
1. https://github.com/ros-planning/navigation2/blob/master/Dockerfile
2. https://navigation.ros.org/build_instructions/index.html#build
3. https://github.com/ros-planning/navigation2/tree/master/nav2_system_tests
4. Individual server/component tests such as: https://github.com/ros-planning/navigation2/tree/master/nav2_map_server/test

SteveMacenski, 2020-05-15:
Mostly just the getting started, but those other things could also be good references depending on what you're trying to do. The dockerfile is a good reference for figuring out the CLI for colcon I suppose, but there's a bunch of other stuff happening in those I wouldn't necessarily point anyone to that. One of the goals of an actual webpage would be to explicitly walk through these things instead of having people fish through code. 

Is that something you'd be open to working on? Basically explaining some of the basics to get up and running, colcon, building workspace (getting started helps with that) running tests / linters, process for working on tickets / submitting PRs? Shouldn't be that much work and could help you concretely formalize that knowledge you've gained and help others learn it!

Shru, 2020-05-18:
Thanks Steve. Yes, I will work on this. I will start documenting building Navigation 2 from master (ros2 workspace, Navigation 2 depends workspace, Navigation 2 workspace), some colcon basics, docker basics. Shall I make a github page and raise a PR? Would that be a good workflow?

SteveMacenski, 2020-05-18:
I think first outlining what youre goals are to accomplish so that we have a clear scope would be good. Then a bullet point outline of what you want your page(s) to talk about, in order. Then when you go to write it, its just executing on your plan and so we can discuss what belongs / doesn't / order before you go and write up stuff.

Shru, 2020-05-23:
@SteveMacenski That sounds good. Based on your feedback, please find below the outline for developer guide for Navigation 2. I have also used [ROS Developer Guide](https://index.ros.org/doc/ros2/Contributing/Developer-Guide/) as reference. 

Please share your feedback. If something is out of place/ needs to be reworked, please let me know and I will make changes accordingly. 
I have not yet mentioned the system tests, component tests, subsystem tests, unit tests. However, for each of these, I plan to mention:
a. Scope of each of those tests
b. How/Where are these tests are integrated in the code base?
c. How to run each of these tests?
d. How to interpret the test results. 

-----------------------------------------------PAGE1-------------------------------------------------------------

### Getting started with Navigation2:

**1. Prerequisites:**
Explain in the form of a table target/OS dependencies, ROS2 versions that Navigation 2 supports. 

**2. Install Navigation 2:**
Mention various available options in terms of installing latest stable release or building Navigation 2 from master.  
**2a.** Install latest stable release: list steps
**2b.** Build and install for a specific released ROS distribution: list steps
**2c.** Build and install from script: list steps
**2d.** Build and install manually from master: Complete 3a and refer to 3b for the steps. 

*Best Practice* - Latest OS, ROS2 Master, Navigation 2 Master (2d)

-----------------------------------------------PAGE2-------------------------------------------------------------

**3. Colcon build system and how to use it for Navigation 2 development?**
**3a.** Basics: Talk about workspaces in general, building packages. Some useful links on Colcon for ROS:
https://vimeopro.com/osrfoundation/roscon-2019/video/379127725
https://colcon.readthedocs.io/
https://design.ros2.org/articles/build_tool.html
https://index.ros.org/doc/ros2/Tutorials/Colcon-Tutorial/
**3b.** Create colcon workspaces and build both ROS2 and Navigation 2 manually from master: list steps.
(For missing packages, establish ROS dependencies and not install individual packages.)
**3c.** Colcon package testing:
Explain colcon test, colcon test-result, testing individual packages with ctest from Navigation 2 perspective.
**3d.** Building and testing troubleshooting: 
https://navigation.ros.org/build_instructions/build_docs/build_troubleshooting_guide.html#build-troubleshooting-guide

-----------------------------------------------PAGE3-------------------------------------------------------------

**4. Staying upto date with ROS and Navigation 2.**
https://index.ros.org/doc/ros2/Installation/Maintaining-a-Source-Checkout/#maintainingsource
*Best Practice* -

**5. Tutorials and Applications:**
https://github.com/ros-planning/navigation2/blob/master/nav2_bringup/bringup/README.md
https://navigation.ros.org/tutorials/index.html

-----------------------------------------------PAGE4-------------------------------------------------------------

**6. Using Docker for development:**
Explain in a few lines what docker is and how docker containers are useful in development process (faster builds, isolated environment, ability of have multiple dockers, etc)

Basic docker operations:
**6a.** Install docker: list steps
**6b.** Use an existing docker image in docker hub: list steps. Also talk about tags in the docker hub.
*Best Practice* - Mention the docker image (most upto date and suitable for testing) 
**6c.** Building docker container and push image to docker hub: list steps
**6d.** Run and test navigation 2 in the docker container: list steps
http://wiki.ros.org/docker/Tutorials/Docker

**7. Continues Integration and Circle CI:**
Explain automation of build and test process. Not more from a new version release standpoint but more from merging new code to the navigation 2 master.
https://vimeo.com/187705230
https://app.circleci.com/pipelines/github/ros-planning/navigation2

-----------------------------------------------PAGE5-------------------------------------------------------------

**8. Testing Navigation2 package:**
**8a.** Running build tests: Colcon test (3c), running the test from within Docker container (6d), test by Circle CI.
**8b.** Linters:
- Basics:
- Integrating Linters for Navigation 2 Testing: Talk about the ament_lint_auto package and how to use it. 
https://github.com/ament/ament_lint/blob/master/ament_lint_auto/doc/index.rst
https://answers.ros.org/question/306194/ros1-and-ros2-linting/
- Set of linters used so far in Navigation 2: list them and explain what they do. 
- Running Linters: Refer 8a.

SteveMacenski, 2020-05-26:
We try pretty hard to make sure we have as little redundancy as possible, both to reduce overhead of work and to make sure that things don't get out of sync. Page 1 seems like the existing "getting started" pages, I think those can just stay where they are? 

I think it would be good to rather think of this as being an addition to the "Getting Involved" section that we'd rename "Developer Guidance" or similar. The landing page I think would direct people to the getting started pages if required, that should outline the basic install stuff you have on Page 1. The "best practices" bullet you added for Page 1 would also be good for that landing page. A few paragraphs of the developer recommended setup of the "getting started" options. 

For page 2, I don't think we need to get into colcon basics too much, I think just giving them a link to the colcon docs and the ROS2 tutorial for it would be helpful. We should assume they have some basic understanding of it since we need to start at nav2 uses rather than from the technology basics. We can definitely link the docs for the basis though and ask them to read it if they haven't yet. Then on page 2, I think its mostly showing how to use them (colcon build, colcon test, colcon test-result) and the flags that are useful to know (package-select, verbose) that we often use. Also address linting here since that's part of the colcon/ament workflow. I think this page could be named something similar to "Build and Test Workflow". This could end up being a very long page and we might want to split into 2 subpages, one on colcon basics / flags / build, and the other on test and linting. But they should be one after another. 

For page 3, can you explain what your goal is with this page and how it fits into the developer process workflow? I agree staying up to date is important, but not sure if that could be lumped into another page (1 or 2) because its a short area. We should definitely not cover tutorials for the same reason as "getting started", developers will go to tutorials if they're looking for those.

For page 4, mostly good. I think between 6a and 6b, you're missing a how to build the docker image to work with (6c). 6c should not discuss pushing to dockerhub, that's only for admins, so I think you can remove that if you move the building portion between A and B. When a developer builds and works in a docker environment, they shouldn't be pushing a development workspace to dockerhub. I think this page can also include some tips / workflows for how to develop using the docker containers (e.g. build it locally, then go into an instance and take over the overlay workspace with your branch / work and develop internally). I don't think we should discuss CI, that's a very complicated topic that itself needs its own separate documentation. We don't use the ROS2 build farm in the way Dirk describes in that talk. I think of this page as answering the question "How to work with navigation2 and docker?" (what is docker, building an image, running an image, developing in an image, important flags to note and use).

Page 5 I think can be pushed into Page 2. Or reorder to be after page 2, see the comments on page 2.

I think a couple pages you could make at that point:
- Issue workflow. Explaining to fill out the template and include XYZ information (if crashing, give traceback, if issue, give way to reproduce, if feature, describe in detail with implementation considerations).
- PR workflow. Explaining to fill out the template and include XYZ information (image / gif if appropriate, tests, update documentation or migration guide, any new issues it may cause, if adding new parameter update docs, any future work)

SteveMacenski, 2020-06-09:
@Shru Hi just checking back in, any updates? I would love to have us finish designing this and start working on it!

Shru, 2020-06-09:
@SteveMacenski , really sorry I will pick up pace. :D

Ok agree, let’s make it as lean as possible. I have also removed the need to cross reference pages a lot.

The updated layout and details based on your feedback is below. Let me know if we should rework any part.

I think the only thing missing is details of test framework such as functional tests, integration tests unit tests. In Colcon test, we explain how to run some of these tests but we don’t have a page about the scope of these tests and how to interpret results, Codecov report etc. 
Should I club all these under Colcon test section because the tests we have right now are run by using the colcon test command/ctest command? Or shall we keep the 2 separate; like just introduce colcon test CLI as in **6c** and provide a link to another page which has details of test framework? I think the latter is better for making it future proof if we expand our test framework, please let me know your opinion. 

-----------------------------------------------LANDING PAGE------------------------------------------------
Bunch of other stuff…..
….
If you would like to install and try out Navigation 2 please visit **<Page 1 – Getting Started>**

To view other demonstrations and explore more capabilities of Navigation 2, please refer the <**Tutorials page**>.

For detailed setup options and for developers wanting to contribute to Navigation 2, please visit **<2. Prerequisites>**. Mention _Best Practice_ - Latest OS, ROS2 Master, Navigation 2 Master **<3d>**, use script **<3c>** to do the same for ease henceforth, other recommendations for developers.

-----------------------------------------------PAGE1 – GETTING STARTED-------------------------------
**SUBPAGE1:**

**1.** **Getting started with Navigation2:**
Let’s keep the minimal example (even though it is actually **3a**) so it’s nice for someone who would just like to see a turtle bot move on their screen with minimum setup details. 
**1a.** Installation
**1b.** Running the example
**1c.** Navigating

**SUBPAGE2:**

**2.** **Prerequisites:**
Explain in the form of a table target/OS dependencies, ROS2 versions that Navigation 2 supports.

**3.** **Build and Install Navigation 2:**
Mention various available options in terms of installing latest stable release or building Navigation 2 from master. 
**3a.** Install latest stable release: list steps
**3b.** Build and install for a specific released ROS distribution: list steps
**3c.** Build and install from script: list steps
**3d.** Build and install manually from master: We use the Colcon build system to build and test Navigation 2. For details about Colcon and to understand the steps listed below, please visit <**Page 2 – Build and Test Workflow>**
List steps.

**4.** **Build Troubleshooting Guide**: https://navigation.ros.org/build_instructions/build_docs/build_troubleshooting_guide.html#build-troubleshooting-guide
(For missing packages, establish ROS dependencies and not install individual packages.)

**5.** **Staying upto date with ROS and Navigation 2**
https://index.ros.org/doc/ros2/Installation/Maintaining-a-Source-Checkout/#maintainingsource
Best Practice – How often should we import ros2 repos and rebuild while developing navigation2?

-----------------------------------------------PAGE2 – BUILD AND TEST WORKFLOW---------------
**SUBPAGE1:**

**6. Colcon build system and how to use it for Navigation 2 development?**

**6a.** Links to familiarize with Colcon build system and its use in ROS:
https://vimeopro.com/osrfoundation/roscon-2019/video/379127725
https://colcon.readthedocs.io/
https://design.ros2.org/articles/build_tool.html
https://index.ros.org/doc/ros2/Tutorials/Colcon-Tutorial/

**6b.** Colcon build and flags (symlink-install, packages-skip) 

**SUBPAGE 2:**

**6c.** Colcon package testing: Colcon test, test-result, flags (verbose, packages-select, retest-until-pass, cmake specific arguments), ctest (specific packages)

**6d.** Linters:
    • Basics and scope of linters(code quality etc).
    • Integrating Linters for Navigation 2 Testing: Talk about the ament_lint_auto, ament_lint_common packages and how to use/run them.
https://github.com/ament/ament_lint/blob/master/ament_lint_auto/doc/index.rst
https://answers.ros.org/question/306194/ros1-and-ros2-linting/ 
    • Set of linters used so far in Navigation 2: list them(ament_cpplint, ament_uncrustify, ament_cppcheck, ament_cmake_gtest) and explain what they do. 
[... 12 lines cut ...]
http://wiki.ros.org/docker/Tutorials/Docker

**SUBPAGE2:**

**7e.** Developing in an image: Talk about underlay and overlay, _workflow tips_ about using overlay workspace. The essence is explained here https://index.ros.org/doc/ros2/Tutorials/Workspace/Creating-A-Workspace/ Have to give a similar Navigation 2 related example in concise manner. 
Mention docker flags as and when necessary within the above steps. 

-------------------------------------PAGE4 – DEVELOPER PROCESS----------------------------------
**SUBPAGE1:**

**8. Getting Involved:** Retain the current getting involved part.

**9. Process:** Overview
**9a.** Issue workflow: Explaining to fill out the template and include XYZ information (if crashing, give traceback, if issue, give way to reproduce, if feature, describe in detail with implementation considerations). 
**9b.** PR workflow: Explaining to fill out the template and include XYZ information (image / gif if appropriate, tests before submitting a PR and any common troubleshooting steps, update documentation or migration guide, any new issues it may cause, if adding new parameter update docs, any future work) 

**SUBPAGE2:**

**10. Licensing**

**11. DCO**

**SUBPAGE3:**

**12. Projects for 2020 Summer Student Program**

Shru, 2020-06-15:
Landing page – Sorry, I got confused that this is the overall website landing page. 

Page 1 – Cool, so we go from Landing Page for developer guide --> Build and Test Workflow --> Docker --> Developer Process. 
As far as content of this page "Page 1 - Getting Started" is concerned, I plan to leave most of things as it is, in the current website. I will only make sure that the steps listed in this section yield the desired results and add a few minor points. 

So are we OK to start writing this? :)  

---------------------------------------LANDING PAGE FOR DEVELOPER GUIDE --------------------
Introduce what this section of the website is about. Have some paragraph that this is explaining the developer workflow for setting up your environment, compiling, testing, linting, and working on tickets/Prs. Link those sections as shown below, where ever necessary.
If you would like to install and try out Navigation 2 please visit **<Page 1 – Getting Started>**
To view other demonstrations and explore more capabilities of navigation 2, please refer the **<Tutorials>**
For detailed setup options and for developers wanting to contribute to Navigation 2, please visit **<2. Prerequisites>**. Mention _Best Practice_ - Latest OS, ROS2 Master, Navigation 2 Master **<3d>**, use script **<3c>** to do the same for ease henceforth, other recommendations for developers.
Just to give them some knowledge about "hey, what's this all about".

-----------------------------------------------PAGE1 – GETTING STARTED-------------------------------
**SUBPAGE1:**
**1.** Getting started with Navigation2 (First Demo):
Let’s keep the minimal example (even though it is actually **3a**) so it’s nice for someone who would just like to see a turtle bot move on their screen with minimum setup details. 
**1a.** Installation
**1b.** Running the example
**1c.** Navigating

**SUBPAGE2:**
**2.** Prerequisites:
Explain in the form of a table target/OS dependencies, ROS2 versions that Navigation 2 supports.
**3.** Build and Install Navigation 2:
Mention various available options in terms of installing latest stable release or building Navigation 2 from master. 
**3a.** Install latest stable release: list steps
**3b.** Build and install for a specific released ROS distribution: list steps
**3c.** Build and install from script: list steps
**3d.** Build and install manually from master: We use the colcon build system to build and test Navigation 2. For details about colcon and to understand the steps listed below, please visit **<Page 2 – Build and Test Workflow>**
List steps.
**4.** Build Troubleshooting Guide: https://navigation.ros.org/build_instructions/build_docs/build_troubleshooting_guide.html#build-troubleshooting-guide
(For missing packages, establish ROS dependencies and not install individual packages.)
**5.** Staying upto date with ROS and Navigation 2.
https://index.ros.org/doc/ros2/Installation/Maintaining-a-Source-Checkout/#maintainingsource
_Best Practice_ – How often should we import ros2 repos and rebuild while developing navigation2?

-----------------------------------------------PAGE2 – BUILD AND TEST WORKFLOW---------------
Start with developer workflow in terms setup environment and repeat the Best Practices part here. 
**SUBPAGE1:**
**6.** Colcon build system and how to use it for Navigation 2 development?

**6a.** Links to familiarize with colcon build system and its use in ROS:
https://vimeopro.com/osrfoundation/roscon-2019/video/379127725
https://colcon.readthedocs.io/
https://design.ros2.org/articles/build_tool.html
https://index.ros.org/doc/ros2/Tutorials/Colcon-Tutorial/

**6b.** Colcon build and flags (symlink-install, packages-skip) 

**SUBPAGE 2:**
**6c.** Colcon package testing: Colcon test, test-result, flags (verbose, packages-select, retest-until-pass, cmake specific arguments), ctest (specific packages)
**6d.** Linters:
    • Basics and scope of linters(code quality etc).
    • Integrating Linters for Navigation 2 Testing: Talk about the ament_lint_auto, ament_lint_common packages and how to use/run them.
https://github.com/ament/ament_lint/blob/master/ament_lint_auto/doc/index.rst
https://answers.ros.org/question/306194/ros1-and-ros2-linting/ 
    • Set of linters used so far in Navigation 2: list them(ament_cpplint, ament_uncrustify, ament_cppcheck, ament_cmake_gtest) and explain what they do. 
      
-----------------------------------------------PAGE3 - DOCKER----------------------------------------------

**SUBPAGE1:**
**7.** Using Docker for development:
Explain in a few lines what docker is and how docker containers are useful in development process (faster builds, isolated environment, ability of have multiple dockers, etc). 
Basic docker operations:
**7a.** Install docker: list steps
**7b.** Build a docker image locally: list steps.
**7c.** Use an existing docker image present in docker hub: list steps. Also talk about tags in the docker hub.
_Best Practice_ - Mention the docker image (most upto date and suitable for testing)
**7d.** Run and test navigation 2 in the docker container: list steps
http://wiki.ros.org/docker/Tutorials/Docker

**SUBPAGE2:**
**7e.** Developing in an image: Talk about underlay and overlay, workflow tips about using overlay workspace. The essence is explained here https://index.ros.org/doc/ros2/Tutorials/Workspace/Creating-A-Workspace/ Have to give a similar Navigation 2 related example in concise manner. 
Mention docker flags as and when necessary within the above steps. 

[... cut ...]
**9a.** Issue workflow: Explaining to fill out the template and include XYZ information (if crashing, give traceback, if issue, give way to reproduce, if feature, describe in detail with implementation considerations). 
**9b.** PR workflow: Explaining to fill out the template and include XYZ information (image / gif if appropriate, tests before submitting a PR and any common troubleshooting steps, update documentation or migration guide, any new issues it may cause, if adding new parameter update docs, any future work) 

**SUBPAGE2:**
**10.** Licensing
**11.** DCO

SteveMacenski, 2020-06-15:
Again, Page 1 should not exist. You should only have the landing page reference the existing getting started page to get setup. Remove Page 1 from your overview. Landing page should have the recommended build instructions & link to the build page, which also already exists. 

No other changes

SteveMacenski, 2020-06-18:
Sound good @Shru ?

Shru, 2020-06-22:
@SteveMacenski Yes, perfect. I will keep Page 1 out of this. Page 1 is already present in the website and it will not be a part of the developer workflow. So we go from Landing Page -> Page 2 with proper links in the landing page as you mentioned.
I have started to document. I will get back to you with the link and any questions if I have during the process. Once I have one or two pages in, I will get back to you and you could take a look and comment on whether the representation of the content is OK and any missing information and so on. Please let me know if you have questions/suggestions in the mean time and I will take them into consideration while writing.

SteveMacenski, 2020-06-22:
Awesome, sounds good :-) Thanks for taking the lead on this. It will be really helpful to link people to this instead of answering it every time someone asks.

Shru, 2020-06-23:
No problem. This is a cool assignment to work on :)

SteveMacenski, 2020-07-08:
How's it going @Shru ?

SteveMacenski, 2021-08-23:
Covered in tickets in navigation.ros.org for documentation / tutorials

closed 2021-08-23 by SteveMacenski