ros-navigation/navigation2 #1739: Build errors in test_map_server_node

fmrico, 2020-05-15:
Dear All

I am building master from source, and test_map_server_node.cpp:74 makes fail the compilation

```
if (rclcpp::spin_until_future_complete(node, result) == rclcpp::FutureReturnCode::SUCCESS) {
```

This error comes from a PR merged in rclcpp 22 days ago [here](https://github.com/ros2/rclcpp/pull/1083), that moved `FutureReturnCode` from `rclcpp::executor` namespace to `rclcpp`. 4 days ago, we incorporated that update in #1718. 

I think that this change should (only the namespace change) be reverted. Otherwise, only people running ros2 (master) from sources can compile Nav2.

Francisco

shivaang12, 2020-05-15:
[My proposal]
After reverting, removing the deprecate flag will help "from source" people to compile it smoothly too.

fmrico, 2020-05-15:
It is not the only build issue compiling with eloquent. #1669 makes also fail compilation for an upgrade to foxy.

@SteveMacenski Should we move to Foxy to continue contributing? Maybe it is reasonable.

SteveMacenski, 2020-05-15:
> Otherwise, only people running ros2 (master) from sources can compile Nav2.

So for Nav2 **master** that's intended behavior. I agree that change is a pain so you can't use eloquent binaries anymore, but prior to this last 6 months, you couldn't have done that anyhow because master was changing so much as had to keep up. Things have stabilized more now allowing that to be possible. 

Yes, you should move to Foxy or ROS2 master. Hopefully theres less breaking changes in the future, I thought that breaking change was really silly and they should have just left that in the namespace. It seems to have such little value in that particular change.

SteveMacenski, 2020-05-18:
@fmrico does that answer the question? Can we close this?

fmrico, 2020-05-19:
@SteveMacenski Yes, it solves my question. Thanks!!

closed 2020-05-19 by fmrico