ros-navigation/navigation2 #1740: Add BT nodes to adjust replanning rate according to speed

naiveHobo, 2020-05-15:
## Feature request

#### Feature description

A ```SpeedController``` node (analog to ```RateController``` and ```DistanceController```) should be added which would allow adjusting the replanning rate based on the current speed. ```RateController``` and ```DistanceController``` nodes have some drawbacks when used individually (#1701). Speed based rate of replanning seems like the natural solution to these drawbacks.

#### Implementation considerations

The ```SpeedController``` node can be implemented as an extension to ```RateController``` where the rate is scaled according to the current speed.
- By setting limits on the rate, it ensures the replanning happens even if the robot is not moving, offering a solution for the drawback of the ```DistanceController```.
- When the robot is moving at high speeds, the rate is increased accordingly and replanning is done more frequently, thus retaining the benefit of using the ```DistanceController```.
- When at a constant speed, the ```SpeedController``` boils down to a ```RateController```.

naiveHobo, 2020-05-15:
I have an initial implementation for the node proposed. Here's how the replanning rate scales according to the current speed:

[image]

The robot is stuck towards the end so the rate of replanning in that case is held at 0.1hz.

SteveMacenski, 2020-05-15:
Can you link a PR to this?

naiveHobo, 2020-05-16:
Yup, the scales can be set as node parameters. The replanning rate is in the range [0.1, 1.0] hz scaled to according to speed in the range [0, 0.25] m/s.

closed 2020-06-12 by naiveHobo