ros-navigation/navigation2 #1757: 3D lidar support / vendors

SteveMacenski, 2020-05-21:
We should consider doing 3D lidar demos analog to the VSLAM demos currently happening. In 3D SLAM-world the suggestions I'd throw out to consider:
- HDL SLAM / Localization https://github.com/koide3/hdl_graph_slam
- LeGO-LOAM https://github.com/RobustFieldAutonomyLab/LeGO-LOAM
- lidar slam ros2 https://github.com/rsasaki0109/lidarslam_ros2

with emphasis on the options that include a localization mode or are compatible with another open-source high quality localizer.

SteveMacenski, 2020-07-22:
The major deliverables:

    1 or 2 3D SLAM systems with some parameter profiles that work well for mobile robots
    Documentation about 3D SLAM uses in navigation (what are they, why, sensors to use, how, maybe some parameter help)
    Development required to get that all working well (3D static map server, 3D static map costmap layer, anything else)
    Some data set / demos of using a robot with one of the 3D SLAM systems to do some task

Some of the metrics to compare by:

    CPU / GPU utilization (does it use a GPU if available?)
    Scaling - does it work reliably in larger indoor environments like a warehouse / retail store / etc
    map quality - does it actually map the space well and are the loop closures working / stable (may require parameter tuning to get good results for the mobile robot use)

peroveRedRockAI, 2021-09-09:
Hi Steve

Thanks for all the work on Nav2, slam toolbox and ouster driver and more. Is there any update on this developement? 

Are you familiar with the cmu exploration project and the [Ground-based Autonomy Base Repository](https://github.com/jizhang-cmu/ground_based_autonomy_basic) by Ji Zhang (LOAM/Kaarta) at CMU.

[CMU Exploration](https://www.cmu-exploration.com/development-environment)

Also Livox have published a Lidar simulation for Gazebo ([Livox Laser Simulation](https://github.com/Livox-SDK/livox_laser_simulation)) and some slam implementations on their Github [(LIO-Livox)](https://github.com/Livox-SDK/LIO-Livox), [(Livox-Mapping)](https://github.com/Livox-SDK/livox_mapping), [(LOAM-Livox)](https://github.com/Livox-SDK/loam_livox)

SteveMacenski, 2021-09-13:
No current work on this, its future work at this point unless you're interested in contributing!