ros-navigation/navigation2 #1762: Segfault on planner server on Eloquent

blacksoul000, 2020-05-22:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Version or commit hash:
  - from source: ea1667e510335789f932594b6604b177c4592763
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
Follow waypoints for some minutes (about 2 min).

#### Expected behavior
No segfault.

#### Actual behavior
Segfault

#### Additional information
geometry2 updated to bab90b3b74174599866cb76675bdcceac9efad1e

[nav2.zip](https://github.com/ros-planning/navigation2/files/4666811/nav2.zip)

`
#0  __memmove_avx_unaligned_erms () at ../sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S:306
#1  0x00007fffa43dcb6d in nav2_navfn_planner::NavfnPlanner::smoothApproachToGoal(geometry_msgs::msg::Pose_<std::allocator<void> > const&, nav_msgs::msg::Path_<std::allocator<void> >&) ()
   from /home/user/workspace/nav2/install/nav2_navfn_planner/lib/libnav2_navfn_planner.so
#2  0x00007fffa43de6be in nav2_navfn_planner::NavfnPlanner::makePlan(geometry_msgs::msg::Pose_<std::allocator<void> > const&, geometry_msgs::msg::Pose_<std::allocator<void> > const&, double, nav_msgs::msg::Path_<std::allocator<void> >&) ()
   from /home/user/workspace/nav2/install/nav2_navfn_planner/lib/libnav2_navfn_planner.so
#3  0x00007fffa43de8c5 in nav2_navfn_planner::NavfnPlanner::createPlan(geometry_msgs::msg::PoseStamped_<std::allocator<void> > const&, geometry_msgs::msg::PoseStamped_<std::allocator<void> > const&) ()
   from /home/user/workspace/nav2/install/nav2_navfn_planner/lib/libnav2_navfn_planner.so
#4  0x00007ffff7b8a63d in nav2_planner::PlannerServer::computePlan() ()
   from /home/user/workspace/nav2/install/nav2_planner/lib/libplanner_server_core.so
#5  0x00007ffff7b9da1b in nav2_util::SimpleActionServer<nav2_msgs::action::ComputePathToPose, rclcpp::Node>::work() () from /home/user/workspace/nav2/install/nav2_planner/lib/libplanner_server_core.so
#6  0x00007ffff7b9e07c in std::_Function_handler<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> (), std::__future_base::_Task_setter<std::unique_ptr<std::__future_base::_Result<void>, std::__future_base::_Result_base::_Deleter>, std::thread::_Invoker<std::tuple<nav2_util::SimpleActionServer<nav2_msgs::action::ComputePathToPose, rclcpp::Node>::handle_accepted(std::shared_ptr<rclcpp_action::ServerGoalHandle<nav2_msgs::action::ComputePathToPose> >)::{lambda()#1}> >, void> >::_M_invoke(std::_Any_data const&) ()
   from /home/user/workspace/nav2/install/nav2_planner/lib/libplanner_server_core.so
#7  0x00007ffff7b90e19 in std::__future_base::_State_baseV2::_M_do_set(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*) ()
   from /home/user/workspace/nav2/install/nav2_planner/lib/libplanner_server_core.so
#8  0x00007ffff73b7827 in __pthread_once_slow (once_control=0x7fffb8001e18, 
    init_routine=0x7ffff6ec98a0 <__once_proxy>) at pthread_once.c:116
#9  0x00007ffff7b924dd in std::thread::_State_impl<std::thread::_Invoker<std::tuple<std::__future_base::_Async_state_impl<std::thread::_Invoker<std::tuple<nav2_util::SimpleActionServer<nav2_msgs::action::ComputePathToPose, rclcpp::Node>::handle_accepted(std::shared_ptr<rclcpp_action::ServerGoalHandle<nav2_msgs::action::ComputePathToPose> >)::{lambda()#1}> >, void>::_Async_state_impl(std::tuple<nav2_util::SimpleActionServer<nav2_msgs::action::ComputePathToPose, rclcpp::Node>::handle_accepted(std::shared_ptr<rclcpp_action::ServerGoalHandle<nav2_msgs::action::ComputePathToPose> >)::{lambda()#1}>&&)::{lambda()#1}> > >::_M_run() () from /home/user/workspace/nav2/install/nav2_planner/lib/libplanner_server_core.so
#10 0x00007ffff6eca6df in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#11 0x00007ffff73af6db in start_thread (arg=0x7fff697fa700) at pthread_create.c:463
#12 0x00007ffff692588f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
`

SteveMacenski, 2020-05-22:
@blacksoul000 this is duplicate of #1608 that was fixed with https://github.com/ros-planning/navigation2/pull/1612 in master. We did not backport this to eloquent but we'd be happy to. Can you cherry pick that commit to Eloquent and submit a PR?

blacksoul000, 2020-05-26:
Done. #1767

closed 2020-05-28 by SteveMacenski