ros-navigation/navigation2 #1763: Use of get_parameter_or not working.

tgreier, 2020-05-22:
Where get_parameter_or is used, those values do not seem to be read from yaml.
It seems to be a problem perhaps since introduction of declare_parameter?

## Bug report

**Required Info:**

- Operating System:
 Ubuntu 18.04
- Version or commit hash:
eloquent-devel

#### Steps to reproduce issue
Try to set from yaml:
odom_topic
goal_reached_tol
required_movement_radius
movement_time_allowance_

Also, underscore at end of "movement_time_allowance_" should be removed?

#### Expected behavior
Values will be set
#### Actual behavior
Defaults are retained
#### Additional information

SteveMacenski, 2020-05-22:
Please give exact examples and links to their lines in source for review.

tgreier, 2020-05-22:
I'm sorry, I am using eloquent-devel branch so these changes are not in that branch. The change you referenced I believe is in master branch.

However, I originally was having a problem with "odom" topic, which is here:

https://github.com/ros-planning/navigation2/blob/7c0ac74514506591c0d508a6bc10e42d1d67162f/nav2_dwb_controller/nav_2d_utils/include/nav_2d_utils/odom_subscriber.hpp#L69

Should master branch be used for latest ros2 work? I apologize for my newbie question.

SteveMacenski, 2020-05-22:
Master should be used for latest ros2 work, yes. But you can cherry pick that PR's commit to Eloquent and then submit a PR and I'll merge it to fix eloquent. 

I agree, I think in that case there's an issue because the `odom_topic` hasn't been declared. Without declaring, you can't get it from the file so it'll always use the default or the topic provided in the constructor. 

You can submit a PR to fix that using the nav2_utils declare if not declared method before the `get_parameter_or` to declare it and then it should work. 

Please update the original ticket with the link to each of these and I can help figure out which/what the solution is for you to patch in a PR.

closed 2020-05-27 by SteveMacenski