ros-navigation/navigation2 #1773: BT navigator nodes with use_sim_time

daisukes, 2020-05-27:
I have been struggling with getting time from `/clock` in my custom BT nodes.
I found that the default parameter file does not include `use_sim_time` for `bt_navigator_client_node`, which can be obtained as `node` on the blackboard of BT.

Do you think this is worthy to include as a default parameter?
```
bt_navigator_client_node:
  ros__parameters:
    use_sim_time: true
```

SteveMacenski, 2020-05-27:
Sure, but are you sure its not 

```

bt_navigator_rclcpp_node:
  ros__parameters:
    use_sim_time: True
```

which is in the file? Or is that name of the node incorrect. All of the things we refer to as "client nodes" are listed under `XYZName_rclcpp_node` in the configuration file. Are these all wrong?

daisukes, 2020-05-27:
https://github.com/ros-planning/navigation2/blob/master/nav2_bt_navigator/src/bt_navigator.cpp#L76
https://github.com/ros-planning/navigation2/blob/master/nav2_bt_navigator/src/bt_navigator.cpp#L117

`client_node_` is created with the name `bt_navigator_client_node` and set as `node` of the blackboard.

SteveMacenski, 2020-05-27:
I think what could be done instead is remove the `client_node_` and use `rclcpp_node_` instead. I think that will fix your issue since `rclcpp_node_` is created from the `true` here https://github.com/ros-planning/navigation2/blob/master/nav2_bt_navigator/src/bt_navigator.cpp#L33 but not used in this code. So you can use `rclcpp_node_` rather than making a new one. Then the namespace parameter for `bt_navigator_rclcpp_node` would be correct.

SteveMacenski, 2020-05-27:
I think that same thing should happen in the `waypoint_follower` node, which has the same other issue.

daisukes, 2020-05-27:
`costmap_2d_ros` also creates another node, which has a separate parameter in the nav2_params.yaml and each layer is initialized with a client node and a rclcpp node.
https://github.com/ros-planning/navigation2/blob/2113b6a7d9a9f8ab2f482047e312a72af87a3b2e/nav2_costmap_2d/src/costmap_2d_ros.cpp#L79

https://github.com/ros-planning/navigation2/blob/fc762f659d98ce4ca2ebc5d87e6fc556742ffc9a/nav2_bringup/bringup/params/nav2_params.yaml#L175

https://github.com/ros-planning/navigation2/blob/2ac07b2a8a8f718edd596c9363a53d2d6998163b/nav2_costmap_2d/src/layer.cpp#L48

I'm not sure why but it seems to intentionally create a non-spinning node rather than using a spinning node (`rclcpp_node_`). This sounds to me like there is a reason to create a node.
https://github.com/ros-planning/navigation2/blob/fc762f659d98ce4ca2ebc5d87e6fc556742ffc9a/nav2_bt_navigator/include/nav2_bt_navigator/bt_navigator.hpp#L132

SteveMacenski, 2020-05-27:
Costmap 2d is a special case, I think that actually requires 3 nodes. There's alot going on there. the `rclcpp_node_` is used in the backend for TF and something intentionally open for the plugin layers to use for their needs since that's in-determinant at compile time.

daisukes, 2020-05-27:
I have tried the change you suggested then got this error.
It looks like two threads are trying to use the node. 
```
[bt_navigator-10] [ERROR] [1590607379.467104705] []: Original error: Node has already been added to an executor.
```

SteveMacenski, 2020-05-27:
What did you try to do, exactly that caused that? Can you send me a link to your branch?

daisukes, 2020-05-27:
Pushed the change here and got this error with the system test of navigation2.
https://github.com/CMU-cabot/navigation2/tree/bt-navigator-fix2

SteveMacenski, 2020-05-27:
Huh, I know why. The BT node that's given to the blackboard is spinning for the services / actions / subscribers that use it (e.g. there are instances of spin_some() and others in the BT node plugins). So the `rclcpp_node_` when created is given to a thread that's spinning it for us, so the exception is being thrown that we're trying to spin it twice (or N times). 

We have  couple options. Option 1 is that we add a parameter about whether or not to start spinning the NodeThread [here](https://github.com/ros-planning/navigation2/blob/master/nav2_util/src/lifecycle_node.cpp#L58) and set that to false for this node. Option 2 is that we change the `true` to `false` in the `LifecycleNode` constructor for the BT navigator and we readd the `client_node_` for use (since the `rclcpp_node_` isn't useful to us in that context). For this, we could either name it `_client_node` or `_rclcpp_node` depending on if we want to keep things inline with other nodes in the configuration file.

My suggestion is turning the `true` to `false` and adding back in the `client_node_` under the name `rclcpp_node_` so we don't need to make any configuration file changes and its in line with all the other processes. 

I don't think you'll hit this issue though in the waypoint_follower analog.

daisukes, 2020-05-27:
> My suggestion is turning the `true` to `false` and adding back in the `client_node_` under the name `rclcpp_node_` so we don't need to make any configuration file changes and its in line with all the other processes. Path of least resistance.

Sounds good to me.

> I don't think you'll hit this issue though in the waypoint_follower analog.

waypoint_follower also has `spin_some`
https://github.com/ros-planning/navigation2/blob/879a9379ee0a8931fc73315c4eee477ea902049a/nav2_waypoint_follower/src/waypoint_follower.cpp#L212

This change looks good at least the system test. So I will make a PR.

SteveMacenski, 2020-05-27:
If you change to the `rclcpp_node_` internal node, then you can remove the spin some.

Optionally, you can just turn the `true` to `false` and just call it done (path of least resistance). The goal is to reduce from 3 nodes to 2, I don't think it really matters which.

SteveMacenski, 2020-05-27:
Thanks! You knocked out 2 nodes from the network, that's actually a pretty real performance booster.

closed 2020-05-27 by SteveMacenski