ros-navigation/navigation2 #1779: Recoveries always return success, regardless if they're able to complete their required tasks or not

SteveMacenski, 2020-05-28:
We should investigate the return statuses of the recoveries. They will all return success, even if in fact they halt early due to a potential collision. Ex. 

```
  if (!isCollisionFree(distance, cmd_vel.get(), pose2d)) {
    stopRobot();
    RCLCPP_WARN(node_->get_logger(), "Collision Ahead - Exiting BackUp");
    return Status::SUCCEEDED;
  }
```

I remember in the early days we accepted this but I wonder now if we should have it fail and figure out the necessary changes to the behavior tree to handle this properly (use of different or custom control nodes for recoveries). 

Within the context of the BT, the state return SUCCESS or FAILURE probably doesn't matter (?) since its trying to do something to fix a bad situation. Even if it only partially moves or particularly spins, that could still successfully unblock the robot, so even a FAILURE is a SUCCESS, maybe. That's perhaps an argument for having the recovery BT nodes handle the returned action status and always return a certain value or use a control flow node that will not do "bad things" due to recovery failures. 

This is a real problem though if its called outside the context of the behavior tree. If I'm calling backup from a controller, I really want to know if it backed-up properly. Same with others. Right now the logic for always succeed is in the action server for the recovery itself, not in the BT node or the condition node hosting recovery BT nodes. This is making is so that when I write a test to intentionally collide with something, I still get success even though it failed, making it impossible to test failure conditions of the recoveries.

My suggestion - and open for discussion - is that we:
- Properly return the success / failure state in the recovery plugins in the recovery server
- In the Behavior Tree, either mask these values on return to always be success, or modify the control flow node used so that a failure of a recovery doesn't negatively impact performance.

vinnnyr, 2020-06-29:
I am starting to work on this issue.

vinnnyr, 2020-07-06:
@SteveMacenski Is an appropriate resolution to change the logic in the Recovery Control Node?

Changes to the logic in bold:
“The RecoveryNode is a control flow node with two children. It returns SUCCESS if and only if the first child returns SUCCESS. The second child will be executed only if the first child returns FAILURE If the second child SUCCEEDS **OR FAILS**, then the first child will be executed again.“

SteveMacenski, 2020-07-06:
I think we need to change the way the recoveries work first, since they always return success. We need them to return failure when they actually fail to process (like spin due to collision or clear costmap because it couldn't call the  service). At that point, we need to figure out some changes to the behavior tree to support it, which would probably involve changing that logic, yes.

vinnnyr, 2020-07-07:
Yes, sorry, I didn’t explain that I was able to identify the areas in the action server recovery plugins that need to be changed.

Where I might need some advice is what kind of behavior tree masking makes the most sense. There are a few places that might make sense. @gimait and @naiveHobo , if you have a chance, would love some advice. 

1. Change the logic in the RecoveryNode as described above
2. Change the logic in the BT recovery actions themselves? Perhaps change the behavior such that on tick it always returns `SUCCESS` (In the context of the BT)

gimait, 2020-07-09:
I think you can change the return statements to failure in the cases set here, but we need to be aware of one thing: Once that is done, we cannot differentiate between an error in the server and exiting the action cleanly.

If that's okay and intended, then it's simple (until you look at the BT :wink:):
1. Change the recovery plugins to return failure instead of success as needed.
2. I don't think you need to change the RecoveryNode, instead, change the SequenceStar (the one that is parent of the recovery action nodes) and change it with something that continues the sequence even if there is a failure, but that returns failure if all children fail. I believe you could do this with the default control nodes and decorators, it might just be necessary to give it a lot of thought. If this is not an option, then I would write the appropriate control node to do this.

naiveHobo, 2020-07-09:
> and change it with something that continues the sequence even if there is a failure, but that returns failure if all children fail.

@gimait have a look at the `RoundRobin` control node. I worked on it a while ago and that seemed like a good option for the recovery branch. It pretty much works the way you described.

Other than that, I agree with @gimait's suggestions, changes in the `RecoveryNode` are not required. Recovery plugin fixes should be enough.

SteveMacenski, 2020-07-09:
I think if the recovery passes or fails, that should be propagated from the server to the BT node and the BT node should fail / succeed based on it. It might be important for general use that someone knows if the robot was unable to call a service for help, clear a costmap, or spin in place due to collision.

SteveMacenski, 2021-01-14:
Just an update @vinnnyr @naiveHobo this is now the only blocking work for the V1.0! 

Exciting days :-) The ROS 2 action deadlock issues have been resolved so excited!

closed 2021-02-12 by SteveMacenski