ros-navigation/navigation2 #1780: Select behavior Tree in NavigateToPoseAction

fmrico, 2020-05-28:
## Feature request

#### Feature description

I would like to open a discussion (or continue the one in #1748) about how the users of Navigation2 can select the desired robot's behavior during navigating. This behavior is coded by the BTs XML file and selected in nav2_params.yaml.

By now, we consider navigating is going from point A to point B (or patrol several points, but it is the same function), and finish the navigation. #1748 is different. It uses navigation to follow a dynamic point to a certain distance, and this is not going from A to B and finish. Using navigation instead of a reactive behavior provides a way to create paths and to avoid obstacles, using all the algorithms that we are developing in this repo. My approach is using a different BT for navigating, composing the necessary nodes to accomplish this task.

Navigation2 users could need to change the navigation behavior during robot navigation. For example:

1. Navigate from the entrance to the living-room.
2. Following a dynamic goal for 1 minute.
3. Return to entrance.

BT in 1 and 3 is different from 2. I believe that the difference is the BT used to navigate. How could the Navigation2 user do this at runtime? My opinion is that this may be done by selecting in NavigateToPoseAction the BT that should be used to navigate.

#### Implementation considerations

- [ ] Changes in  NavigateToPose.action to include a field `string behavior_tree` or something like that in the request field. If it is void, the use the BT by default.

- [ ]  Use a list of BTs, in the similar way as plugins (the first could be the BT by default):

```
    bt_xml_filenames: ["navigate_w_replanning_and_recovery.xml", "dynamic_goal.xml"]
    bt_ids: ["default", "dynamic_goal"]
```

- [ ]  In `bt_navigator`  make the call  `bt_->buildTreeFromText(..)` in `navigateToPose` method if the selected BT changes.

fmrico, 2020-05-28:
A possible implementation is here: https://github.com/fmrico/navigation2/tree/select_bt , but I would like to wait for discussing this before any PR

gramss, 2020-05-28:
Can this be extended to use "external" BT that are not know at compile time. BT can be created at runtime through xml. But all control/decorators and nodes must be known during compile time.

This would be the next step after what you have already done here. I really like your thinking of using a dynamic usage of different BT. I'm currently thinking about how to further improve such dynamic adding of functionalities through BT at runtime.

I'd be happy to chat with you about such things in a longer conversation, if you would like so! :)

fmrico, 2020-05-29:
Hi @gramss 

It sounds very interesting. Your idea of creating dynamically BTs remember me [this paper.](https://arxiv.org/pdf/1611.00230.pdf). 

One extension of my PR could be detecting if the `behavior_tree` field in the action is not an ID, but an XML containing the BT, and use it.

> I'd be happy to chat with you about such things in a longer conversation, if you would like so! :)

Yes!! I'd like to hear your ideas :)

Best

gimait, 2020-06-01:
Hi, I saw #1784 and then ended here because I am quite interested in bts and I would like to join this conversation.
I really like the idea of updating the behavior tree dynamically, that would be a great upgrade I think.

However, I'm concern about providing a list of trees to choose from at runtime. What would be the advantage of loading several behavior trees over defining the complete robot behaviour on one behavior tree in this case? The current behavior trees only take in consideration a simple task because they define an example behavior (I guess), but you could easily define a behavior tree that can have different navigation settings (different planner, recovery and controller servers) for each of the cases you describe in this ticket or any other.

I saw that you are giving a list of BTs to the `bt_navigator` in #1784. Wouldn't you need to hard code behaviors if you load several trees? How would you define when to use one or other behavior tree?

fmrico, 2020-06-01:
Hi @gimait 

> but you could easily define a behavior tree that can have different navigation settings (different planner, recovery and controller servers) for each of the cases you describe in this ticket or any other.

It would be a huge BT. I am not sure if this solution is scalable. I don't see the advantage to having a library of BT and select which one you want to use. 

> I saw that you are giving a list of BTs to the bt_navigator in #1784. Wouldn't you need to hard code behaviors if you load several trees?

I don't understand this question. BT defines a navigation behavior.

> How would you define when to use one or other behavior tree?

BT is defined in a field of the NavigateToPose action. If no BT is selected in the action, the default BT will be used.

SteveMacenski, 2020-06-01:
@fmrico I 100% agree with the intent of this ticket. I will say however, that you "could" have all 3 BTs in 1 tree with logic for which subtree to enter, so it should be possible for you to have `N` distinctive tasks in a single tree. I think that's not great and this is something we should enable. 

Do you think the navigation request should get:
- A file path to a XML
- A mapped name of an XML
- The BT itself in a message form?

I don't think that the mapped name is really important, but we could do that. 

Open questions: 
Can you please profile and give us an understanding of the overhead per Bt of having a map of BTs? Do you think we need to have N loaded or just have the current one loaded when requested? How much time does it take for the BT to build from XML?

SteveMacenski, 2020-06-01:
Last question would be how to deal with parameterization? Should we have a parameter for the BT to load as the default one or should all navigate to pose requests required to mention what to use? That's a pretty big API change.

fmrico, 2020-06-01:
Hi @SteveMacenski 

> - A file path to a XML <-- my preference
> - A mapped name of an XML <-- not my preference but I could live with it
> - The BT itself in a message form? <-- very not my preference

Ok to "A file path to a XML". I have to change the PR then. Now it is implemented "A mapped name of an XML"

> Can you please profile and give us an understanding of the overhead per Bt of having a map of BTs? 

Ok. I will work on it. 

> Do you think we need to have N loaded or just have the current one loaded when requested? How much time does it take for the BT to build from XML?

My first measures gave me a result that loading a BT takes a very short time. I will come back with numbers.

> Last question would be how to deal with parameterization? Should we have a parameter for the BT to load as the default one or should all navigate to pose requests required to mention what to use? That's a pretty big API change.

I would add a parameter with the default BT.

SteveMacenski, 2020-06-01:
@fmrico if you think we should just give it a filepath, then you don't need to analyze the overhead of multiple BTs, but still the load time. 

I'd be OK with a map of them, but there are some logistics we need to work out. Either way, we should have some kind of default BT in case none is given. So maybe rather than `bt_xml_file` its `default_bt_xml_file` or something as the default in case no BT is provided in the action.

gimait, 2020-06-02:
I think I understand the issue now. I know that the tree needed for this would be huge, I just wanted to ask what would be the workflow for accomplishing multiple tasks with different behaviors automatically. What I meant with hard coding behaviors is that selecting one or another behavior tree will also make a part of the robot behavior, and it is necessary to place that behavior selection somewhere. I didn't realise that is intended here is to have an external system requesting the specific goal-behavior pair.

I really like the idea of being able to update the behavior tree online and I think this is going to be a great upgrade. I also agree that it would be nicest to send directly a xml file path on the navigation request (seems a more flexible solution, probably will reduce the number of places you need to touch if you want to add a new bt).

SteveMacenski, 2020-06-02:
I still think the mappings is something to be discussed. There's consistency reasons to do that here if it were to make sense but it kind of depends on how you think about the autonomy system that's calling the NavigateToPose node. I don't think loading all `N` nodes at startup so we have a literal map of trees is probably smart from a resource utilization standpoint (requires analysis). An  implementation like this could be done by having the  `filename` and `name` in a parameter file, and then the a single tree is loaded from that cache of `filename` to `name/id` for use at any given time (or not load another if its the same). The alternative would be to just supply the filename to use and be done with it without any extra configs in the yaml files, which can also be a benefit or a drawback since you're then pushing that responsibility up a layer. Having a pre-defined set of them to use and abstracting away the specific implementation to the role it fits was a rationale the Intel guys fully held and I agreed with.

closed 2020-06-12 by SteveMacenski

rucha6g, 2023-08-07:
I am new to Nav2. Please forgive me if I am asking very basic questions. Currently, I am using Turtlebot waffle pi, which successfully navigates from the initial position to the goal pose. I am using `navigate_w_replanning_and_recovery.xml` Behavior tree. I want to use the Turtlebot for the application of obstacle detection. I want the turtlebot to detect the obstacle in front and stop till the obstacle clears out. The turtlebot should stop only if the obstacle belongs to a certain class (I am using Computer vision for this). Once the path is clear I want to continue the navigation to the goal pose. Could someone please suggest as to how I can achieve this by modifying the Behavior tree?

Thank you in advance!