ros-navigation/navigation2 #1781: Investigate GPU optimizations

SteveMacenski, 2020-05-28:
We should investigate which areas within navigation and the ecosystem that could benefit from GPU optimization. Initial ideas:
- AMCL
- Voxel / obstacle layers
- Inflation layer
- DWB / TEB

Most but the controllers wouldn't require much GPU, the embedded GPU in modern Intel CPUs should be enough to do some low resolution ray casting like those would require or processing inflation cells against some function. 

Ideally, these would be done with OpenCL so that they would be usable for all forms of GPUs. I would prefer not Cuda for that reason, but for exceptional speedups and new features, I would also be ok with Cuda.

SteveMacenski, 2020-05-28:
From another ticket:

[Your link](https://github.com/AutoRally/autorally#1-install-prerequisites) makes alot of sense, things like DWAs or controllers that are sampling a bunch of paths, GPUs make a ton of sense. And especially because TEB, DWB, etc never run as fast as I want, optimizing there would be a really high-impact role. If we can speed up the controllers and we can speed up ray casting for the costmap layers, we could potentially run the local costmap at 10hz or something and the controller at 50hz

SteveMacenski, 2020-06-17:
I do like this as a line of thinking, but I'm going to close .This strikes me as a pipe dream ticket that will likely not be addressed until Nvidia decides that open source is in their best interest or I have an army of Cuda interns.

closed 2020-06-17 by SteveMacenski

saratpoluri, 2022-05-27:
@SteveMacenski Can we discuss your pipe dream. If you had a wish, which local planner would you choose for GPU offload and why?

SteveMacenski, 2022-05-31:
There was a comment here on a discussion I was going to follow up on -- but it appears to be gone now and I'm not sure who that user was... If you're looking at this now, please restart that thread here or in Slack.

saratpoluri, 2022-05-31:
Reposting:
I am looking to implement a local planner on the GPU but I wanted to understand the major pain points when TEB/DWB is running on CPU.

What are the metrics you would like to see improve as a result of GPU offload of the local planner? What is "fast enough" according to your experience?

SteveMacenski, 2022-05-31:
Its running at a sufficiently high rate, while using less CPU resources, and able to evaluate more potential paths. By far, the 2 highest areas of computation in Nav2 are in the costmap layers and in the controller algorithms. The planners and other stuff are pretty much inconsequential related to those two items and the things going on in them are often well suited for parallelization.

Ideally, we would like the local trajectory planners to be running as fast as possible, but a high-aim would be 50hz. I don't think that would be strictly required, but if you aimed for that, you'd certainly be in a good spot. 20hz is a drop dead minimum and many algorithms / hardware (raspberry pi; smaller than a modern intel i5) aren't able to achieve that with DWB and certainly not TEB. Most algorithms need to run relatively fast in order to be stable, so 20hz is that minimum drop-dead but not great performance. For instance, if you run any slower than ~15hz with TEB, it will stop being essentially functional

With that, CPU utilization is a big issue. Even running at only 20hz on a modern intel CPU can use quite a bit of the available resources and typically we have to pare down expectations in the number of potential paths to generate and score -- or time ahead we can simulate as a result of the constraint. Even with making those practical trade offs, the controller is by far one of the most computationally intensive tasks in Nav2 (along with the costmaps). I haven't formally benchmarked the costmaps, but I have a intuition alot of it is around raytracing and iterating over the points in a 3D voxel grid which could also be a target for GPU optimization. 

We're also doing some work on a new controller https://github.com/artofnothingness/mppic if that's of interest

saratpoluri, 2022-06-01:
Understood. DWB will be the initial target then. Costmap is something I haven't looked at yet but I will start with perf analysis to identify the concrete hotspots and see how 3D voxel grid operations can be optimized.

saratpoluri, 2022-06-30:
@SteveMacenski I have profiled the following ROS components with varying parameters and used vtune to understand what are the most time consuming functions:
i. Local Planner (DWB)
ii. Local Costmap
iii. AMCL

I can share the numbers with you over a call. 

For now, I had a question about AMCL. I ran it with default settings and found the avg cpu util = 5%. I increased the max and min particles by 10x and avg cpu util increased to 10%. I increased the update rate by 10x and the cpu util increased to 25%. I gather from sources that default settings for AMCL for the most part are acceptable. Perhaps the higher update rate is useful. Motivation for offloading AMCL seems weak from my measurements. 

I am told a larger map will increase the cpu util and that is where the advantage of GPU offload wrt AMCL will be seen. Is this your expectation and reasoning behind identifying AMCL as a potential target? If yes, I will experiment with how AMCL scales with map size and profile to understand which function within AMCL increases in compute time as result.

SteveMacenski, 2022-06-30:
I agree with your AMCL analysis. 

Costmaps and the local trajectory planners are the biggest hogs of CPU time. AMCL is just an obvious target for GPU optimization given the particle filter, that's mostly why I have that on the list. It does increase a bit over time, but rarely is AMCL the reason I have resource issues. Its usually costmap update rates or local trajectory planners can't run as fast as I'd like them too before they start missing their update look schedule. 

Note: There's discussions about a localization framework (https://github.com/ros-planning/navigation2/pull/1930) that would use a PF as well which would be my personal preference for GPU optimization in localization if that was something you were going to do. With that said, that project is currently on hiatus but it is my intention to get back to that in the long-term future (maybe 2023, if not 2024 for my scheduling unless others could dedicate professional time to it) to replace AMCL. So overall, I agree that spending too much time on AMCL would have diminishing results, DWB & the costmaps would be the best places.

I would suggest the costmaps as the first target, since that's not going anywhere and the basis for all other potential trajectory planners / path planners. DWB is also a great target, though I would bet some non-trivial improvements could be made even without GPU (but GPU might be part of it). I haven't sat down and profiled DWB before, but I also don't think anyone else has either to determine where compute time is being spend and if shuffling around some code would improve performance significantly.