ros-navigation/navigation2 #1783: parameter not declared in dwb_local_planner

Marwan99, 2020-05-29:
https://github.com/ros-planning/navigation2/blob/a9171d0233de1aac5467cb0cf89adc1de0e73dfb/nav2_dwb_controller/dwb_core/src/dwb_local_planner.cpp#L175

`< dwb plugin >.default_critic_namespaces` parameter doesn't seems declared, does it need to be declared?

SteveMacenski, 2020-05-29:
Can you explain the use of this? I don't actually see in the param file where this is used or how it might be used.

Marwan99, 2020-05-29:
From my understanding `<dwb plugin>.default_critic_namespaces` param is expecting a vector of strings representing the namespaces that should include the critics defined in `<dwb plugin>.critics` param. The `DWBLocalPlanner::resolveCriticClassName` method resolves which critic plugin belongs to which namespace (this method implementation can potentially resolve to the wrong critic if two namespaces have the same critic name).

Currently `<dwb plugin>.default_critic_namespaces` is hard coded to `["dwb_critics"]` as in here:
https://github.com/ros-planning/navigation2/blob/a9171d0233de1aac5467cb0cf89adc1de0e73dfb/nav2_dwb_controller/dwb_core/src/dwb_local_planner.cpp#L177

Since the `<dwb plugin>.default_critic_namespaces` param is not declared you can not amend this param from YAML .

I think the infrastructure to allow having critics from different namespaces is already in place, we just need to enable it through exposing the `<dwb plugin>.default_critic_namespaces` param by declaring it.

Does that sound right?

Marwan99, 2020-05-29:
FYI, there is another parameter (`<dwb plugin>.<critic name>.class`) that is declared and read, which can be used to explicitly define the critic class name and namespace
https://github.com/ros-planning/navigation2/blob/a9171d0233de1aac5467cb0cf89adc1de0e73dfb/nav2_dwb_controller/dwb_core/src/dwb_local_planner.cpp#L190-L193

SteveMacenski, 2020-05-29:
What's your suggestion? Just add a declare? That sounds reasonable.

Marwan99, 2020-05-29:
Yeah pretty much.

closed 2020-05-29 by SteveMacenski