ros-navigation/navigation2 #1787: Create 3D static costmap layer and 3D map server

SteveMacenski, 2020-05-30:
For running 3D SLAM, visual slam, and others, it would be useful to have a 3D static layer taking in a 3D map from a map server and populating it with the same mechanics as the existing static layer. For 3D SLAM, its loading the dense map, but VSLAM, its loading the sparse feature map. 

This would involve:
- Using PCL IO PCD to load maps from file and publishing probably a pointcloud to a topic
- Creating a costmap layer to take in this (very, very large) pointcloud and populate it into a 2D or 3D occupancy grid for use in updating the costmap's static map

This is especially important for localization for the global costmap. The global costmap is generally sized by the map size, so we need something to provide the structure of the environment as well as the actual places we can't go from static obstacles. For SLAM sessions, its much less important since the depth information processed into a Voxel Layer will largely create a rough map of the environment as new information gets in the view of the cameras.

sunbo57123, 2020-07-16:
hi, 
From 
>Using PCL IO PCD to load maps from file and publishing probably a pointcloud to a topic

does it mean to add a new msg file for pointcloud, through which the map server could publish pointcloud file information?

naiveHobo, 2020-07-16:
Currently 2D maps are stored in the PGM and published by the `map_server` as `nav_msgs/OccupancyGrid` msgs. Analog to that, the 3D map server should be able to load 3D maps which are stored in the [PCD format](http://pointclouds.org/documentation/tutorials/pcd_file_format.html) and the `map_server` should publish these 3D maps as [`sensor_msgs/PointCloud2`](https://github.com/ros2/common_interfaces/blob/master/sensor_msgs/msg/PointCloud2.msg) msgs.

SteveMacenski, 2022-04-04:
This is baked into https://github.com/ros-planning/navigation2/issues/1757 so closing as redundant to group related topics together

closed 2022-04-04 by SteveMacenski