ros-navigation/navigation2 #1788: ClearCostmapService is not correctly working

daisukes, 2020-05-30:
## Bug report

**Required Info:**
#### problem happens with at least these two commits
- OS: Ubuntu20.04 with ROS2 foxy source build
- Version or commit hash:
  - 4708b5939fbf9be184af29cdf4e48586e9d2d907 (latest master)
  - 55edf58be9eaec3bb12a81e86a97f049e67a8daf (May 18)
- DDS: Fast-RTPS

#### works fine
- OS: Ubuntu18.04 with ROS2 eloquent 
- 8a8810c454c5854907165a0dfd5a7ba4df84e3e2 (Mar. 12)

#### Steps to reproduce issue
```
# Launch tb3 simulation and provide an initial location to start lifecycle
$ ros2 launch nav2_bringup tb3_simulation_launch.py

# Then see the node info
$ ros2 node info /global_costmap/global_costmap_rclcpp_node
```

#### Expected behavior
```
$ ros2 node info /global_costmap/global_costmap
/global_costmap/global_costmap
  Subscribers:
    /clock: rosgraph_msgs/msg/Clock
    /global_costmap/footprint: geometry_msgs/msg/Polygon
    /global_costmap/parameter_events: rcl_interfaces/msg/ParameterEvent
    /map: nav_msgs/msg/OccupancyGrid
  Publishers:
    /global_costmap/clearing_endpoints: sensor_msgs/msg/PointCloud
    /global_costmap/costmap: nav_msgs/msg/OccupancyGrid
    /global_costmap/costmap_raw: nav2_msgs/msg/Costmap
    /global_costmap/costmap_updates: map_msgs/msg/OccupancyGridUpdate
    /global_costmap/global_costmap/transition_event: lifecycle_msgs/msg/TransitionEvent
    /global_costmap/parameter_events: rcl_interfaces/msg/ParameterEvent
    /global_costmap/published_footprint: geometry_msgs/msg/PolygonStamped
    /global_costmap/rosout: rcl_interfaces/msg/Log
    /global_costmap/voxel_grid: nav2_msgs/msg/VoxelGrid
  Service Servers:
    /global_costmap/clear_around_global_costmap: nav2_msgs/srv/ClearCostmapAroundRobot
    /global_costmap/clear_entirely_global_costmap: nav2_msgs/srv/ClearEntireCostmap
    /global_costmap/clear_except_global_costmap: nav2_msgs/srv/ClearCostmapExceptRegion
    /global_costmap/get_costmap: nav2_msgs/srv/GetCostmap
    /global_costmap/global_costmap/change_state: lifecycle_msgs/srv/ChangeState
    /global_costmap/global_costmap/describe_parameters: rcl_interfaces/srv/DescribeParameters
    /global_costmap/global_costmap/get_available_states: lifecycle_msgs/srv/GetAvailableStates
    /global_costmap/global_costmap/get_available_transitions: lifecycle_msgs/srv/GetAvailableTransitions
    /global_costmap/global_costmap/get_parameter_types: rcl_interfaces/srv/GetParameterTypes
    /global_costmap/global_costmap/get_parameters: rcl_interfaces/srv/GetParameters
    /global_costmap/global_costmap/get_state: lifecycle_msgs/srv/GetState
    /global_costmap/global_costmap/get_transition_graph: lifecycle_msgs/srv/GetAvailableTransitions
    /global_costmap/global_costmap/list_parameters: rcl_interfaces/srv/ListParameters
    /global_costmap/global_costmap/set_parameters: rcl_interfaces/srv/SetParameters
    /global_costmap/global_costmap/set_parameters_atomically: rcl_interfaces/srv/SetParametersAtomically
  Service Clients:

  Action Servers:

  Action Clients:

```

#### Actual behavior
```
$ ros2 node info /global_costmap/global_costmap
/global_costmap/global_costmap
  Subscribers:
    /clock: rosgraph_msgs/msg/Clock
    /global_costmap/parameter_events: rcl_interfaces/msg/ParameterEvent
    /map: nav_msgs/msg/OccupancyGrid
  Publishers:
    /global_costmap/clearing_endpoints: sensor_msgs/msg/PointCloud
    /global_costmap/costmap: nav_msgs/msg/OccupancyGrid
    /global_costmap/costmap_raw: nav2_msgs/msg/Costmap
    /global_costmap/costmap_updates: map_msgs/msg/OccupancyGridUpdate
    /global_costmap/global_costmap/transition_event: lifecycle_msgs/msg/TransitionEvent
    /global_costmap/published_footprint: geometry_msgs/msg/PolygonStamped
    /global_costmap/voxel_grid: nav2_msgs/msg/VoxelGrid
    /parameter_events: rcl_interfaces/msg/ParameterEvent
    /rosout: rcl_interfaces/msg/Log
  Service Servers:
    /global_costmap/global_costmap/change_state: lifecycle_msgs/srv/ChangeState
    /global_costmap/global_costmap/describe_parameters: rcl_interfaces/srv/DescribeParameters
    /global_costmap/global_costmap/get_available_states: lifecycle_msgs/srv/GetAvailableStates
    /global_costmap/global_costmap/get_available_transitions: lifecycle_msgs/srv/GetAvailableTransitions
    /global_costmap/global_costmap/get_parameter_types: rcl_interfaces/srv/GetParameterTypes
    /global_costmap/global_costmap/get_parameters: rcl_interfaces/srv/GetParameters
    /global_costmap/global_costmap/get_state: lifecycle_msgs/srv/GetState
    /global_costmap/global_costmap/get_transition_graph: lifecycle_msgs/srv/GetAvailableTransitions
    /global_costmap/global_costmap/list_parameters: rcl_interfaces/srv/ListParameters
    /global_costmap/global_costmap/set_parameters: rcl_interfaces/srv/SetParameters
    /global_costmap/global_costmap/set_parameters_atomically: rcl_interfaces/srv/SetParametersAtomically
  Service Clients:

  Action Servers:

  Action Clients:
```
#### Additional information
If you call `ros2 service list`, you can see the missing services but not listed in the node info. It rarely success to call the service. Once it waits the service, it will never return. It success like less than 5%.

It works fine with my old workspace with a commit in March. 
Small service examples work fine with my Ubuntu20.04/foxy environment.
But, the costmap services are not working.

The code is not changed since Feb. 
https://github.com/ros-planning/navigation2/blob/master/nav2_costmap_2d/src/clear_costmap_service.cpp

SteveMacenski, 2020-06-01:
So what you're saying is that between Eloquent and Foxy, this stopped working reliably? Any other services that also have similar behavior? I'm wondering if this is actually a Nav2 issue or a Foxy services issue.

daisukes, 2020-06-01:
So far, I tested it with three machines and it works with a better CPU. 
It looks like a Foxy services issue. Should I post this issue to [rclcpp](https://github.com/ros2/rclcpp/issues)?

CPU|Host OS|Result
---|---|---
i7-9750|Ubuntu16.04|fail 95%
i7-6700|Ubuntu18.04|fail 100%
i9-7920|Ubuntu18.04|success 95%

SteveMacenski, 2020-06-01:
Yes, that seems like something very good to point out to rclcpp and ping people, that's a **really** big issue. 

So just to confirm, you're saying that there's nothing wrong with clear costmap in particular, or any service in Nav2 in particular; but rather that all services are having this issue?

daisukes, 2020-06-01:
I haven't checked other services, but I'm checking the default QoS of services.
According to the [QoS policies](https://design.ros2.org/articles/qos.html) and [implementation](https://github.com/ros2/rmw/blob/a3774531e9b7289574501275aafbc3f93120360e/rmw/include/rmw/qos_profiles.h#L64), service is published with volatile durability. 

Costmap2D node might be somewhat busy to handle service request and missed them.
It could explain the results I got, better CPU can handle more requests.
What do you think?

SteveMacenski, 2020-06-01:
Maybe, but that still seems really not good. The services should be successfully called. We could change the QoS to make it more stable.

daisukes, 2020-06-01:
First, launch tb3 simulation and gave an initial position, then call some services.

All `get_state` service responded. (btw, costmap nodes are unconfigured. Is this okay?)

```
$ ros2 service list | grep get_state | while read e; do echo $e; ros2 service call $e lifecycle_msgs/srv/GetState "{}"; done
/amcl/get_state
waiting for service to become available...
requester: making request: lifecycle_msgs.srv.GetState_Request()

response:
lifecycle_msgs.srv.GetState_Response(current_state=lifecycle_msgs.msg.State(id=3, label='active'))

/bt_navigator/get_state
waiting for service to become available...
requester: making request: lifecycle_msgs.srv.GetState_Request()

response:
lifecycle_msgs.srv.GetState_Response(current_state=lifecycle_msgs.msg.State(id=3, label='active'))

/controller_server/get_state
requester: making request: lifecycle_msgs.srv.GetState_Request()

response:
lifecycle_msgs.srv.GetState_Response(current_state=lifecycle_msgs.msg.State(id=3, label='active'))

/global_costmap/global_costmap/get_state
waiting for service to become available...
requester: making request: lifecycle_msgs.srv.GetState_Request()

response:
lifecycle_msgs.srv.GetState_Response(current_state=lifecycle_msgs.msg.State(id=1, label='unconfigured'))

/local_costmap/local_costmap/get_state
waiting for service to become available...
requester: making request: lifecycle_msgs.srv.GetState_Request()

response:
lifecycle_msgs.srv.GetState_Response(current_state=lifecycle_msgs.msg.State(id=1, label='unconfigured'))

/map_server/get_state
waiting for service to become available...
requester: making request: lifecycle_msgs.srv.GetState_Request()

response:
lifecycle_msgs.srv.GetState_Response(current_state=lifecycle_msgs.msg.State(id=3, label='active'))

/planner_server/get_state
waiting for service to become available...
requester: making request: lifecycle_msgs.srv.GetState_Request()

response:
lifecycle_msgs.srv.GetState_Response(current_state=lifecycle_msgs.msg.State(id=3, label='active'))

/recoveries_server/get_state
waiting for service to become available...
requester: making request: lifecycle_msgs.srv.GetState_Request()

response:
lifecycle_msgs.srv.GetState_Response(current_state=lifecycle_msgs.msg.State(id=3, label='active'))

/waypoint_follower/get_state
waiting for service to become available...
requester: making request: lifecycle_msgs.srv.GetState_Request()

response:
lifecycle_msgs.srv.GetState_Response(current_state=lifecycle_msgs.msg.State(id=3, label='active'))

```

[... 184 lines cut ...]
response:
rcl_interfaces.srv.ListParameters_Response(result=rcl_interfaces.msg.ListParametersResult(names=['use_sim_time'], prefixes=[]))

/turtlebot3_joint_state/list_parameters
waiting for service to become available...
requester: making request: rcl_interfaces.srv.ListParameters_Request(prefixes=[], depth=0)

response:
rcl_interfaces.srv.ListParameters_Response(result=rcl_interfaces.msg.ListParametersResult(names=['use_sim_time'], prefixes=[]))

/turtlebot3_laserscan/list_parameters
waiting for service to become available...
requester: making request: rcl_interfaces.srv.ListParameters_Request(prefixes=[], depth=0)

response:
rcl_interfaces.srv.ListParameters_Response(result=rcl_interfaces.msg.ListParametersResult(names=['use_sim_time'], prefixes=[]))

/waypoint_follower/list_parameters
waiting for service to become available...
requester: making request: rcl_interfaces.srv.ListParameters_Request(prefixes=[], depth=0)

response:
rcl_interfaces.srv.ListParameters_Response(result=rcl_interfaces.msg.ListParametersResult(names=['loop_rate', 'stop_on_failure', 'use_sim_time'], prefixes=[]))

```

SteveMacenski, 2020-06-02:
That's alot of text but you didn't answer the question. I don't really need to see any of that. You didn't explain what that is seeking to test, please delete that text since it's filling the ticket but doesn't have value in debugging.

daisukes, 2020-06-02:
I have not tried custom QoS, because I may need to learn more about it.

> So just to confirm, you're saying that there's nothing wrong with clear costmap in particular, or any service in Nav2 in particular; but rather that all services are having this issue?

So I tried to answer this question. It looks like not only clear costmap services, but maybe some specific services.

SteveMacenski, 2020-06-02:
Specific ones? Ok, that would be good to figure out some that are fine and some that are not to compare.

daisukes, 2020-06-03:
This problem does not appear if I use Fast-RTPS v1.10.0 instead of 2.0.x. 
https://github.com/ros2/ros2/commit/c7c115a7f9d78985903bac05d7dd0d886b808b61

This upgrade probably introduces this issue.

SteveMacenski, 2020-06-03:
Fun, in that case you should **definitely** file a ticket on ros2 and/or fast-rtps immediately and tag me in them.

SteveMacenski, 2020-06-08:
I rolled this ticket into #1772 and renamed it to fit the scope of the single problem. We should continue discussion there.

closed 2020-06-08 by SteveMacenski

joespeed, 2020-08-18:
> This problem does not appear if I use Fast-RTPS v1.10.0 instead of 2.0.x.
> [ros2/ros2@c7c115a](https://github.com/ros2/ros2/commit/c7c115a7f9d78985903bac05d7dd0d886b808b61)
> 
> This upgrade probably introduces this issue.

does switching to cyclonedds solve the problem?

daisukes, 2020-08-18:
@joespeed Yes, I had been working with cyclonedds until Fast-RTPS is fixed.

SteveMacenski, 2020-08-18:
> until Fast-RTPS is fixed.

I wouldn’t expect that to be soon

JaimeMartin, 2020-08-18:
Hi @SteveMacenski 

Actually, it is fixed. Fast DDS 2.0.1 includes the fix and it is currently included in Foxy. 

@daisukes can confirm this, and he tested the branch I think. 

Anyhow, the open issue is this one https://github.com/ros-planning/navigation2/issues/1772

This issue is actually closed.

daisukes, 2020-08-18:
@JaimeMartin 
Yes, I already confirmed that Fast DDS 2.0.1 fixed this problem.

I was not sure @joespeed 's intention here, but I just wanted to say, it worked well cyclonedds in those days.