ros-navigation/navigation2 #1791: Update parameter maps (controller, planner, recovery plugins) to use a map rather than 2 lists

SteveMacenski, 2020-06-01:
Right now we have our parameters being read in as 2 vectors that we cross correlate. This is obviously unideal, we'd much rather use a map but that's not supported right now by rclcpp parameters. 

We have 2 options:
- Implement the map type parameter in ROS2
- Use namespaces to our advantage. Have a root namespace of the file and a map entry to have a similar parsing schema to a map. This could have a nav2_util and used by recoveries, planners, controllers, and maybe BTs for loading mapped values in navigation2. This mechanic is also used for the costmap sensor data entries, so there's good precedence for it. 

Ex.
```
param_namespace:
  plugins: [plugin_1, plugin_2]
  plugin_1:
    name: "FollowPath"
    plugin: "DWB/DWB_controller"
  plugin_2:
    name: "DockingController"
    plugin "my_custom/DockingController"

```

mikeferguson, 2020-06-17:
I'll note that we use something very close to your second interface in [robot_controllers](https://github.com/fetchrobotics/robot_controllers)

An additional upside is that if you end up wanting to add other non-string parameters for each plugin you can -- I imagine that getting maps with different value types would be a huge mess to implement. For an example, here's what it looks like to [define an entire robot's controllers](https://github.com/fetchrobotics/robot_controllers), including nested namespaces of controllers and all kinds of data types.

SteveMacenski, 2020-06-17:
I don't see an example configuration file in the robot_controllers package to get a good indicator (might be good to poke someone to add, there's no [documentation](http://wiki.ros.org/robot_controllers) on it at all as far as I can tell about how to use / configure it. In my opinion, its almost like you didn't open source it if you don't provide a way for folks to use it without reading all the source code :shrug: 

That is a plus. I'd rather just use a `map` input type, but this would let us expand in the future if we liked.

naiveHobo, 2020-06-17:
An interface along these lines would make more sense I think:

```
param_namespace:
  plugins: [FollowPath, DockingController]
  FollowPath:
    plugin: "DWB/DWB_controller"
    # DWB parameters
  DockingController:
    plugin "my_custom/DockingController"
    # custom params
```

This will also let us keep the plugin-specific parameters in the proper namespace without having to mention the name explicity again and again like what's being done for DWB right now.

https://github.com/ros-planning/navigation2/blob/dd0d9fd9b8415e0bdfaf5d7f373967a65b779900/nav2_bringup/bringup/params/nav2_params.yaml#L92-L95

SteveMacenski, 2020-06-18:
that's essentially what already exists. All of the `FollowPath.XYZ` can be represented as 

```
FollowPath:
  XYZ: 0.5
```

I think what you're suggesting is that we make the namespace / name for it count as well. I would generally agree if the names were arbitrary like in costmap_2d, but the names here matter since they're supposed to be  concepts mapped to implementations. If we did that, my concern would be that people would be more interested in naming the `name` parameter to be convenient in configuration files rather than abstracting the implementation in the behavior tree

naiveHobo, 2020-06-18:
> If we did that, my concern would be that people would be more interested in naming the name parameter to be convenient in configuration files rather than abstracting the implementation in the behavior tree

I understand but isn't that true for any interface we decide to go for? Even with the current setup for example, `FollowPath` could be changed to any arbitrary name in `nav2_params.yaml` without affecting anything, right?

SteveMacenski, 2020-06-18:
Well if we dont make the namespaces dependent on the name for the declaration (e.g. `voxel_layer` namespace is used for nothing other than parameter mapping), then its not coupling the namespace applying this map to the name of the plugin in the server. 

I think I could be convinced of doing it this way pretty easily, admittedly.

naiveHobo, 2020-06-18:
I've implemented an example of the interface I talked about for the controller server in #1808. If it looks reasonable, I can go ahead and do the same for the entire stack.

closed 2020-07-01 by SteveMacenski