ros-navigation/navigation2 #1802: Remove final orientation movement from controller plugins

fmrico, 2020-06-11:
## Feature request

#### Feature description
All plugins in the Controller, when running `FollowPath` node in BT, must implement the final orientation maneuver. This is a very basic maneuver. In some cases, being part of every Controller plugin must be an inconvenience:

- Visiting waypoints. The robot does not need to do this maneuver in the intermediate points.
- When the final orientation is not relevant.
- When the final maneuver could be special.
- The final maneuver could always return RUNNING (e.g. following a dynamic point to a distance, where the final orientation is not the one in the last pose of the path)

I propose to create a BT control node to do the final orientation maneuver, instead of being part of FollowPath. The BTs will include this node in a sequence after FollowPath. This will let people to create their own final maneuver if needed.

I can be working on this next week if you find it interesting.

#### Implementation considerations

SteveMacenski, 2020-06-11:
> must implement the final orientation maneuver.

I suppose they don't _have_ to, but it would be correct if they did since you're going to a pose not a point. But you could create a goal checker plugin that only checks for the XY position and not the orientation and disable the rotate to pose in the controller.

Certainly that could be capability added for a specific application. Keep in mind that the BT node ticks at 10 hz, so then the controller could only run at 10 hz unless its another BT action node calling another action server to do it. In fact, you could just implement this as another controller plugin in the controller server that just handles the rotate the goal. Then this is called in the BT after the XY pose is met using the goal checker with orientation set to null. 

I'm not sure though if that would be the best default behavior for the majority of users though. We could make any new parameterizations required to enable you to do it though (e.g. param in the DWB controller for whether to rotate to goal)

fmrico, 2020-07-08:
Hi @SteveMacenski 

I have closed this issue because I got the desired behavior in #1859 without this functionality.

Anyway, maybe we still consider it useful to decouple the final maneuver from the controller functionality. Or perhaps with another BT node after reaching the Pose is enough.

Anyway, I am closing it.

closed 2020-07-08 by fmrico