ros-navigation/navigation2 #1803: [nav2_map_server] Any reason for using a custom find*.cmake instead of the system one

doisyg, 2020-06-12:
nav2_map_server package uses a custom `find*.cmake` 
https://github.com/ros-planning/navigation2/blob/master/nav2_map_server/cmake_modules/FindGRAPHICSMAGICKCPP.cmake
Is it still needed ?

SteveMacenski, 2020-06-12:
@rotu any input?

ShivamPR21, 2020-06-12:
@doisyg hope this link will help [https://stackoverflow.com/a/23838081](either try to find pkgConfig.cmake or try to find it in default paths of your system)

rotu, 2020-06-12:
I don't know what alternative you mean in the title. I wasn't able to find a CMake package for GraphicsMagick at the time I developed this, and if I did, I would have preferred to use that.

SteveMacenski, 2020-06-12:
There you have it @doisyg. If there's another option, we'd be open to hearing about it

SteveMacenski, 2020-06-18:
@doisyg have you found an alternative we should consider?

doisyg, 2020-06-18:
Not 100% sure but I will try to PR next week or close this otherwise

closed 2020-06-23 by SteveMacenski

doisyg, 2020-06-24:
Sorry, I confused GraphicsMagick and ImageMagick which have some header files in common (i did a non trinary implementation of the map saver a couples of years ago with ImageMagick)