ros-navigation/navigation2 #1804: /map topic in the master branch

rjshim, 2020-06-15:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Version or commit hash:
  - from master source
- DDS implementation:
  - default Fast-RTPS

#### Steps to reproduce issue

```
$ ros2 launch nav2_bringup bringup_launch.py with the map location as an argument
$ ros2 topic echo /map
```

#### Expected behavior
The terminal should print out the /map topic data 

#### Actual behavior
No data printed out
It also didn't work when I tested it with Rviz showing error code "No tf data. Actual error: Frame [map] does not exist"

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

Thank you for help in advance!
Ryan

SteveMacenski, 2020-06-15:
Please provide a full exact command & the terminal output. Did you wait until the lifecycle nodes fully transitioned up? Are you sure that the map path you provided was taken / valid?

rjshim, 2020-06-19:
@SteveMacenski 

I will wait for the foxy release first and reopen this issue if I need help. For now, it doesn't work but it might be just because I do not understand the package well. The path to the map file shouldn't be the issue because it works in the Dashing version.  

Thank you for your help. 
Ryan

closed 2020-06-19 by rjshim

rjshim, 2020-06-30:
@SteveMacenski 
 
I was running in my launch file some files from this repository, and I hadn't realized that some of their paths changed. I fixed them and everything worked perfectly fine. Thank you very much for your help.

Ryan