ros-navigation/navigation2 #1810: Push test coverage > 80%

SteveMacenski, 2020-06-18:
Right now, test coverage is pretty OK for an open source ROS project, but that's a pretty low bar. We have 66% test coverage as of writing with some real efforts to get it up to that point. Now we should make an actual plan of things we need to add tests for to push this higher since many of the easier nickel-and-diming tests are already written. 

- [ ] nav2_behavior_tree at 57% coverage, need to add more tests to our behavior tree nodes
- [ ] Costmap 2D only has 60% coverage
- [ ] Nav2 Core is at 85%, we need to call 1 extra function to get 100%
- [ ] DWB controller is at 70% coverage
- [ ] nav2_util is at 65% coverage, we need to test the param dump utility
- [ ] rviz plugins have 0% coverage

SteveMacenski, 2020-06-18:
These ar esome of the ones we should look at, but any and all additional tests are desired

naiveHobo, 2020-06-21:
> Nav2 Core is at 85%, we need to call 1 extra function to get 100%

`nav2_core::GoalChecker::reset()` is the missing call. It doesn't get called since the derived classes `dwb_plugins::SimpleGoalChecker` and `dwb_plugins::StoppedGoalChecker` have their own `reset()` functions defined and the base class `reset()` (which is empty anyway) doesn't get called.

We can make `nav2_core::GoalChecker::reset()` a pure virtual function to get to 100% coverage.

SteveMacenski, 2020-06-22:
Is reset the only function in goal checker with a default empty method? If so, sounds like a plan.

SteveMacenski, 2020-08-18:
Still more tests we could make, but we just hit ~80%

closed 2020-08-18 by SteveMacenski